JMRI/JMRI #6134: Deadlock in operations "Import Cars" on macOS

bobjacobsen, 2018-11-12:
I'm seeing an intermittent deadlock when running graphical AllTest on macOS Mojave with Java 8. Thread info follows.  The cause seems to be opening a FileChooser dialog off the Swing thread.

 I'll look at this ASAP, but might be a day or two.  

Deadlock report:
```
     [java] Found one Java-level deadlock:
     [java] =============================
     [java] "Test Import Cars":
     [java]   waiting to lock monitor 0x00007fa3e6402ee8 (object 0x00000006da6e9190, a java.lang.Object),
     [java]   which is held by "AWT-EventQueue-0"
     [java] "AWT-EventQueue-0":
     [java]   waiting to lock monitor 0x00007fa3e254cb98 (object 0x00000006c0af5488, a java.awt.Component$AWTTreeLock),
     [java]   which is held by "Test Import Cars"
```

First locked thread:
```
     [java] "Test Import Cars":
     [java] 	at com.apple.laf.AquaFileSystemModel.getRowCount(AquaFileSystemModel.java:193)
     [java] 	- waiting to lock <0x00000006da6e9190> (a java.lang.Object)
     [java] 	at javax.swing.JTable.getRowCount(JTable.java:2662)
     [java] 	at javax.swing.plaf.basic.BasicTableUI.createTableSize(BasicTableUI.java:1692)
     [java] 	at javax.swing.plaf.basic.BasicTableUI.getPreferredSize(BasicTableUI.java:1733)
...
     [java] 	at javax.swing.JFileChooser.createDialog(JFileChooser.java:821)
     [java] 	at javax.swing.JFileChooser.showDialog(JFileChooser.java:749)
     [java] 	at javax.swing.JFileChooser.showOpenDialog(JFileChooser.java:656)
     [java] 	at jmri.jmrit.operations.rollingstock.ImportRollingStock.getFile(ImportRollingStock.java:40)
     [java] 	at jmri.jmrit.operations.rollingstock.cars.tools.ImportCars.run(ImportCars.java:91)
```

Second locked thread:
```
     [java] "AWT-EventQueue-0":
     [java] 	at java.awt.Component.invalidate(Component.java:2921)
     [java] 	- waiting to lock <0x00000006c0af5488> (a java.awt.Component$AWTTreeLock)
     [java] 	at java.awt.Container.invalidate(Container.java:1585)
     [java] 	at javax.swing.JComponent.revalidate(JComponent.java:4853)
     [java] 	at javax.swing.JTable.tableRowsInserted(JTable.java:4482)
     [java] 	at javax.swing.JTable.tableChanged(JTable.java:4407)
     [java] 	at javax.swing.table.AbstractTableModel.fireTableChanged(AbstractTableModel.java:296)
     [java] 	at javax.swing.table.AbstractTableModel.fireTableRowsInserted(AbstractTableModel.java:231)
     [java] 	at com.apple.laf.AquaFileSystemModel$DoChangeContents.run(AquaFileSystemModel.java:458)
     [java] 	- locked <0x00000006da6e9190> (a java.lang.Object)
     [java] 	- locked <0x00000006deb2ac30> (a java.lang.Object)
     [java] 	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
     [java] 	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:744)
...
     [java] 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
     [java] 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
     [java] 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
     [java] 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

The main (test) thread is at:
```
     [java] "main" #1 prio=5 os_prio=31 tid=0x00007fa3e1000000 nid=0x2503 in Object.wait() [0x000070000c3fa000]
     [java]    java.lang.Thread.State: WAITING (on object monitor)
     [java] 	at java.lang.Object.wait(Native Method)
     [java] 	at java.lang.Object.wait(Object.java:502)
     [java] 	at java.awt.EventQueue.invokeAndWait(EventQueue.java:1301)
     [java] 	- locked <0x00000006deb2b690> (a java.awt.EventQueue$1AWTInvocationLock)
     [java] 	at java.awt.EventQueue.invokeAndWait(EventQueue.java:1282)
     [java] 	at javax.swing.SwingUtilities.invokeAndWait(SwingUtilities.java:1348)
     [java] 	at jmri.util.ThreadingUtil.runOnGUI(ThreadingUtil.java:123)
     [java] 	at jmri.util.JUnitUtil.dispose(JUnitUtil.java:1026)
     [java] 	at jmri.util.JUnitUtil.resetWindows(JUnitUtil.java:973)
     [java] 	at jmri.util.JUnitUtil.tearDown(JUnitUtil.java:272)
     [java] 	at jmri.jmrit.operations.OperationsTestCase.tearDown(OperationsTestCase.java:36)
     [java] 	at jmri.jmrit.operations.rollingstock.cars.tools.ImportCarsTest.tearDown(ImportCarsTest.java:115)
     [java] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
```

silverailscolo, 2018-11-12:
Not sure this is the same, but it's also a File Chooser dialog staying open on screen: https://github.com/JMRI/JMRI/pull/6115#issuecomment-437516678

bobjacobsen, 2018-11-13:
Thanks!  #6139 solves the formal deadlock, but hangs with this File Chooser on the screen:
[image]

If I navigate down and select the ExportOperationsCarRoster.csv file
[image]
the window closes and the test completes OK.

DanielBoudreau, 2018-11-14:
Could be related to how the test code was written:

       // opens file chooser path "operations" "JUnitTest"
        JFileChooserOperator fco = new JFileChooserOperator();
        String[] path = OperationsXml.getOperationsDirectoryName().split(Pattern.quote(File.separator));
        fco.chooseFile(path[0]);
        fco.chooseFile(path[1]);
        fco.chooseFile(ExportCars.getOperationsFileName());

Most likely need to wait or pause between each file chooser operation.  Not sure how to fix.

Dan

DanielBoudreau, 2018-11-14:
I might have a fix for this:
Fix dialog window issue #6142