JMRI/JMRI #6135: Signal Mast - Output Matrix - XpressNet - does not sent all commands

sidlo64, 2018-11-12:
PanelPro 4.13.4
Signal mast defined as controlled by Output Matrix with 5 logic outputs XT116, XT117, XT118, XT119 and XT120.
Changing aspect with more than 3 logic output at the same time does not work reliably.
Aspect are controlled from Table Signal Mast - column Aspect.
Aspect 00000 - Stůj
Aspect 11110 - 40 a opakovaná výstraha

Procedure:
0) Set aspect with output 00000 - OK
1) Set aspect with output 11110 - OK
2) Set aspect with output 00000 - error - result is 00100, command for XT118 was not send.

See monitor log.
Table configuration is in test-bug.xml.

[monitorLog.txt](https://github.com/JMRI/JMRI/files/2573029/monitorLog.txt)
[test-bug.xml.txt](https://github.com/JMRI/JMRI/files/2573032/test-bug.xml.txt)

silverailscolo, 2018-11-12:
Thanks for pointing this out. Using the XPressNet Simulator this is not seen on my iMac in 10 tries. What hardware are you using for this? Perhaps it's serial comm or stationary decoder related? I could easily add a delay between each of the output commands for one aspect, but you would want to set its length somewhere other that the mast config pane.
As I also saw this mentioned in a couple of other recent jmriusers threads, it could perhaps be implemented as a system wide setting?

pabender, 2018-11-13:
@silverailscolo I doubt you will ever see this on a simulator because of the way the responses are generated on real hardware.

@sidlo64 I don't see aspect 00000 defined in the file.  

One thing you can try is setting the feedback mode of the turnouts involved to either DIRECT or SIGNAL mode.  This will bypass some of the feedback responses, which may be the cause of the issue.  (The messages that are bypassed are intended to make sure we don't burn out solenoid switch machines. with the way that accessory decoders work on the Lenz LZV100 and LZ100... )

I'll see if I can play with this on real hardware later today or tomorrow.

sidlo64, 2018-11-13:
@pabender I did some setup attempts with DIRECT an SIGNAL, but the results were the same.
The SIGNAL setting is not described in the documentation [http://jmri.org/help/en/html/doc/Technical/TurnoutFeedback.shtml](url)

With connection Simulator I had same result as @silverailscolo 
[monitorLog-simulator.txt](https://github.com/JMRI/JMRI/files/2577821/monitorLog-simulator.txt)
Interestingly, the commands are sorted sequentially.

In the recent past, I used the LI100F interface connection on the Paco Canada design [http://usuaris.tinet.cat/fmco/lokmaus_en.html#genli](url) 
[monitorLog-LI100F.txt](https://github.com/JMRI/JMRI/files/2577841/monitorLog-LI100F.txt)
Because the connection was slow 9600 baud and there were trouble with the signals, I left this solution. I suspected GenLI100 was to blame.

Now I'm using GenLI101 from [http://www.digi-cz.cz/obchod/obvod-rozhrani-gen-li-101/](url) This interface is completely different. Unfortunately, it has the same problems.
[monitorLog-LI101.txt](https://github.com/JMRI/JMRI/files/2577867/monitorLog-LI101.txt)

@pabender You can use any signal system [http://jmri.org/help/en/html/tools/signaling/AspectSignaling.shtml](url) For aspect test set one aspect code 00000 and the other 11110. It does not matter what you use the aspects.
I have used the SŽDC signal [http://jmri.org/xml/signals/SZDC-2015-zakladni/index.shtml](url) system, but it is not a prerequisite.

silverailscolo, 2018-11-25:
After PR #6177 is merged, please try it and see what interval (in Milliseconds) keeps the individual turnout commands from being lost over the connection. The interval (wait) is applied _before_ each output command.

pabender, 2018-11-26:
@sidlo64 

I've finally had a chance to try this with an LZV100 and an LIUSB-ETHERNET (in Ethernet mode) and I can replicate what you are seeing.

Where the turnouts don't all change to the direction they should in response to setting the signal, I see a request to operate the turnout, such as:

23:48:01.262: [packet:52 19 8B C0]   Accessory Decoder Operations Request: Turnout Address 102(Base Address 25,Sub Address 1) Turn Output 1 On.

followed by a command station busy response from the command station:

23:48:01.265: [61 81 E0]   Command Station Busy

JMRI's response to the Command Station Busy is to retry the command.  

It might fail more than once and then we eventually get a response back from the command station indicating the turnout status changed.  When I see an issue, the command station is reporting feedback for the address that is opposite to the expected value.  

I'm curious as to why you never see the command station busy messages (I think you should)

I am not working where I could actually see anything operate on the layout (which is in the basement, and I'm not, but Watching the turnout table, I can actually see the turnout go to the correct state and then flip to the incorrect state based on the feedback.  

That said, I can explain why you are seeing the behavior you see and will give you two options to fix this:

First, the reason WHY this is happening is that XPressNet command stations that send feedback responses to turnout messages (which bsaically is the Lenz LZ100 and LZV100) return the feedback for a pair of turnouts.  

When a turnout fails to end up showing in JMRI the correct feedback state, it is because of the order in which the feedback messages were processed.  This is a race condition I need to track down, but it is of pretty low priority right now.

Here are two options for how to work around this.

1) Set the feedback mode to either DIRECT or SIGNAL feedback mode.  With both of these modes, the values in the feedback is ignored.
2) Skip addresses so you don't end up with two addresses in the same pair on the signal mast.   Instead of using XT116,XT117,XT118,XT119,XT120 use XT116,XT118,XT120,XT122,XT124 or XT117,XT119,XT121,XT123,XT125.

I know you already said you tried DIRECT and SIGNAL mode feedback for the turnouts involved and that didn't work, but it really should work (it works on my test case, even using your signal definition).  

Also, I am curious, did you try this with an actual signal head or just by looking at messages?  

I should be able to try a real signal head, but it will be another few days.

sidlo64, 2018-12-01:
@pabender 
I have newer recieved response from command station that is a busy. As you already write, the answer is always for couple of turnouts. Log of communication is here 
[monitorLog-all.txt](https://github.com/JMRI/JMRI/files/2635784/monitorLog-all.txt)

I am working with real decoder like this [https://sites.google.com/site/sidloweb/elektrika/28-uni16ard-nav-mx/uni16ard-nav-mx-nastaveni](url)

I changed the decoder program. Now decoder listens only for even addresses. Everything works as should. Log of communication is here 
[monitorLog-even.txt](https://github.com/JMRI/JMRI/files/2635790/monitorLog-even.txt)

Your advice work around with even addresses works perfect.

sidlo64, 2019-06-19:
PR #7006 fix this issue.
Tested with 
command station NanoX-S88 [http://www.digi-cz.cz/obchod/centrala-nanox/](url)
interface Gen-Li 101 [http://www.digi-cz.cz/obchod/obvod-rozhrani-gen-li-101/](url) 
signal decoder UNI16ARD-NAV-MX [https://sites.google.com/site/sidloweb/elektrika/28-uni16ard-nav-mx](url)
delay between turnouts commands 500 ms.

closed 2019-06-19 by sidlo64