JMRI/JMRI #6144: Engine Driver and JMRI fail when stealing an engine

frankfezzie, 2018-11-14:
Using Engine Driver connected through JMRI, when I attempt to acquire an engine that is "In Use" on a Digitrax throttle I see a "steal" message on my phone with Y or N options.  Regardless of which option I choose, Engine Driver will not acquire the engine and any further attempt to acquire any engine with Engine Drive fails, even after stopping and restarting Engine Driver.  In addition, JMRI will no longer allow any throttle on the computer screen to acquire any engine.  The only way I can find to resume normal operation is to close and restart JMRI.

I have confirmed this problem on my home layout and two separate club layouts, all running similar hardware and software.  I also have reports from another club experiencing the same problem.

Setup
Digitrax DCS100 command station
Digitrax DT402D throttle running rev 21 software and operating plugged into the Loconet
HP Laptop running Windows 10
JMRI versions  Production 4.12, test 4.13.3, and test 4.13.4  - all show the same problem
Locobuffer USB interface
Engine Driver 2.22.69 running on Samsung Galaxy S2 with Android version 4.1.2

Frank Fezzie 
_._,_._,_

frankfezzie, 2018-11-15:
At Steve Todd's request I did an additional test.   If I open a new throttle in JMRI on the computer screen and use it to try to acquire an engine that is “In Use” on a Digitrax throttle the Steal feature seems to work properly if I choose the "Y" option.  If I choose the "N" option then the JMRI throttle on screen will not allow any engine to be selected.  This test was done with test release 4.13.4

Frank Fezzie

Pugwash1, 2018-11-15:
@mstevetodd #4722
The start of the problem happens in DeviceServer.java Line 710, notifyControllerAddressDeclined, where it cancels the throttle request, before recieveing a steal/nosteal. When the Steal is chosen MultiThrottleController then issues a StealThrottleRequest that cannot be executed due to the cancelledthrottlerequest... I dont understand why it feels it needs to cancel the request.

BHoffman351, 2018-11-15:
> On Nov 15, 2018, at 9:50 AM, Steve G <notifications@github.com> wrote:
> 
> I dont understand why it feels it needs to cancel the request.

This is in caps because it is not being heard:

BECAUSE STEAL IS NOT IMPLEMENTED!

Observe this line (539) from jmri.jmrix.loconet.LnThrottleManager.java:

log.warn("user agreed to steal address {}, but no code is in-place to handle the 'steal' (yet)",address.getNumber());

Everyone is asking why it does not work correctly.  That is why.  It is not done.  Why the withrottle package of JMRI does not follow the precise steps is because this is untestable.  It is making the best of the situation.  I can refine the WiThrottle Server code to work properly but it still does not make the code that is not there steal an address.  The car’s engine is seized, switching to premium fuel is not going to help.

There were underlying problems with JMRI's LocoNet handling that needed to be fixed and steal was put on the shelf.

I will take a look and see if there is a clean way for me to temporarily remove the “steal” process from the WiThrottle Server until such time as steal is working.  Hopefully that will stop the endless barrage of support cases that Steve and I are receiving.

Regards,
Brett

pabender, 2018-11-15:
All,

One of the things I am working on now is tests for the steal process on the UI and the WiThrottle server.  This is something that is not currently tested at all and it needs to be.

The LocoNet Throttle Manager also needs to have tests in place to verify it does the right thing, but I am not sure I know enough about the process to write that.

pabender, 2018-11-16:
All,

I have written a tests for the JMRI throttle GUI ( see #6149 ), and that works as expected (per the sequence diagram in the ThrottleListener javadoc that is (see http://jmri.sourceforge.net/JavaDoc/doc/jmri/ThrottleListener.html ), so the issues with the JMRI throttle may very well be tied to the LocoNetThrottleManager implementation.

The issue with the WiThrottleServer however is different.  WiThrottleServer doesn't actually follow the expected sequence for the steal request.  This was documented previously in #4722 (which @Pugwash1 mentioned ).  The sequence should look like the first one mentioned (what happens on the Digitrax LNWI) but it doesn't.

The tests that I'm adding are using a derivative of the DebugThrottleManager that REQUIRES the steal to be processed as designed (see https://github.com/JMRI/JMRI/blob/master/java/test/jmri/managers/StealingThrottleManager.java )

I see two issues we need to solve.

1) the issue that prevents correct handling of the "yes" case in the WiThrottle Server is one of not maintaining state properly.  When the server notifies the device that a steal is required, the expected response from the client to confirm the steal is to repeat the request but that request executes exactly the same code as the original request.  

As a result of this sequence, WiThrottle never actually notifies the ThrottleManager whether it wants to steal or cancel the request.  

Because there is no notification to the manager of the decision of the client, The manager isn't able to clean up the open request for the address.

2) If I am reading the protocol correctly, there is no message the client can send to reject the steal request.  We could handle that with a timer or, perhaps, waiting for the next request to come in with a different address, but neither of those solutions is ideal.

My test for throttle stealing in the WiThrottle Server is in PR #6150.  I'm happy to make the modifications required to make that test pass.  That should resolve the issue in #4722, when yes is selected in response to the steal question, but we need to figure out how to handle the rejection case, which is the issue here.

BHoffman351, 2018-11-16:
> On Nov 15, 2018, at 1:28 PM, Steve G <notifications@github.com> wrote:
> 
> I hate to be the bearer of bad news, but it does issue a steal request when the user responds "Y", it just causes a NPE dues to the slotForAddress failing.
> The line after the warn is "commitToAcquireThrottle(slotForAddress.get(address.getNumber()));" that works correctly for the standard JMRI throttle. interface.

Steve, That is not being the bearer of bad news.  If someone had continued work on the steal process, I was not aware of it.  

> The start of the problem happens in DeviceServer.java Line 710, notifyControllerAddressDeclined, where it cancels the throttle request, before recieveing a steal/nosteal. When the Steal is chosen [...]

This is because the choice is made earlier in the ThrottleController.java.  The method notifyStealThrottleRequired() will just pass it up to notifyControllerAddressDeclined(). This was the correct behavior without the steal process working.  If it could not steal, it had to clean up the request and should not give the option to steal.  With the additional code that was added, essentially it broke the process as the sequence is now altered.

And yes, that is the intended behavior.  It cancels the original request and then the user choses to make a new request as a steal or make no request.  This prevents needing some form of negative request from the user and leaving address requests hanging in limbo.  I will work on seeing what needs to change at this point.

The LNWI was the first item to have steal in it that the WiThrottle protocol was extended for.  The LNWI does not have recognition for a negative response from the user.

In looking at this last night, the option of “No” when choosing to steal does not work and locks down the LocoNet.  This is with a JMRI on-screen throttle.  Whether WiThrottle sends a no, or the WiThrottle Server sends a no, does not matter.  This causes address selection to fail from that point onward.  Even for other devices.  Even for on-screen throttles.

Regards,
Brett

Pugwash1, 2018-11-16:
Brett, I spent some time last night and again this morning working thru this, before reading your comments, and guessed that the cancel  needed to be as No didn't return anything.
I think the more work, but less hassle for the future, way would be add a method for creating anew throttle with an automatic steal, this would prove useful going forward. There is a boolean for readonly throttles that gets passed, if that was an object it could carry some other info, like whether to autosteal, never steal, prompt , share etc, it wouldn't take too long.

pabender, 2018-11-16:
@Pugwash1 that actually was the original steal implementation.  The code to do that is commented out here:

https://github.com/JMRI/JMRI/blob/1a18219b785a17534ac3cf01109b9ba2c4683566/java/src/jmri/jmrit/withrottle/ThrottleController.java#L216-L223

pabender, 2018-11-16:
@Pugwash1 I don't actually agree that going back to an automatic steal is the right thing to do.

Since the JMRI throttle also appears to have issues on LocoNet, the LocoNet steal implementation does need to be fixed. The JMRI throttle front end does the steal correctly as designed.

Pugwash1, 2018-11-16:
Sorry, I wasn't clear.
1 provide for create throttle with an option that can be steal, prompt, never steal , share ad nauseam...
2 aquireing throttles would  normally start with prompt, return a throttle or ask for what to do. The code could then steal, or not. In the Wi-Fi case it would release the request when it asks for the prompt and then if ever gets a steal command from the device it would aquire with steal.  I think it should have been done that way from the start, illiminating the need for timers to release requests. 
This also decomplicates JYTHON type calls, which might default to share.

grandle, 2018-11-16:
I agree that withrottle is not maintaining its state on callback.  But after some real world testing of this problem this is what I found the problem to be -- 

The throttle in JMRI does steal a digitrax throttles engine, if you answer yes to the steal question.  If you answer no, it hangs the JMRI throttle and your only option is to restart JMRI.

As you are aware from the bug report, if you try and steal an address from any of the wifi apps it hangs the wifi server and you can do nothing with apps until you restart JMRI.

Also, if JMRI has ever used this engines address during its current running session, it will not ask the steal question at all and it will simply share the address.

The problem is in the LnThrottleManager.java code, code segment follows

----------------------------

    public void notifyChangedSlot(LocoNetSlot s) {
        log.debug("notifyChangedSlot - slot {}, slotStatus {}", s.getSlot(), Integer.toHexString(s.slotStatus()));
        // This is invoked only if the SlotManager knows that the LnThrottleManager is
        // interested in the address associated with this slot.

        // need to check to see if the slot is in a suitable state for creating a throttle.
        if (s.slotStatus() == LnConstants.LOCO_IN_USE) {
            // loco is already in-use
            log.warn("slot {} address {} is already in-use.",
                    s.getSlot(), s.locoAddr());
            // is the throttle ID the same as for this JMRI instance?  If not, do not accept the slot.
           if ((s.id() != 0) && s.id() != throttleID) {
                // notify the LnThrottleManager about failure of acquisition.
                // NEED TO TRIGGER THE NEW "STEAL REQUIRED" FUNCITONALITY HERE
                //note: throttle listener expects to have "callback" method notifyStealThrottleRequired
                //invoked if a "steal" is required.  Make that happen as part of the "acquisition" process
               slotForAddress.put(s.locoAddr(),s);
               notifyStealRequest(s.locoAddr());
              // return; --- THIS IS THE RETURN THAT I AM TALKING ABOUT -----------
            }
        }
        commitToAcquireThrottle(s);
        
    }
-----------------------------------------------------------------------------------------------------
If I comment out the return in the above code and everything appears to work correctly.  Nothing hangs and you get the expected result == you can control the engine and a no response to either the app or the internal throttle does not hang the system.  

I am not sure why the return is needed because it destroys the state of the thread.  It may be nice to have a discussion about the all of the high level issues in the what the code is doing, but this fix solves the problem and a working code that does not hang JMRI and require a restart of the code is probably a better goal to want to achieve.

The comment about the WARNING that says that the code has not be written is a red herring -- somebody spent some time trying to implement the steal and it does work if the "RETURN" is removed.

pabender, 2018-11-16:
Sent from my iPhone

> On Nov 16, 2018, at 3:20 PM, grandle <notifications@github.com> wrote:
>            slotForAddress.put(s.locoAddr(),s);
>            notifyStealRequest(s.locoAddr());
>           // return; --- THIS IS THE RETURN THAT I AM TALKING ABOUT -----------
>         }
>     }
>     commitToAcquireThrottle(s);
>     
> }

> I am not sure why the return is needed because it destroys the state of the thread. It may be nice to have a discussion about the all of the high level issues in the what the code is doing, but this [...]

Removing the return basically ignores the system specific behavior of stealing, in other words, it ignores what the throttle does in response to the notifyStealRequest() method above the return.

This was the behavior before the steal was implemented.

There is another callback that the front end makes to the backend to finish the steal request.

I can look at that later, but I am a small screen right now, which makes it harder to look for the code.

pabender, 2018-11-16:
@grandle 

Here is some history of how we got here:
https://github.com/JMRI/JMRI/issues/2436
https://github.com/JMRI/JMRI/issues/4062

There is probably more out there...

I did the work to add the basic support for stealing to the throttle listener, along with putting it in place in the  I understand the problem at that level, but I won’t go into the system specific part of this other than to say this LocoNet is really the only system that uses it. @Pugwash1 @devel-bobm and @bobjacobsen can give you a better idea of the LocoNet specific implementation.

There is a sequence diagram in the ThrottleListener JavaDoc that explains what should happen:
http://jmri.sourceforge.net/JavaDoc/doc/jmri/ThrottleListener.html

My bottom line is that there really are two issues we are looking at:
1) the LocoNet implementation appears to have a bug.
2) the WiThrottle server doesn’t do all of the communication it should with the throttle manager.

In the LocoNet specific code, this should be executed after the steal request:

https://github.com/JMRI/JMRI/blob/cfa170cd341ad833214e1a92b0083441f182c7a5/java/src/jmri/jmrix/loconet/LnThrottleManager.java#L529-L542

I would look there for issues.

BHoffman351, 2018-11-16:
> On Nov 16, 2018, at 10:37 AM, Steve G <notifications@github.com> wrote:
> 
> I spent some time last night and again this morning working thru this, before reading your comments, and guessed that the cancel needed to be as No didn't return anything.
> I think the more work, but less hassle for the future, way would be add a method for creating anew throttle with an automatic steal, this would prove useful going forward. There is a boolean for readonly throttles that gets passed, if that was an object it could carry some other info, like whether to autosteal, never steal, prompt , share etc, it wouldn't take too long.

Steve, That is exactly what was there.  
WiThrottle requests address
JMRI says “no, if you want it - gotta steal”
JMRI forgets the request even happened, no change to address
WiThrottle has a choice
No - sends no response, everything continues as before
Yes - sends “Give Me That Address!”
If JMRI sees the forceful request, it does a steal
It was set up to do this with the existing code, but the steal mechanism being broken causes the lockup in JMRI.  The cancel request is broken and I use that prior to deciding to steal or not.  That is why WiThrottle causes the lockup either way.  Everything should be available in JMRI already, it just needs some adjustment in the withrottle package once the steal is actually working.
That is how AJ and I discussed and implemented steal on our respective ends when the LNWI was in development.

Regards,
Brett

pabender, 2018-11-16:
Thinking about this, I wonder if the issue in the decline to steal case is here:

https://github.com/JMRI/JMRI/blob/cfa170cd341ad833214e1a92b0083441f182c7a5/java/src/jmri/jmrix/loconet/LnThrottleManager.java#L534

The failedThrottleRequest is in the AbstractThrottleManager.

I think I have a pretty quick test I can write that would verify whether or not this occurs correctly.

pabender, 2018-11-17:
All, here is a question:

Should the result of saying no to the steal be to cancel the request or to fail the request.

As implemented in the LocoNet code, a failedThrottleRequest is issued, which was discussed in one of the prior threads on this (I think it may have even been my idea).

But, by definition, saying no to the steal is the user saying "no, I really don't want to do that now", which is canceling the request....

The real question becomes if we have n throttle listeners waiting for an address, and one of them decides it doesn't want to steal, should we invalidate all the other waiting listeners?  This is what calling failedThrottleRequest does.  (failedThrottleRequest also has the side effect of poping up a dialog that says there was an error acquiring the address and the error sent by the LocoNet code is "user chose not to 'steal' the throttle" which the user knows, because they just said no to the request....)

(remember that the throttleListener objects represent some piece of code that is waiting for the throttle object.  Unless we are loading a throttle from a file (either a script or by loading saved throttles) it is unlikely that we will have more than one.

At a minimum, I think the request from the non-stealling throttleListener should be canceled before any failedThrottleRequest calls are made to avoid notifying the user of the decision that was just made.

Opinions?

pabender, 2018-11-17:
Ok Folks, I found the cause of not being able to ask for a second throttle on the LnThrottleManager.

LnThrottleManager's requestThrottleSettup starts like this:
https://github.com/JMRI/JMRI/blob/aee4f6b6fac2dab17b12a1205a5da980e3c6156a/java/src/jmri/jmrix/loconet/LnThrottleManager.java#L65-L68

that flag, requestOutStanding, was reset when a steal was accepted by the user, but never reset if the steal was rejected.... I am putting together a pull request to fix that.

pabender, 2018-11-17:
PR #6154 resets the requestOutstanding flag when a steal is rejected.

I tested this works using the LnThrottleManager on a Roco Z21 system (with a DT402 attached to play second throttle).

That should take care of the issues with the mechanism with regard to loconet.  

I also commented out the the warning message before the commitToAquire call in the stealThrottleRequest method and added a comment as to why.  Having the warning there is confusing because we can steal slots, we just don't implement the ZapSteal (I think that is right term) of the newer Digitrax command stations.

I haven't started playing with the WiThrottle Server yet to trace for myself what it is doing that may cause an error (beyond the missing steps shown in the test of #6150 ).  I'll look into that tomorrow.

icklesteve, 2018-11-17:
For info, in the next few months I'd like to introduce the same functionality for JMRI MERG CBUS throttles as the MERG CBUS CAB handsets have.

When a throttle request to command station fails as currently in use, the user has option to steal, share or cancel the request.

( I can see the stealThrottleRequest methods in ThrottleManager + diagram in ThrottleListener however no shareThrottleRequest methods? )

With the original question posed by @pabender in mind, 
A cancel request suggests single throttle request cancel ( while still in dialogue box? ), no other throttles registered to this session affected.
A failed request indicates a failure in hardware / protocol ( after a dialogue box selection? ), other throttle (s) registered to the session may be affected if their session had already been cancelled.

Steve.

pabender, 2018-11-17:
Sent from my iPad

> On Nov 17, 2018, at 8:17 AM, steve young <notifications@github.com> wrote:
> For info, in the next few months I'd like to introduce the same functionality for JMRI MERG CBUS throttles as the MERG CBUS CAB handsets have.
> 
> When a throttle request to command station fails as currently in use, the user has option to steal, share or cancel the request.
> 
> ( I can see the stealThrottleRequest methods in ThrottleManager + diagram in ThrottleListener however no shareThrottleRequest methods? )
> 
There hasn’t ever been a need for the share.  Only LocoNet currently implements the steal.

XpressNet currently implements a share, but it follows the practice of Lenz handheld throttles and does this automatically.
> With the original question posed by @pabender in mind,
> A cancel request suggests single throttle request cancel ( while still in dialogue box? ), no other throttles registered to this session affected.
> 
The cancelThrottleRequest isn’t something specific to the steal process ( it was there long before the steal was added ).  The original idea was that if a user asked for an addresses and then changed his mind before the request was fulfilled, the request could be canceled.
> A failed request indicates a failure in hardware / protocol ( after a dialogue box selection? ), other throttle (s) registered to the session may be affected if their session had already been cancelled.
> 
Usually, it indicates something that can’t be recovered from.  An example is the SPROG Throttle Manager ( for when the SPROG is used as a stand alone programmer/decode tester ), which can only support 1 throttle at a time.  A request for a second throttle isn’t valid, so a failedThrottleRequest is issued.

bunnshill, 2018-11-17:
Many thanks to you all for looking into this problem.  At 2 recent train shows my JMRI crashed with errors due to the steal problem I think.  (Raspberry Pi running 4.12 using both Apple and Android phones with a Digitrax DSC100).

It seems to me the problem is not just about stealing.  I have experimented today and I realise there is also a problem also with dispatching.

So I started with the DCS100 with its memory reset and a MacBookPro using JMRI 4.12 and Locobuffer. I ran Withrottle on my iPhone.  The JMRI slot monitor showed no slots selected. I selected loco 2201 on the iPhone app.  All was well and it showed up in the loconet monitor and the slot monitor.  I then released it on the iPhone and it disappeared from the WiThrottle server and the slot monitor. It can then be reselected on the iPhone and all is well.

But if I dispatch it from the iPhone it disappears from the active slot monitor and the WiThrottle server.  I cannot then reacquire the loco through the WiThrottle app.It appears to work on the app  - the loconet monitor shows a response to the throttle changing but the loco does not respond. I can acquire it and release it using a Digitrax throttle.   But whatever I do with the WiThrottle app shows up in loconet monitor and the WiThrottle server but not in the slot monitor and the loco does not move.  This is not stealing.  I cannot then use this loco on the Withrottle app again until  JMRI is rebooted.

The system console does not show an error, unless the following is a clue:

WARN - current stored duration is not a valid number "null" [thread 20]

I hope this helpful.

David Matthews

bunnshill, 2018-11-17:
I have repeated this with the same results with 4.13.4.

The loconet slot monitor shows the loco dispatched by the Withrottle app in the unused slots of the slot monitor as idle but the speed can vary and the loconet monitor reflects this.   The loco does not move.

David Matthews

pabender, 2018-11-17:
@bunnshill I opened a new issue for that, #6155

devel-bobm, 2018-11-17:
@pabender wrote:

> Having the warning there is confusing because we can steal slots, we just don't implement the ZapSteal (I think that is right term) of the newer Digitrax command stations.

"StealZap" is a feature which only comes into play when a throttle uses LocoNet "expanded slot" messaging and some other throttle "steals" its loco (i.e. slot).  At the moment, JMRI's LocoNet support does _not_ use "expanded slot" messaging, so this is not (yet) an issue for JMRI-managed throttles.

As a "thief" stealing a slot, and working with "standard slot" messaging (as currently done), JMRI's LocoNet code need not do anything other than "steal the slot".  The command station takes care of the rest, regardless of the type of messaging any other throttle is using to control that slot.  

- If there is a throttle using the slot with "expanded slot" messaging, then the command station will send it a message that approximately says "your slot was stolen" and the throttle gives the user a "StealZap" message.  

- If the throttle is being used only by throttles using "standard slot" messaging, nothing will happen and all throttles will retain influence over the slot.  (And throttle control becomes chaotic as each throttle periodically tries to exert influence, only to be overridden by another throttle's influence...)

As a "victim" of slot stealing, current JMRI code need do nothing, regardless of the messaging type used by the "thief" throttle.  JMRI does not get a message saying the slot was stolen.  JMRI will continue to provide periodic updates to the slot.  

- If the slot was stolen by a throttle using "standard slot" messaging, JMRI's periodic updates will be accepted by the command station and will cause throttle control chaos as any throttle sending "standard slot" messaging will be accepted and sent to the loco.

- If the slot was stolen by a throttle using "expanded slot" messaging, JMRI's periodic updates will "fall on deaf ears" and be a waste of available LocoNet bandwidth.  Only the throttle sending "expanded slot" messaging will be able to influence the loco.

**Once JMRI supports JMRI and WiThrottleServer throttles using "expanded slot" messaging**, JMRI's LocoNet support will need to be able to "zap" a JMRI throttle or WiThrottleServer-managed throttle **if and only if JMRI is using "expanded slot" messaging to control the throttle**.

If the LocoNet "You are the victim of a stot steal" "expanded slot" message comes along for a slot currently used by a JMRI-managed throttle or WiThrottleServer-managed throttle, then that throttle will be unable to control the slot.  The only sensible thing to do is to let the throttle user know that the throttle is no longer controlling the loco...

In other words, **JMRI support for "StealZap" is an exercise for inclusion in future "expanded slot" support efforts.**

pabender, 2018-11-17:
PR #6154 fixes the Loconet Aquisition issue, and drops the warning message.

PR #6150 should now take care of the sequence difference between steal on LNWI and JMRI noted in #4722.

I am going to go ahead and merge #6154 so I can fully test #6150 (which is on a different branch) with real hardware.

In addition to my open question about whether we should use cancel or fail after a steal is refused, 
there is one minor issue that may need to be dealt with still.  If a WiThrottle/EngineDriver user does either of the following things:
1) Never chooses whether to steal or not
2) Declines the steal and then leaves the app open without selecting a new address (i.e. never sends a Q to disconnect).

The steal process won't be completed, which potentially leaves the LnThrottleManager waiting indefinitely for the steal response.  The only way I see to fix that is with a timer, which needs to be started when the WiThrottle Server is notified of the steal.

BHoffman351, 2018-11-17:
I would appreciate an opportunity to test this with the existing WiThrottle Server code prior to #6150 being merged.

Thanks,
Brett

> On Nov 17, 2018, at 4:07 PM, Paul Bender <notifications@github.com> wrote:
> 
> PR #6154 <https://github.com/JMRI/JMRI/pull/6154> fixes the Loconet Aquisition issue, and drops the warning message.

pabender, 2018-11-17:
@BHoffman351  yes, please do test first.

pabender, 2018-11-17:
My testing with real hardware shows there is still an issue with #6150.  

The code now passes the test on stealing within the MultiThrottleController class, but it appears that the MultiThrottle class creates a new MultiThrottleController for the second address request from the device, so the state saved in the first MultiThrottleController is lost when the request is made....

BHoffman351, 2018-11-18:
@pabender <https://github.com/pabender> wrote:
> yes, please do test first.

I found that the corrections to the cancel of a steal request (#6150 <https://github.com/JMRI/JMRI/pull/6150>) had the desired effect of cleanly removing the request for an address that needs steal.  It also made it so that I was able continue my work and add a boolean flag (similar to what is in #6154 <https://github.com/JMRI/JMRI/pull/6154>) that allows the steal request to be completed when confirmed in WiThrottle and Engine Driver.  All tested on real hardware.  It also cleanly handles:
> Never chooses whether to steal or not
> Declines the steal and then leaves the app open without selecting a new address (i.e. never sends a Q to disconnect).
It does this without the need for a timer or other wizardry.

This is available in the branch:
BH_Steal <https://github.com/BHoffman351/JMRI/tree/BH_Steal>

It currently does not handle the choice of a separate lead loco for an Advanced Consist, but I can add that in as well.

Regards,
Brett

pabender, 2018-11-18:
@BHoffman351 please go ahead and add a pull request for that, so we can merg it in.

Also, I just want to verify I am reading what your message states correctly:

Are your changes are in addition to my changes to ThrottleController and MultiThrottleController changes in #6150 
Or
Do your changes replace my changes to ThrottleController and MultiThrottleController?  

I just want to make sure the right changes are put in place when All is said and done.

( the other changes in #6150 are tests for steal in the server code... )

BHoffman351, 2018-11-18:
> On Nov 18, 2018, at 1:41 PM, Paul Bender <notifications@github.com> wrote:
> please go ahead and add a pull request for that, so we can merg it in.
> 
I will mark as WIP, and I’ll take a look into consist lead address.
> Also, I just want to verify I am reading what your message states correctly:
> 
My apologies, you read it correctly but I wrote it wrong.  Where it said “Merge branch ‘master' into…” I read as “Merged ... into master”.
> Are your changes are in addition to my changes to ThrottleController and MultiThrottleController changes in #6150 <https://github.com/JMRI/JMRI/pull/6150>
> Or
> Do your changes replace my changes to ThrottleController and MultiThrottleController?
> 
My changes were in addition to #6154 <https://github.com/JMRI/JMRI/pull/6154>, I did not use what was in #6150 <https://github.com/JMRI/JMRI/pull/6150>.
> 
> I just want to make sure the right changes are put in place when All is said and done.
> 

Also, I have not run any automated tests against it.

Regards,
Brett

pabender, 2018-11-24:
Close via #6150 and #6154

closed 2018-11-24 by pabender