JMRI/JMRI #6152: RaspberryPi: "ant package-windows" NSIS file InstallJMRI.nsi plugin path error.

ClockLoop, 2018-11-17:
If you build the windows package on a raspberrypi with latest NSIS version 2.51-1.
It will fail due to 2 issues.
Both are located in the InstallJMRI.nsi file.
1. the path to nsis plugin folder is actually /Plugins   not /plugin   
2. the way it presents the files with "quotes" makes it not find the files also.
Build Fail:
ReserveFile /plugin "System.dll"
ReserveFile /plugin "NSISdl.dll"
ReserveFile /plugin "UserInfo.dll"

FIX: Edit InstallJMRI.nsi  and replace above with below.
Build working:
ReserveFile /usr/share/nsis/Plugins/System.dll
ReserveFile /usr/share/nsis/Plugins/NSISdl.dll
ReserveFile /usr/share/nsis/Plugins/UserInfo.dll

I tried to just put the correct /Plugins, but it still fails to find the file.  the FULL path is the issue.
I also tried to copy the files to the location it wanted.... the quote is the issue.
Nothing worked until I gave it the direct paths to the files. I don't know enough about linux or nsis or reservefile command to do anything else.

Build Fail trimmed error log.
> $ ant package-windows
> Buildfile: /home/pi/JMRI-s/JMRI/build.xml
> package-windows:
>     [mkdir] Created dir: /home/pi/JMRI-s/JMRI/dist/Windows/JMRI
>      [copy] Copying 10789 files to /home/pi/JMRI-s/JMRI/dist/Windows/JMRI
>      [copy] Copying 110 files to /home/pi/JMRI-s/JMRI/dist/Windows/JMRI
>     [mkdir] Created dir: /home/pi/JMRI-s/JMRI/dist/Windows/JMRI/lib
>      [copy] Copying 68 files to /home/pi/JMRI-s/JMRI/dist/Windows/JMRI/lib
>     [mkdir] Created dir: /home/pi/JMRI-s/JMRI/dist/Windows/JMRI/lib/windows
>      [copy] Copying 17 files to /home/pi/JMRI-s/JMRI/dist/Windows/JMRI/lib/windows
>     [mkdir] Created dir: /home/pi/JMRI-s/JMRI/dist/release
>      [exec] ----------------------------------------------------------------------
>      [exec] NSIS String Functions Header File 1.09 - Copyright 2004 Diego Pedroso
>      [exec] ----------------------------------------------------------------------
>      [exec]  (/usr/share/nsis/Include/StrFunc.nsh:52)
>      [exec] $ {StrStr} - Copyright 2004 Diego Pedroso - Based on functions by Ximon Eighteen (macro:STRFUNC_FUNC:11)
>      [exec] $ {UnStrStr} - Copyright 2004 Diego Pedroso - Based on functions by Ximon Eighteen (macro:STRFUNC_FUNC:5)
>      [exec] ReserveFile: "/usr/share/nsis/plugin" -> no files found.
>      [exec] Usage: ReserveFile [/nonfatal] [/r] [/x filespec [...]] file [file...]
>      [exec] Error in script "InstallJMRI.nsi" on line 448 -- aborting creation process
> 
> BUILD FAILED
> /home/pi/JMRI-s/JMRI/build.xml:2224: exec returned: 1

devel-bobm, 2018-11-17:
Part of your problem is typically handled via the "local.properties" file in the "root" of your local JMRI repository.  That is the file which allows you to "localize" the paths to your installs of the "helper tools" JMRI uses.  See the comments in build.xml (approx. lines 30-55) which give a little bit of description.

For example, you would prsumably need to modify your local.properties file to include the line:

nsis.home=/user/share/nsis/

But this will not resolve the issue you have with "plugin" versus "Plugins".  Apparantly the NSIS project has changed their directory naming in newer versions than the one(s) JMRI is expecting.

Per the comment (near line 20) in LaunchJMRI.nsi, you **must** use the "Large strings" build of NSIS.  If there is a "Large strings" build for this newer version, then perhaps there should be consideration of moving to the newer version.  (Or simply work with a "Large strings" version from a build with older-style path naming.

ClockLoop, 2018-11-17:
> Per the comment (near line 20) in LaunchJMRI.nsi, you **must** use the "Large strings" build of NSIS. If there is a "Large strings" build for this newer version, then perhaps there should be [...]

I actually saw that, and found a link to an experimental version of nsis from one of the old issues here, but the link was broken, and I tried to search using keywords related, nothing...  while searching, i manually edited the paths, and solved it before finding any large strings version.
I wasn't sure if this was the nsis install, or the script.

bobjacobsen, 2018-11-17:
On a Mac, using "brew install makensis --with-large-strings” to install and build NSIS 3.03 with the large-string option works with the existing files.

klk32003, 2018-11-17:
@ClockLoop See [this commit](https://github.com/JMRI/JMRI/commit/31532e37e05a2c2d8e3e596ba22eb6ce0bbabbe1#diff-735059f12e8148114622f09cfca2fea4). Reverting it might help.

bobjacobsen, 2018-11-17:
I think all of the problems here are caused by trying to use a NSIS 2 processor instead of a NSIS 3 one. JMRI requires NSIS 3 and has for a while.

https://github.com/JMRI/JMRI/blob/master/scripts/HOWTO-distribution.md

ClockLoop, 2018-11-17:
> I think all of the problems here are caused by trying to use a NSIS 2 processor instead of a NSIS 3 one. JMRI requires NSIS 3 and has for a while.
> 
> https://github.com/JMRI/JMRI/blob/master/scripts/HOWTO-distribution.md

I see, ....
Ok, so the issue is raspbian's latest build of nsis is only 2.51-1.
Thanks, I just won't compile for any target but the one im running on...(linux)

closed 2018-11-17 by ClockLoop