JMRI/JMRI #6158: Clock Information Save Reminder gives misleading advice

davidzuhn, 2018-11-17:
This is JMRI 4.12 running on a Raspberry Pi on Raspbian 9 via PanelPro.

If I make any changes on the Fast Clock Setup window, when I close that window I'm prompted to "Remember to save your Clocks information in your configuration.   (choose Save > Save Configuration... from the File menu)"

However, I don't know where that File menu is supposed to be.   It's not the File menu in the main Panel Pro window (which only has "Print decoder definitions", "Print Preview decoder definitions" and "Quit" as options).    

It's not on the Fast Clock Setup windows, which doesn't have a File menu, and doesn't have a "Save Configuration..." under either of the two menus available [Window, Help].   

I don't see as "Save Configuration..." menu item anywhere.

silverailscolo, 2018-11-18:
- in the main DecoderPro menu bar select Panels > Save Panels...
- with a Panel + Editor open, choose File menu > Save Panels...
- from any open table, select File menu > Store > Store Configuration and Panels To File...

Agreed that instead of "Save..." the dialog should say "Store.." with some more context.

davidzuhn, 2018-11-21:
None of those three options match the text in the dialog box (which should either have a correct path or a general message).  Giving a specific path that doesn't exist is the worst case.  Perhaps the dialog box should have a "Save now" button instead of just a reminder.   Hiding the clock config behind panels would be confusing -- I'm not use any panels and so I wouldn't expect the clock config message to require panels.

silverailscolo, 2018-11-22:
I'm preparing to correct the reference to the menu item and add a File menu to the frame.

silverailscolo, 2018-11-22:
PR #6188 fixes this, plus some more refinements (Store menu + dialog that lets you choose either Store... method, or Cancel).

PR #8419 (2020-04-24): Fast Clock Setup Window + Clock Javadoc
Update for Fast Clock Setup Frame layout, close window handling.

Moves Load / Store Strings to higher Bundle level.
Javadoc for various Clock related classes.

Fixes #6158
Fixes #6259

[image]
closed 2020-04-24 by bobjacobsen