JMRI/JMRI #6159: DCC++ Current sensing broken

tman0, 2018-11-17:
Latest JMRI v4.13.4 testing release, originally discovered in 4.12 release
Latest official DCC++ Git release
Platform: Linux 4.9.59-v7+ on Raspberry Pi

Looks like the parsing of DCC++ current sensing responses is broken.
DCC++ traffic log:
```
[TX: c]   Read Track Current Cmd 
[RX: a0]   Current: 0 / 1024
[TX: 1]   Track Power ON Cmd 
[RX: p1]   Power Status: ON
[TX: c]   Read Track Current Cmd 
[RX: a14]   Current:  / 1024
[TX: c]   Read Track Current Cmd 
[RX: a16]   Current:  / 1024
[TX: 0]   Track Power OFF Cmd 
[RX: p0]   Power Status: OFF
[TX: c]   Read Track Current Cmd 
[RX: a0]   Current: 0 / 1024
```

When the response isn't 0, JMRI throws this error:
```
2018-11-17 23:20:40,392 dccpp.DCCppReply                      ERROR - DCCppReply value index too big. idx = 2 msg = a16 [AWT-EventQueue-0]
2018-11-17 23:20:40,396 dccpp.DCCppReply                      ERROR - DCCppReply value index too big. idx = 2 msg = a16 [AWT-EventQueue-0]
2018-11-17 23:20:40,398 jmrix.AbstractMRTrafficController     ERROR - Unexpected exception in invokeAndWait: {}java.lang.reflect.InvocationTargetException [serial.SerialDCCppPacketizer Receive thread]
java.lang.reflect.InvocationTargetException
	at java.awt.EventQueue.invokeAndWait(EventQueue.java:1319)
	at java.awt.EventQueue.invokeAndWait(EventQueue.java:1294)
	at javax.swing.SwingUtilities.invokeAndWait(SwingUtilities.java:1348)
	at jmri.jmrix.AbstractMRTrafficController.distributeReply(AbstractMRTrafficController.java:1013)
	at jmri.jmrix.AbstractMRTrafficController.handleOneIncomingReply(AbstractMRTrafficController.java:1056)
	at jmri.jmrix.AbstractMRTrafficController.receiveLoop(AbstractMRTrafficController.java:868)
	at jmri.jmrix.AbstractMRTrafficController$2.run(AbstractMRTrafficController.java:792)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.NumberFormatException: For input string: ""
	at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)
	at java.lang.Integer.parseInt(Integer.java:592)
	at java.lang.Integer.parseInt(Integer.java:615)
	at jmri.jmrix.dccpp.DCCppReply.getCurrentInt(DCCppReply.java:768)
	at jmri.jmrix.dccpp.DCCppMultiMeter.message(DCCppMultiMeter.java:39)
	at jmri.jmrix.dccpp.DCCppTrafficController.forwardReply(DCCppTrafficController.java:85)
	at jmri.jmrix.AbstractMRTrafficController.notifyReply(AbstractMRTrafficController.java:295)
	at jmri.jmrix.AbstractMRTrafficController$RcvNotifier.run(AbstractMRTrafficController.java:1223)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:301)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

tman0, 2018-12-20:
Working on it now. Looks like `DCCppReply.isNamedCurrentReply()` is returning true no matter whether it's named or not. Both `a14` and `a MAIN 14` match the named current reply regex `DCCppConstants.CURRENT_REPLY_NAMED_REGEX`.

closed 2018-12-22 by bobjacobsen

positron96, 2019-07-21:
Sorry to dig it up, but what pieces of software produce named current reading? Also, where do the specs for DCC++ come from? I thought this
site https://github.com/DccPlusPlus/BaseStation/wiki/Commands-for-DCCpp-BaseStation is the official protocol documentation, but there is no current sensing there.

Pugwash1, 2019-07-21:
@positron96  Unsure where its documented but in SerialCommand.cpp its send <c> , reply <aX>
where X is between 0 and 1024 and it represents X/1024 of total current.

positron96, 2019-07-29:
@Pugwash1 thanks! What about named current reading though?

PR #6339 (2018-12-22): Fix named current reply regex being too broad
The `DCCppConstants.CURRENT_REPLY_NAMED_REGEX` was too broad, so it captured both named and unnamed current replies, making `DCCppReply.isNamedCurrentReply()` always return true for any current reply. Also added tests to `DCCppReplyTest` to verify this in the future. Fixes #6159.