JMRI/JMRI #6178: Create Test Release 4.13.6

bobjacobsen, 2018-11-21:
This is the next release in the 4.14 cycle. It's intended to be released in early from HEAD of master as soon as some WIP issues are resolved.

bobjacobsen, 2018-12-05:
The release-4.13.6 branch has been created. 

From now on, please document your changes in the [jmri4.15.1.shtml](https://github.com/JMRI/website/blob/master/releasenotes/jmri4.15.1.shtml) release note file.

Maintainers, please set the 4.15.1 milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

Jenkins will be creating files shortly at the [CI server](http://builds.jmri.org/jenkins/job/TestReleases/job/4.13.6/)

If you're developing any additional (post-4.13.6) changes that you want in the JMRI 4.14 production release, please start from this branch, i.e. do `git checkout -b release-4.13.6` to start your work.

Pugwash1, 2018-12-05:
> ---------- Original Message ----------

Digitrax pr3/dcs51
Roster->identify->No Service mode programmer available.
Occurs on this
HEAD is now at 1314f333a8 Merge pull request #6161 from
bobjacobsen/manual-internal-11
Defaults all good.
Steve G.

bobjacobsen, 2018-12-05:
First JMRI 4.13.6 files are available in the usual way at:

http://builds.jmri.org/jenkins/job/TestReleases/job/4.13.6

Feedback appreciated. Checking that various connections actually work (i.e. that internals changes haven’t broken anything) particularly appreciated.

bobjacobsen, 2018-12-05:
@Pugwash1 Thanks!  Looking into it.  Won't be at hardware until this evening.

Pugwash1, 2018-12-08:
@bobjacobsen 
Regarding the PR3. The cause is that it is being asked what it can provide before the adapter has been configured, and it defaults to PR3 mode , which I think is stand alone pr3, which has no programmers.
If it defaults to MS100 mode it would say it has programmers, but then tries to use them when in standalone mode. One way forward would be to make the choice of standalone pr3/locobuffer or pr3/locobuffer at a higher level.

closed 2018-12-17 by bobjacobsen