JMRI/JMRI #6197: Engine Driver and JMRI 4.13.5 fail when attempting to steal an engine

frankfezzie, 2018-11-26:
Using Engine Driver connected through JMRI ver 4.13.5, when I attempt to acquire an engine that is "In Use" on a Digitrax throttle I see a "steal" message on my phone with Y or N options.  

When I choose the "Y" option to the steal message, Engine Driver will not acquire the engine and any further attempt to acquire any engine with Engine Drive fails.  After 5 to 10 seconds Engine Driver reports that it lost contact with the server and reestablishes contact.  At this point Engine Driver resumes normal operation and will acquire any engine except one that is "In Use" on a Digitrax throttle.

When I choose the "N" option to the steal message, Engine Driver returns to the throttle screen showing no engine is selected and will allow me to acquire any engine except one that is "In Use" on a Digitrax throttle.

I also tested the JMRI throttle on the computer screen.  The JMRI computer throttle appears to be mostly well behaved regarding steals IF one always uses the RELEASE button to free a selected engine from the throttle.  Both "Y" and "N" options to the steal message appear to function properly.

However, if you use the DISPATCH button to free a selected engine from the computer throttle, then operation of the computer throttle gets weird after a steal message.  Here is one test sequence.
8840 is In Use on Digitrax throttle.  On the computer throttle do these steps;
1. attempt to acquire 8840.  Get steal message.  Answer "Y".  Confirm control of engine.  Slot monitor shows slot In Use. Hit DISPATCH to free engine.  Slot monitor shows status as IDLE.
2. acquire 8841.  Confirm control of engine.  Slot monitor shows slot in Use.  Hit DISPATCH to free engine.  Slot monitor shows status as IDLE.
3. acquire 8840.  You will not get steal message which is correct.  Computer throttle appears to show that engine is acquired but you will have no control and slot monitor will show status as IDLE.
Testing will show that you can properly acquire any engine number ONE time and then use DISPATCH button to free it.  But second attempt to acquire the number will fail as described above.

Once the computer throttle starting failing to properly acquire engines, then Engine Driver will exhibit the same failures.  Only way to fix it is to close and then restart JMRI.

Setup
Digitrax DCS100 command station
Digitrax DT402D throttle running rev 21 software and operating plugged into the Loconet
HP Laptop running Windows 10
JMRI version Test 4.13.5
Locobuffer USB interface
Engine Driver 2.22.69 running on Samsung Galaxy S2 with Android version 4.1.2
I placed two engines on the track, 8840 and 8841, so that I could observe results of throttle commands on the engines as I attempted to acquire the engines in various sequences.  8840 was acquired on the Digitrax throttle.

Frank Fezzie

pabender, 2018-11-26:
The dispatch issue is related to #6155, so that part of this issue is a duplicate.

Use the release button as a workaround.

The engine driver issue should be reported as an issue on the engine driver issue tracker ( 
https://github.com/JMRI/EngineDriver/issues ) as I believe WiThrottle is working properly now.

BHoffman351, 2018-11-26:
> On Nov 25, 2018, at 9:20 PM, frankfezzie <notifications@github.com> wrote:
> Using Engine Driver connected through JMRI ver 4.13.5, when I attempt to acquire an engine that is "In Use" on a Digitrax throttle I see a "steal" message on my phone with Y or N options.
> 
> When I choose the "Y" option to the steal message, Engine Driver will not acquire the engine and any further attempt to acquire any engine with Engine Drive fails.
> 
This was the issue corrected in #6168 <https://github.com/JMRI/JMRI/pull/6168> but it was not merged in prior to version 4.13.5 being released.

Regards,
Brett

pabender, 2018-11-26:
@frankfezzie it is possible that fixes for both of the issues you see have been made, but did not make 4.13.5.  Could you please try one of the daily builds from:

http://jmri.tagadab.com/jenkins/job/Development/job/Packages/

And let us know if the problems are solved.

frankfezzie, 2018-11-26:
Gentlemen,
Per your request I downloaded 4.13.6ish and repeated my tests.  A lot of the problems have been fixed!  A few remain, possibly related to problem #6155.

Fixed - Using a Digitrax DT402D throttle and Engine Driver I tested various combinations of having an engine "in Use" on one throttle and attempting to steal it with the other throttle.  I tested both "Y" and "N" answers to Steal message.   Releasing or Dispatching a shared engine from either throttle appears to work properly.  Shared engines that were Released or Dispatched could be reacquired by either throttle.   It all seemed to work as it should.

I then tested using a JMRI computer throttle and a Digitrax DT402D throttle.  Two sequences cause the JMRI computer throttle to fail with the same result in both cases.
1. If an engine that is IN USE on the JMRI computer throttle is Dispatched using the DISPATCH button then JMRI will no longer properly acquire that engine.  It appears to be acquired on the screen but the Slot Status is not changed to IN USE when the SET button is pressed.  Any other engine can still be properly acquired by JMRI.
2. If an engine is being shared with a JMRI computer throttle and a Digitrax DT402D throttle and the engine is either Released or Dispatched from the Digitrax throttle FIRST then JMRI will no longer properly acquire that engine.  It appears to be acquired on the screen but the Slot Status is not changed to IN USE when the SET button is pressed.  Any other engine can still be properly acquired by JMRI.

Once the JMRI throttle has failed in this manner and will no longer properly acquire a specific engine, then attempting to acquire the engine with Engine Driver has the same result.  It appears to be acquired on Engine Driver but the slot status is not changed to IN USE.  The only way I could get JMRI to properly acquire the specific engine was to shutdown and restart JMRI.

Frank Fezzie

frankfezzie, 2018-11-27:
Paul Bender,

I tested 4.13.6ish per your request and posted my results to github.  Hope you saw them.  Let me know if you did not.

Frank Fezzie

From: Paul Bender 
Sent: Monday, November 26, 2018 6:58 AM
To: JMRI/JMRI 
Cc: frankfezzie ; Mention 
Subject: Re: [JMRI/JMRI] Engine Driver and JMRI 4.13.5 fail when attempting to steal an engine (#6197)

@frankfezzie it is possible that fixes for both of the issues you see have been made, but did not make 4.13.5. Could you please try one of the daily builds from:

http://jmri.tagadab.com/jenkins/job/Development/job/Packages/

And let us know if the problems are solved.

pabender, 2018-11-27:
@frankfezzie yes, I saw your message.

@Pugwash1 It appears there is still something wrong with dispatch on LocoNet, even after #6166 has been merged.  See https://github.com/JMRI/JMRI/issues/6197#issuecomment-441741761 above.

Pugwash1, 2018-11-27:
@pabender Looking at the problem now.

Pugwash1, 2018-11-27:
@pabender I can correct it so that a throttle dispatched from jmri can be aquired by a Digitrax throttle or a JMRI throttle in PanelPro, but there are issues if that throttle is already in use by a WiiFi or Web throttle. When the throttle is dispatched, the slot status changes in the command station to common so it disappears off the loconet slotmon. Depending on the command station the Wifi or web throttle will still be able to control the loco.

pabender, 2018-11-27:
@Pugwash1 you might need to do reference counting to make sure we only dispatch when there is a single JMRI user of the slot.

It might also be possible to use something like a vetoable change listener to stop the dispatch process if there is a listener using the throttle (see https://docs.oracle.com/javase/7/docs/api/java/beans/VetoableChangeListener.html ). 

This doesn't help with throttles that acquire the throttle and never register a property change listener, but those that do could choose to stop the dispatch.  it would probably be a good idea to add that to release as well.

Pugwash1, 2018-11-28:
@pabender AbstractThrottle already is a card counter. If I make dispatchthrottle return a value I can do this. The release only release's if its the last throttle. Just doesnt tell anyone. The dispatch is a bit final so has to tell...

Pugwash1, 2018-11-30:
@pabender I will have something by the end of day (EST), just working on the tests.

bunnshill, 2018-12-01:
I have also tested JMRI 4.13.6ish using a DCS 100 and DT400 and UT4D throttles with a Mac Pro and WiThrottle on an iPhone.

The handling of stealing seems to be fixed.  Many thanks.

My observations are the same as Frank Fezzie above, with my comment under his 1) above that the dispatch problems generated are when I use dispatch on the iPhone (not on the Digitrax throttles). I get the same results as his 2) using an iPhone.

David Matthews

Pugwash1, 2018-12-01:
@lavender finally found the problem with the tests. Some are referencing objects that have disposed off. Hopefully done by sunday

Pugwash1, 2018-12-07:
@frankfezzie  @pabender  @bunnshill 
Has anyone tried it in the official test release yet (4-13-6)(no ish)

frankfezzie, 2018-12-07:
I was unaware of official test release 4.13.6 and so I did not test it.
I have downloaded 4.13.7ish and started testing it.  I have found one glitch so far but I need more time.  I also now have an iPad and will be able to test both Engine Driver and WiThrottle with JMRI.
I should have complete report to you by Sunday.
Frank Fezzie

frankfezzie, 2018-12-10:
I have been testing release 4.13.17ish with Engine Driver and WiThrottle Lite.  In general the steal function seems to work well across all combinations of throttles.

One problem still exists, and I am still chasing one other problem.

If an engine is being shared with a JMRI computer throttle and a Digitrax DT402D throttle and the engine is either Released or Dispatched from the Digitrax throttle FIRST then JMRI will no longer properly acquire that engine.  It appears to be acquired on the screen but the Slot Status is not changed to IN USE when the SET button is pressed.  Any other engine can still be properly acquired by JMRI.  Once the JMRI throttle has failed in this manner and will no longer properly acquire a specific engine, then attempting to acquire the engine with Engine Driver or Withrottle Lite has the same result. It appears to be acquired on Engine Driver or WiThrottle Lite but the slot status is not changed to IN USE.  This problem does not occur if the engine is being shared between a Digitrax throttle and either Engine Driver or WiThrottle. 

The only way I could get JMRI to properly acquire the specific engine was to shutdown and restart JMRI.

One hint about the source of the problem.  Once the failure has occurred I would acquire the engine on the JMRI computer throttle.  As reported above the slot is not changed to In Use and I would have no control.  If I hit the DISPATCH button on the JMRI computer throttle then JMRI would indicate that it could not dispatch the engine because the is one other user.  Neither Engine Driver nor WiThrottle Lite was running at this time.

Regarding the problem I am chasing, when throttles are sharing control of an engine, any change made on one throttle such as speed, direction, or function button appears on all throttles sharing the engine.  I get into an occasional failure mode where Engine Driver fails to update the speed indication when another throttles changes the speed.  Direction and functions changes still appear on Engine Driver.  I am still searching for a simple and repeatable sequence to trigger the failure.  Since it only appears in Engine Driver, the problem might lie there rather than in JMRI.  I will keep you advised.

A second note.  When Digitrax DT throttles and Engine Driver and / or WiThrottle are sharing control of an engine, should any throttle dispatch or release the engine then the slot status changes to IDLE or COMMON and no further commands are passed to the engine.  However there is no visible indication on any of the other throttles that the engine has been released.  To all appearances it still looks like the throttles have control of the engine.  To actually regain control you must release the engine from the throttle and then reacquire it.  The JMRI Computer Throttle is an exception to this.  As noted in the failure description above, when any Digitrax throttle releases the engine, the JMRI throttle changes to show that it does not have control of the engine.  This is inconsistent with what the other throttles do.

I see two options.  The first is to change JMRI so that the JMRI computer throttle does not change if another throttle releases the engine.   This would make all throttle types behave the same way.

The second option would be to make Engine Driver and WiThrottle also change display to show the engine is released just like the JMRI Computer Throttle does.  Then all JMRI throttles would be consistent although different from Digitrax throttles.

I have a slight preference for the first option because it is consistent across all throttle types. 

Setup
 Digitrax DCS100 command station
 Digitrax DT402D throttle running rev 21 software and operating plugged into the Loconet
 HP Laptop running Windows 10
 JMRI version Test 4.13.7ish
 Locobuffer USB interface
 Engine Driver 2.22.69 running on Samsung Galaxy S2 with Android version 4.1.2
 WiThrottle Lite ver 2.20 running on an iPad.
 I placed two engines on the track, 8840 and 8841, so that I could observe results of throttle commands on the engines as I attempted to acquire the engines in various sequences. 8840 was acquired on the Digitrax throttle.

Frank Fezzie

Pugwash1, 2018-12-10:
@frankfezzie You are correct. The behavior you note also occurs when a dt402 (XP slot mode off) aquires a loco, a ut4 steals it and the dt402 releases it, but its easier to recover from it. Whether you can still control to loco in jmri varies by command station, with a dcs51 you can still control the loco, with others  you cant.

I think I can at least get it to reacquire the JMRI throttle, but web & wifi might be a problem Throttles also dont respect the "release" button on the slot monitor.

frankfezzie, 2018-12-10:
I have identified a sequence of steps that cause WiThrottle Lite (and sometimes Engine Driver) to fail to display speed changes entered by other throttles sharing the same engine (and slot).

Normally, every throttle sharing an engine will display any speed, direction, or function button change issued by any throttle sharing that engine.  During testing of issues related to Steal, Release, and Dispatch I have noted instances where either Engine Driver or WiThrottle Lite or both would stop displaying speed changes issued by either JMRI computer throttle or Digitrax throttle sharing the same engine.  When this occurs Engine Driver and WiThrottle Lite could still control the speed of the engine and they continued to display any changes to direction or function buttons.  When the failure occurs, normal operation of WiThrottle Lite or Engine Driver can be obtained by releasing the engine and then reacquiring it.  Below is a very specific sequence that causes this failure on WiThrottle Lite every time on my system.  

I have observed other instances where this failure happens but I have not been able to define reliable sequences of steps to make it reoccur.  I have also seen Engine Driver and WiThrottle Lite spontaneously recover and resume displaying speed changes made by other throttles.

I perform these steps in sequence shown and do not press any buttons not called for in the sequence.

Digitrax system is running with a DT402D plugged into the Loconet.  Jmri is running with WiThrottle server running and a JMRI throttle on screen.  Engine Driver is running on a phone and WiThrottle Lite is running on an iPad.  No Engine is acquired on any throttle.  I tested with engine 8840.  Do not enter any command to the engine such as speed or function until told to do so.

1. Acquire 8840 on Digitrax DT402D.
2. Acquire 8840 on JMRI Computer throttle.  You will see a "Steal" message.  Select Yes.
3. Acquire 8840 on Engine Driver
4. Acquire 8840 on WiThrottle Lite
5. On JMRI computer throttle advance the speed slightly.  Note the speed change is displayed on all throttles.  Return speed to zero on JMRI computer screen and note that all throttles now show zero.
6. Press Release button on JMRI Computer Throttle.
7. Increase speed slightly on WiTHrottle and note speed change is displayed on Engine Driver and DT402D.
8. Return speed to zero on WiThrottle Lite and note that other throttles now show zero speed.
9. Increase speed slightly on Engine Driver.  The failure should occur and Withrottle will still show zero speed while DT402D does show speed change.  Return speed to zero on Engine Driver.
10.  Increase speed on WiThrottle.  DT402D will show speed change.  Engine Driver may or may not show speed change.
11.  All throttles continue to show direction and function button changes made on any throttle.

NOTE - this failure seems to be extremely sensitive to keystroke sequences and is almost random sometimes.  I can enter a sequence of steps and get a failure, then repeat the sequence with no failure.  Sometimes Engine Driver fails, sometimes WiThrottle fails, sometimes both fail.

Frank Fezzie

mstevetodd, 2018-12-12:
@frankfezzie Note that until very recently (4.13.5 or so), speed and direction changes from "other" throttles were not forwarded to WiThrottle clients at all. See #6068 
EngineDriver has code in it to request the current speed and direction as part of the heartbeat, so the speed/dir would update every 10 seconds or so (based on "Seconds until eStop" Preference).
For debugging, it would be helpful if you could enable withrottle debugging in default.lcf and advise if the proper speed change message is being sent, but dropped on the client side, or if the server is failing to send the proper message.

frankfezzie, 2018-12-12:
I tested release 4.13.6 for the two failures that I have reported in this thread as existing in 4.13.5 and 4.13.7ish.  Both failures occur in test release 4.13.6 
I uploaded Withrottle debug information under title "Withrottle Speed Failure 4-13-6".  This is first time I have tried to upload such data.  Let me know if I did it right.

Frank Fezzie

mstevetodd, 2018-12-14:
@frankfezzie I don't see the attachment. You can drag and drop files directly into a GitHub issue.
But the file alone isn't going to be much help. What I need is for you to watch the log and see if the "lost" speed change shows up or not. If the lost speed change is in the log, the problem would seem to be in the client, if not, most likely on the server side. It might be easier to watch the JMRI System console window to "catch" the issue.

mstevetodd, 2019-11-21:
I'm going to mark this closed.

closed 2019-11-21 by mstevetodd