JMRI/JMRI #6199: What should be on the Layout Thread?

danielb987, 2018-11-26:
I'm trying to understand the [JMRI threading model](http://jmri.org/help/en/html/doc/Technical/Threads.shtml).

In PR #6177 @pabender writes that AbstractTurnout.setCommandedState() is called from other threads, for example by scripts. I thought that commands on the layout, like throwing turnouts, should always be run on the layout thread.

So my question is: When am I supposed to use ThreadingUtil.runOnLayout() and it's friends? It's clear to me what is GUI stuff, but it's not so clear to me when it's valid to do layout stuff from other threads and when I need to be on the layout thread to do things. What is safe to do in any thread and that requires to be done on the layout thread?

pabender, 2018-11-26:
The separation occurs obased in whether or not you are interacting with the GUI.

For most systems we use the AbstractMRTrafficController which means there are, at a minimum, three threads are involved:
The AWT event thread
A transmit thread
A receive thread

Generally, events triggered on the event thread cause an action in an object which queues a message for the transmit thread.  We can have more than one thread invoking calls on an object however.

It has always been allowed to update object state from the receive threads.  In fact, not updating object state  from the receive thread can cause issues when timing is involved.  E.g. XpressNet turnouts send multiple messages to the command station before the known state is actually set, and these are timing sensitive on layouts using solenoid switch machines.  It’s only the change of known state that should trigger a GUI change in most cases.

When the known state ( in the turnout case ) is set, that should be communicated to the rest of the system on the layout thread, but everything up to that point doesn’t need to be.  

Many objects use property change events to notify of state changes, and those are transmitted using the event queue.

pabender, 2018-11-27:
I missed something in my last reply.  

When the reply is distributed by the receive thread to waiting message listeners, it does so by putting those replies on the swing thread.  This is actually why updates are allowed from  the receive side of AbstractMRTrafficController based systems.

danielb987, 2018-11-27:
Thanks for your answers.

Let me refrase my question. Lets assume the following:
* The layout thread is different from the gui thread
* My code runs in a separate thread that I have created and started and that is different from the layout thread, the gui thread, and the receiving threads.

If I now wants to throw a turnout, set a sensor or set a signal, do I need to call ThreadingUtil.runOnLayout() to do so, or can I call Turnout.setCommandedState() and Turnout.setState() and so on from my thread?

(I'm aware of that the layout thread is the gui thread but I want to know how to write my code in case it isn't).

pabender, 2018-11-27:
You just call setCommandedState ( setState is supposed to be used in scripts only ).

SetCommandedState triggers newCommsndedState, which generates a property change event on the GUI thread as required.  SetCommandedState also forwards the change to the layout, either directly or through an operator .

bobjacobsen, 2018-11-27:
I think this conversation has gotten off-point.

Setting aside the difference between Layout and GUI threads for a moment, the answer is pretty simple: Code in package `X` that is known to have been invoked off the Layout/GUI thread should call everything that is defined in `jmri` on the Layout/GUI thread.  That's almost entirely true of the public interfaces below that too, i.e. `jmri.jmrit.roster`, etc.  (Some day some of those should be promoted, but that's a separate note).

To put it a bit more broadly:  A package (including its subpackages) is welcome to do any threading it wants, but it should return to Layout/GUI before invoking anything outside.

The problem is the exceptions:
- You can often get away with not doing this.  Except when you can't....  So people tend to not thing about it.
- And there are some places that don't care, i.e. lots-but-not-all of the method in the `jmri.util` package.    

Most of JMRI is invoked from the layout or from the GUI, so threading is mostly safe.  It's only when somebody actually writes threaded code that this emerges.  So it hasn't gotten as much attention as perhaps it should.

We have `@InvokeOnGuiThread` and `@InvokeOnLayoutThread` annotations, thought they're not widely used.  Maybe we should add a `@InvokeOnAnyThread` to start flagging ones that are more general (that might be the same as `@ThreadSafe`, or might not be; I haven't thought that through in detail)

On the rephrased question:

- Right now, there's no difference between layout thread and GUI thread, and we don't recommend bothering to transition between them.  
- In the case that you _know_ that your code is running on another thread, then yes, you should use `runOnLayout` to transition _any_ operation on a `Turnout` object to the correct thread.   
   - The rare exception is if you're coding up a new hardware system, which uses threads internally, and you've worked through how all those parts work together to make sure it remains consistent, there's no calling out to listeners, etc.

danielb987, 2018-11-27:
Thanks a lot. That answers my questions.

pabender, 2018-11-27:
PR #6204 adds the `@InvokeOnAnyThread` annotation.

pabender, 2018-11-27:
Also, I don't think `@InvokeOnAnyThread` implies `@ThreadSafe` unless all methods in the class are marked `@InvokeOnAnyThread`.