JMRI/JMRI #6200: JMRI V4.12 Browser stylesheet & Script output

slyguy75, 2018-11-26:
I have upgraded to JMRI V 4.12+Rb6a9bb1 on Ubuntu.

When I run a script, the script output window is blank. A print statement in the script shows nothing in the script window, but it does show in the system console. If I run the script a second time, the script output shows the output twice, but the system console shows nothing. It's not a problem, as I can see the output in the console window.

I'm using stylesheet panelfile.xsl, 28,6k downloaded today, and replaced line 2 in my xml file with <?xml-stylesheet href="panelfile.xsl" type="text/xsl"?>. The browser output for Logix, Set Light is reversed from what the actions description within 'edit logix' show, ie within PanelPro, Set Light "CL2" to On  becomes Set Light "CL2" Off.

rhwood, 2020-01-15:
Is this still an issue?

slyguy75, 2020-01-16:
Hi Randall. It still doesn't work, but it is not an issue
  I fired up a new Ubuntu laptop yesterday, downloaded latest JMRI, same indication, ie script output window does not show print output from script, but does show in syslog, so I can live with it.
  I can't get to the original problem report to close it out. Can you delete it? Thx again.
  
   ---------- Original Message ----------

rhwood, 2020-01-19:
Closing at OP’s request.

closed 2021-04-14 by stale[bot]