JMRI/JMRI #6203: NW-1981 Update/fix

gpmccartney, 2018-11-27:
[NW-1981icons.zip](https://github.com/JMRI/JMRI/files/2618250/NW-1981icons.zip)
[NW1981 updatedxml.zip](https://github.com/JMRI/JMRI/files/2618251/NW1981.updatedxml.zip)

Here is an update to this system which fixes some mapping issues with the color light signals.  Also has an updated icon set to fix any name mismatches and missing icon errors.

bobjacobsen, 2018-11-27:
Are those the complete sets of files?  I.e. can I just replace the resources//icons/smallschematics/aspects/NW-1981 directory with the 1st zip file and the xml/signals//NW-1981 directory with the second one?

Thanks

gpmccartney, 2018-11-27:
Yes Bob you can use those to just replace the existing sets.  Thanks again Bob.  Sorry about all the confusion over this set.

gpmccartney, 2018-11-27:
Hold on Bob, another typo was found.  I will submit a fix to the aspects.xml file to correct the restricting speed.  This has been a tough one.

gpmccartney, 2018-11-28:
[NW-1981revisedxml.zip](https://github.com/JMRI/JMRI/files/2622554/NW-1981revisedxml.zip)
[NW-1981icons.zip](https://github.com/JMRI/JMRI/files/2622555/NW-1981icons.zip)

Hi Bob, hopefully this is it.  The folders inside the zip files can directly replace the existing ones in JMRI.  Hopefully all the bugs are now squashed.  Thanks.

bobjacobsen, 2018-11-28:
The list of changed files is appended below. PR #6208 has the changes.

For the next time you update these, it's better to _not_ have spaces in file names, as they can make various maintenance tasks more complicated.  

Thanks for all the work on this!

closed 2018-11-28 by bobjacobsen