JMRI/JMRI #6223: Simple Programmer doesn't understand if no write programmer is available

bobjacobsen, 2018-11-30:
While working on #6161, Dave Heap discovered that the Simple Programmer doesn't work cleanly when a write-capable programmer isn't available:

> With the Power Pro via USB, The Simple Programmer shows "Write CV" as available, although it doesn't work. It does show "No programmer available!" in the status bar when you try.

[image]

pabender, 2018-11-30:
In that case, when no programmer is available, shouldn’t the menu item for the single CV programmer be disabled?

( the full symbolic programmer can still be available in edit only mode.... )

bobjacobsen, 2018-11-30:
> In that case, when no programmer is available, shouldn’t the menu item for the single CV programmer be disabled?

Sure.  Looks like `SimpleProgAction` is already looking for the "no programmer is _available_" case.  What might be happening here is that the NCE connection is providing a programmer that doesn't do writes, so that could be added to the `SimpleProgAction` tests.

pabender, 2018-11-30:
> Sure. Looks like `SimpleProgAction` is already looking for the "no programmer is _available_" case. What might be happening here is that the NCE connection is providing a programmer that doesn't do writes, so that could be added to the `SimpleProgAction` tests.

True, but in that case, should the NCE connection just say it doesn't provide a programmer.  

It can't read CVs and it can't write CVs, which sounds like a no programmer situation to me.

We have other examples of systems that only conditionally provide a programmer (XpressNet when a Lenz Compact or Atlas Commander is the command station, for example).

closed 2021-04-14 by stale[bot]