JMRI/JMRI #6238: Multimeter units not defined; causes NPE when displayed in the AmpMeter frame.

pabender, 2018-12-03:
This probably is a continuation of the discussion that lead to #6116....

The jmri.MultiMeter interface doesn't define units that the track voltage and track current are provided in.

The jmri.jmrit.ampmeter.AmpMeterFrame displays a percentage (in the range 0.0 to 1.0).  If you provide a value larger than 1.0, a null pointer exception occurs due to the way the digits are displayed (which is done by mimicking a 7 segment display with 3 digits).  

The first MultiMeter implementation was for DCC++, which uses the current sensing capabilities of a motor shield to provide track current, it may or may not be possible for DCC++ to generate a measurement with units (I haven't looked at the code).

Now that a bug has been fixed, the Roco Z21 MultiMeter correctly produces a current measurement in mA.  While a current rating is provided by the Z21's built in booster,  the true maximum value for the Z21s current meter is unknown, so a percentage can't be reliably provided.

I think the right path forward is to:
1) re-write the display code in the AmpMeterFrame so that the number of digits is not fixed
2) add a method where implementations can provide the units for the Current and Voltage that can be displayed alongside the value.

but I'm open to other suggestions (and it's late here, so I won't be working on this until at least tomorrow night).

danielb987, 2018-12-03:
Sounds good. It would be nice to have an enum that could tell whenether the value is relative (0 <= x <= 1.0) or absolute and if so if it's millivolt/milliamp or volt/ampere). It could be useful for other parts of JMRI as well to know the unit of these values.

pabender, 2018-12-03:
> Sounds good. It would be nice to have an enum that could tell whenether the value is relative (0 <= x <= 1.0) or absolute 

My preference here is a method that returns a boolean.  There really aren’t any other values possible for the multimeter.

> and if so if it's millivolt/milliamp or volt/ampere). It could be useful for other parts of JMRI as well to know the unit of these values.

I am actually thinking about just fixing the units for now.  I.e. setting the units so that it is Volts for voltage and Amps for current.  As long as the units are known, it is trivial to convert from one unit type to another.

danielb987, 2018-12-03:
It's fine for me.

pabender, 2018-12-03:
@msunderwd do you know if DCC++ reports current or ADC values?

My reading of the base station code seems to show that it cuts off the power to the motor shield if the current is over a maximum value.

The current monitor code that does this is here:

https://github.com/DccPlusPlus/BaseStation/blob/3c90408653175ef614d472723343d6bc310905d3/DCCpp_Uno/CurrentMonitor.cpp#L29-L36

And the maximum value is defined as 

https://github.com/DccPlusPlus/BaseStation/blob/3c90408653175ef614d472723343d6bc310905d3/DCCpp_Uno/CurrentMonitor.h#L16

Which seems to imply the value may have a maximum of 300, but our code treats the maximum value as 1024 which seems to be based on

https://github.com/DccPlusPlus/BaseStation/blob/3c90408653175ef614d472723343d6bc310905d3/DCCpp_Uno/SerialCommand.cpp#L329-L341

bobjacobsen, 2018-12-04:
> It would be nice to have an enum that could tell whenether the value is relative (0 <= x <= 1.0) or absolute

Could we use the nice AnalogIO interface for this?  A base standard for how we handle AnalogIO would be nice.

danielb987, 2018-12-04:
@bobjacobsen 
Is PR #6243 what you are asking for?

bobjacobsen, 2018-12-05:
PR #6243 is nice, thanks, but I was more asking if the `Multimeter` interface could extend `AnalogIO` so that things like this would be consistent across analog measurements.

danielb987, 2018-12-05:
It would be great.

MultiMeter has two values, voltage and current, where AnalogIO has one value. It seems to me that this change would need to add two new interfaces, CurrentMeter and VoltageMeter, and depricate MultiMeter.

AnalogIO is a named bean but MultiMeter and AbstractMultiMeter doesn't. Which means that AnalogIO needs to have a system name, which the current implementations of MultiMeter doesn't have. CurrentMeter and VoltageMeter would need to have system names, but that is maybe a small problem.

danielb987, 2018-12-05:
It would be great to do this change also from another point of view. On big model railroads, where there are several boosters, it would be nice to have a voltage meter and a current meter after each booster. It would in that case be very useful to have a system name and possible a user name for each meter.

pabender, 2018-12-05:
> On Dec 4, 2018, at 11:56 PM, Daniel <notifications@github.com> wrote:
> It would be great to do this change also from another point of view. On big model railroads, where there are several boosters, it would be nice to have a voltage meter and a current meter after each booster. It would in that case be very useful to have a system name and possible a user name for each meter.
> 

I don’t disagree that this might be nice, but the hardware doesn’t currently exist for that.

The two implantation we currently have only provide a single meter for the booster that is connected directly to the command station.  Secondary boosters on these systems don’t provide the same information ( DCC++ doesn’t really have provisions for a second booster. )

The display side of this only displays the amperage currently.

msunderwd, 2018-12-05:
This could be generalized beyond DCC++ though. I could see a set of
RRAmpMeters (or similar) reporting back from various areas on a large
layout.

KenC57, 2018-12-05:
I'm expecting the next step in the breakers would be adding something to
indicate load.
-ken c

danielb987, 2018-12-05:
> I don’t disagree that this might be nice, but the hardware doesn’t currently exist for that.

I have asked my hardware supplier [Mollehem](http://mollehem.se/index.php/en/control-and-electronics/loconet-components) about this. On european modular layouts (ModulSyd and Fremo), we use LocoNet to distribute the booster signal to the boosters. We have a central (for example Z21) that supply the signal to a booster, which in turn sends the signal to a [RSCLD](http://mollehem.se/index.php/en/control-and-electronics/electronic-utilities/rscld-detail) which then sends the booster signal to the boosters over LocoNet. The TAMS booster B-4 has a version with LocoNet connections with makes this very easy.

With this distribution of the booster signal over LocoNet, it would be very easy to add a card that measure voltage and current from the booster and sends the information over LocoNet.

msunderwd, 2018-12-05:
Indeed... while no one may yet have done it, the technology is obviously
already present, and wants only a demand for it to happen (I could see
myself making such a card just for the fun of it).  Also the utility of
such a thing is also pretty clear.  Being able to remotely monitor voltage
and current on the booster outputs of a large, distributed layout would
definitely be useful.

Therefore, it would make sense to architect whatever this voltage/current
monitoring code is morphing into with that capability in mind.

pabender, 2018-12-05:
Ok, so the first step ( which I am going to work on tonight ) is to make sure the front end doesn’t crash.  That I can take care of pretty quickly.

After that, the next step is going to be to re-write the multimeter code that it works with a manager and analogIO based named beans.

danielb987, 2018-12-05:
@msunderwd 
Off topic of this thread, but since you mention possibly making your own card, I throw out my ideas.

My suggestion to Mollehem is a card with this specification:

Hardware connectors:
* Connector for booster (booster to track)
* Connector for track output
* 2 LocoNet connectors

JMRI interface:
* AnalogIO (input): measure voltage
* AnalogIO (input): measure current
* DigitalIO (input): is the booster delivering power or is power off

If possible, but not needed:
* DigitalIO (input/output): Power on/off to track

It would be useful to have a CV register that tells the card to tell it's measurements with a certain intervall (if the LocoNet protocol supports it), for example a CV register that tells the time in seconds between each notification. If the value is zero, the measurements are not sent unless requested.

I'm no expert in electronics, but I would like to have this card. And I would be happy to work on the JMRI support for this device.

msunderwd, 2018-12-05:
That sounds pretty good and quite do-able. Let me ponder it a bit.

msunderwd, 2018-12-06:
Daniel, contact me off list (msunderwd - at - gmail.com) if you want to
further this board discussion.  I have some thoughts and some questions.

- Mark

danielb987, 2018-12-06:
I have sent you an email. My email is daniel.b at bergqvist.se

Daniel

icklesteve, 2019-02-06:
Hi Folks,

I'm looking at displaying the live track current reports from MERG CBUS Command Stations via the JMRI MultiMeter.

At present I've isolated the mA value ( min 0mA max 65535mA ) from the message stream in the master command station, and created a basic CbusMultiMeter class which extends AbstractMultiMeter.

My issue is the NPE when AmpMeterFrame requires a value between 0.0 and 1.0 to be stored in getCurrent()
Would this make more sense renaming to getCurrentScaleBetween0and1 ?
There is no mention of this percentage requirement in jmri.MultiMeter or AbstractMultiMeter
These work fine and the current value is updated ( just the Frame causing the NPEs ).

I've got no way of knowing what the maximum Amps of the Command station is, ( especially some of the DC setups with multiple CBUS command stations ).

Seeing as I've got the data supplied in mA, I'd also like the user to able to see this value.
How many mA is Train A ( the only train in this case ) using on a 2.5A command station?  
380mA seems easier to quantify than 15.2% , but ideally the user should be deciding the displayed scale.

Current hack -
CBUS reports 457mA
getCurrent returns 457
Frame modded so if getCurrent is 1 or greater assume mA value, 
otherwise assume 0.0 to 0.99999 percentage scale value
This avoids 0.457Amps being misinterpreted as a 45.7%

This is my v quick hack to do basic testing with CbusMultiMeter but if it's likely to be rewritten for multiple meters at some point with a Power / Current Manager, then it might be enough of a quick hack to work until such a day when it moves to managers?
Should I instead add a get / set to see if a scale, mA or mV ?

Would I be OK to put an initial WIP pull in for comment or does someone have something up their sleeve?

Steve

Pugwash1, 2019-02-07:
Steve
I have a similar problem with Digitrax. I will need to report  values from multiple devices, some absolute numbers, some percentage, some with known min and max.
I was thinking about returning an array containing objects, each object containing a description, type, min, max, value, units, blah, blah.
Then have a tabbed display or grid in the GUI, user selectable.
Haven't thought it all way thru yet.
Steve G.

danielb987, 2019-02-09:
@Pugwash1 
It would be good if we can use AnalogIO for multiple values. Each AnalogIO is a NamedBean with an identity.

pabender, 2019-02-10:
In #6533, I just pushed a change to the multimeter UI that allows it to autoscale to the input valuers.

Once that passes tests, this should resolve the issue with the NPE.

The display still shows a % symbol at the end of the value.  It would be easy enough just to remove it.

pabender, 2019-02-10:
#6533 has now been merged.  

@icklesteve  Please verify that solves the NPE for MERG.

I switched the Z21 code back to reporting MA.

I think we should leave this issue open until we do something about the units.  It is Likely switching the Multimeter interface to an AnalogIO as suggested by @danielb987 is the right direction.

icklesteve, 2019-02-10:
Hi @pabender 

The frame now takes the mA value as expected, though I'm still refining the technique for MERG connection Command Station identification before I upload the actual CbusMultimeter.

Thanks, Steve.

closed 2021-03-30 by danielb987