JMRI/JMRI #6245: UncaughtExceptionHandler doesn't log ThreadDeath

danielb987, 2018-12-05:
I was looking at the classes in jmri.util and found this:

`jmri.util.exceptionhandler.UncaughtExceptionHandler` has the following code:

```
public void uncaughtException(Thread t, Throwable e) {
    // see http://docs.oracle.com/javase/7/docs/api/java/lang/ThreadDeath.html
    if (e instanceof java.lang.ThreadDeath) {
        return;
    }
    log.error("Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler", e);
    ....
```

ThreadDeath is thrown in the victim thread when the (deprecated) Thread.stop() method is invoked.

Isn't it an error to call Thread.stop() that should be logged? If some code uses Thread.stop(), it seems to me that there should be a message in the log telling about it. Maybe giving a stack trace to tell where it happens? Or isn't this a problem?

bobjacobsen, 2018-12-05:
This JMRI-specific error handler works consistently with the Java-provided one. From the link in the comment:
```
The top-level error handler does not print out a message if ThreadDeath is never caught.
```

Although `Thread.stop()` is deprecated, calling it isn't an error.  That's how the ThreadMonitor window's stop buttons work.  (It would be nice to clean that up some day...)  IIRC, there are also `stop()` calls in a couple other places.

It wouldn't hurt to add an Info message here saying that a thread has stopped, did you intend that?

danielb987, 2018-12-05:
OK. I read on [Stackoverflow](https://stackoverflow.com/questions/16504140/thread-stop-deprecated) that Thread.stop would unlock all the monitors of the thread which could lead to the objects previously protected by these monitors to be in an inconsistent state. And therefore that the call to Thread.stop() could be dangerous.

I can add a info message but it will need to wait a day or two.

PR #6265 (2018-12-12): Add info message when a thread is forced to stop
This fixes #6245.
closed 2018-12-12 by bobjacobsen