JMRI/JMRI #6249: SF.net jmri-developers list is down

bobjacobsen, 2018-12-05:
It appears that the jmri-developers list at SF.net stopped forwarding email sometime on December 1st.  It's not a complete outage. Other things are working. No prognosis from SourceForge yet.

mattharris, 2018-12-05:
Seems like the mailing list on SF is functioning right now, but it would probably be a good idea to accelerate the switch to groups.io.

Several members have signed up over the past few days since the announcement of the proposed switch.

Thoughts?

bobjacobsen, 2018-12-05:
>  it would probably be a good idea to accelerate the switch to groups.io.

I definitely agree!

mattharris, 2018-12-06:
There's been a flurry of activity on the SF list, so once that calms down will do the switch.

As a reminder, it is necessary for anyone wishing to follow/participate in JMRI developer discussion to manually subscribe to the new list via web at https://jmri-developers.groups.io/g/jmri/join or emailing jmri+subscribe@jmri-developers.groups.io

This is due to SF no longer allowing us to access or export the subscriber list.

mattharris, 2018-12-06:
As just recently posted on the SF list, current plan is to switch the SourceForge list to read-only mode some time around 20h00 UTC today (6th December).

After that time, conversations should take place on the groups.io list.

The list archive will remain on SourceForge, but we will organise an import of the archive to groups.io in the meantime.

mattharris, 2018-12-06:
The SF list should now be placed in read-only mode.

Please move any conversations to the groups.io list.

mattharris, 2018-12-10:
Closing this issue now as we've moved the list over.

closed 2018-12-10 by mattharris