JMRI/JMRI #6253: Dispatcher enhancement

robforeman, 2018-12-06:
I have had a need where I wish to have a slow moving train enter a passing siding to allow a faster moving train to overtake it. 
Currently with Dispatcher I try and achieve this by using a pause in the passing siding section.
The problem I find is because that section is only one block in size the next two sections are allocated to the train that is paused and therefore the train trying to pass cannot progress.

Would it be possible to add another two actions to sections one that stores the allocations and remove them so that the second train can pass and a second action that can place those allocations back in the queue?

If this could be implemented the the actions for the passing siding could be store and remove allocations, pause for x fast minutes and then reallocate those sections.

Do you think this would be feasible?

mstevetodd, 2018-12-06:
@Pugwash1 
I'm not sure why a pause keeps the allocations. Is changing the pause to always drop allocations and then ask for them again when resuming? Or maybe a flag on the pause to hold or drop?

Pugwash1, 2018-12-06:
The pause is for pausing at a station say. At my real local (GO Transit) station the signals are green and the track ahead is allocated, but the train stops and picks up passengers.
The allocate by safe sections is for allocating to a passing track, and not allocating beyond unless its clear all the way to the next safe section/end of transit.
If it pauses between safe spots then if it removes allocations ahead it would allow a train between it and its next safe spot , so defeating the purpose.
Steve G.

Pugwash1, 2018-12-06:
If the fast train is  allocating say 5 ahead, when the slow train in front running by safe section pulls in and releases the section behind that triggers the allocation process then the higher priority train will allocate past it.

robforeman, 2018-12-07:
Steve, I have tried this but the fast train allocation (priority 5) does not seem to override the slow train (priority 1) allocation when the slow train has paused.  Do I need to do something with the new safe sections feature?
Rob

Pugwash1, 2018-12-07:
@robforeman @mstevetodd 
I will investigate why the priorities dont work.
I have given "action" idea some thought, the most useful way of doing something akin to this so it can be reused for multiple scenarios, is to have a transit section to be flagged as "Do not allocate past here if sensor X is active", then logix or actions could be used to control further allocations. The push - pop idea will cause unnecessary  work, plus, this is real time, so the layout situation would have changed and the allocation solution may be different than first done, and trains unnecessarily diverted due to allocations made and cancelled. The earliest for this would be 4.15.1, as they are trying to get 4.14 out for Christmas.

robforeman, 2018-12-07:
@Pugwash1 
If the priorities can work correctly I think that would solve the problem. Agree with your thoughts on the push - pop, could get complicated and cause unwanted consequences due to allocations being made and cancelled.  If you find a problem with the priorities and would like testing done on fixes please let me know.

Pugwash1, 2018-12-10:
I've been playing around with options and find that on my test layout the slow train, travelling by safe sections, needs to have the fast train allocating 6 sections ahead to work. This is due to the way the sections and signals line up, so your situation may be different.
One section for the one the fast train is sitting in, then the switch, then track, then the switch for the passing siding. When that section comes available it needs to allocate that section, the passing section and the one after that (to stop the slow train grabbing it), that's the six sections.
I then tried using the idea of a sensor that must be active or not to control allocating passing a given section, when all was said and done I have to look for some thing within four sections behind. 
I tried using hold on signal, but finished up with the same number of sections between the two trains, three, the one the fast trains in and two to allocate past. Still six.
The best bet for moving forward is the sensor approach as it can be easily manipulated thru logix. 
What do you, or anyone else, think.
Steve G.

Pugwash1, 2018-12-13:
@robforeman I think the only way is to add a new action event. It will stop allocating  for xxxFCmins or if  sensor active. Dont know how it should work if there is also a pause action, the pause will try and stop the train as welll as the signal.

robforeman, 2018-12-13:
@Pugwash1 
Sorry I haven't updated, am away from home until early next week.  What I will do is create a simple panel similar to the area I have the issue and upload it with step by step instructions on how to reproduce.  That way we can both be looking at exactly the same thing.
Rob

robforeman, 2018-12-20:
@Pugwash1
Hi Steve,

I constructed a test panel which I will attach.  It is a simple panel with a passing section.  With the slow train using the Slow Train Transit and a priority of one I can set the sensors for the blocks 1-7 SEN1 -SEN7) sequentially and obtain the required allocations.  If I then run the fast train using the Fast Train transit and a priority of five and set the sensors for blocks 1-8 (skipping 7) sequentially it performs the allocations until section nine which is still allocated to the slow train.
[Test_Passing_Dispatcher-1.xml.zip](https://github.com/JMRI/JMRI/files/2698610/Test_Passing_Dispatcher-1.xml.zip)

robforeman, 2018-12-20:
@Pugwash1

Sorry closed this by accident.
Now reopened.

Pugwash1, 2018-12-20:
@robforeman 
Im just starting to write the tests for pauseing allocation on a sensor being active, so I can see if the new way is going to work, but I will look at this and see if something cant be done know.

closed 2021-04-14 by stale[bot]