JMRI/JMRI #6256: Clock synchronization issues

balazsracz, 2018-12-07:
I found the following problems testing fast clock using the OpenLCB protocol. (4.13.6/ binary 4)

1. When a clock slave (JMRI) is setting the fast clock rate, other masters and slaves do not update the displayed rate on the analog clock face. They *run* at the right rate, just display the rate that was last set on that particular instance of JMRI. Closing and reopening the clock display does not fix the problem, so it seems that JMRI's internal idea about the rate is not bidirectionally synchronized to the rate of the OpenLCB object (which is then correctly synchronized to the network).

2. When a JMRI clock slave loads the panel file with the clock config, it always resets the network clock's rate to whatever is saved in the panel XML (from the clock config dialog). It would be desirable to (have an option to) leave the rate untouched.

3. When a JMRI clock slave loads the panel file with the clock config, it always starts or stops the network clock, depending on whatever is saved in the panel XML (from the clock config dialog). It would be desirable to (have an option to) leave the clock running state as is on the network.

I'm wondering how many of these appear on other peer to peer networks, such as loconet. Is it possible to test loconet with multiple JMRIs but without an actual network?

KenC57, 2018-12-07:
My memory of the clock in JMRI is it didn't expect multiple connections that
would have a clock. I agree the initial setting of the clock should be
controlled. I'd say:
1. Load internal from real-time.
2. Load connections.
3. Load panels, clock data in a panel should have option to set or not set a
default clock time and conditions.
4. Inside the connection settings there would be clock data like slave and
master.

Right now #1 is normal and the clock is running. There aren't connection
based time settings, so #2/#4 would need to change with all connections
supporting clocks. #3 doesn't have the option to not set things yet. So a
fair bit of work will be needed. I know LocoNet, NCE have clocks, not sure
which other connections support clocks.
-ken c

bobjacobsen, 2018-12-07:
> On Dec 7, 2018, at 2:23 AM, Balazs Racz <notifications@github.com> wrote:
> 
> I found the following problems testing fast clock using the OpenLCB protocol. (4.13.6/ binary 4)
> 
> 	• When a clock slave (JMRI) is setting the fast clock rate, other masters and slaves do not update the displayed rate on the analog clock face. They run at the right rate, just display the rate that was last set on that particular instance of JMRI. Closing and reopening the clock display does not fix the problem, so it seems that JMRI's internal idea about the rate is not bidirectionally synchronized to the rate of the OpenLCB object (which is then correctly synchronized to the network).

This is probably a bug in the various display classes, SImpleTimebase fires a “rate” event when the rate changes.  That fires update() in e.g. lcdclock.LcdClockFrame, but it’s not decoding that event name. But I’m not seeing any display of the rate there, so perhaps I’m looking at the wrong display?

> 	• When a JMRI clock slave loads the panel file with the clock config, it always resets the network clock's rate to whatever is saved in the panel XML (from the clock config dialog). It would be desirable to (have an option to) leave the rate untouched.

It looks like no thought went into control over setting the rate at startup. The “Start Up Options” section doesn’t include an option for setting (or not setting) the rate.  That could be added, though.

> 	• When a JMRI clock slave loads the panel file with the clock config, it always starts or stops the network clock, depending on whatever is saved in the panel XML (from the clock config dialog). It would be desirable to (have an option to) leave the clock running state as is on the network.

That seems to be a bug, instead of an oversight.  I would have thought the “Start with Fast Clock Stopped” option controlled that.  http://jmri.org/help/en/package/jmri/jmrit/simpleclock/SimpleClockFrame.shtml

> I'm wondering how many of these appear on other peer to peer networks, such as loconet. Is it possible to test loconet with multiple JMRIs but without an actual network?

Normally a LocoNet command station will carry a clock, which has certain odd behaviors, so there are really three cases (1) with a command station clock as master (2) with a command station clock as slave and (3) without a command station clock.  All three should work, but only the simplest case (3) can be tested without a command station.

Bob

--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

balazsracz, 2018-12-07:
is there a loconet network server I could test with that acts as a hub,
similar to the JMRI OpenLCB hub?

bobjacobsen, 2018-12-07:
> is there a loconet network server I could test with that acts as a hub

I think (have not confirmed) that you can configure one JMRI instance with a LocoNet simulator connection and have it open LocoNet Server, then configure the others to connect to it.

balazsracz, 2018-12-07:
Thanks. I will do some more testing.
We should not hold the release for these problems as we're talking about a
new feature; I will attempt to fix these issues before 4.13.7.

pabender, 2018-12-07:
Just FYI, the SRCP protocol also has a fastclock implementation.

BHoffman351, 2018-12-09:
Just my two cents, perhaps this (and also #6270 ) should not be rushed into the 4.14 release.  That is coming up soon and there are many aspects of the fast clock that are fragile.

This is more appropriate for a separate discussion. My opinion is that most of the fast clock needs to be rethought. One part, that seems to cause issues, is the external time source option. If using an external time source, why would JMRI be able to change the time?

balazsracz, 2018-12-09:
Conceptually, a time source and a time operating console are two different things. It is perfectly possible to have an external time source and remote control it. Multiple DCC systems have their time source built into the command station but with the CS not having any user interface, a throttle is used to remote control the time source.
The property we're looking for when the time source and the operating console is separate, is that the layout shall run fine with or without JMRI; it should be possible to stop and restart JMRI in the middle of an operating session to no detrimental effect for the operators. The quoted PR is attempting to achieve exactly this, with keeping features entirely backwards compatible. If you believe backwards compatibility is broken somewhere, please point it out and I'll be happy to fix.

pabender, 2018-12-10:
Just a question... Is JMRI supposed to be able to act as a Fast Clock master on a standalone loconet?
I don't seem to see an option anywhere to enable that.

KenC57, 2018-12-10:
Bob M,

I don't recall the ability to turn off the 'master' fast clock in a command
station. So we'd only be able to have JMRI as the fast clock master when on
standalone. That's why the funny code to 'adjust' the running fast clock
since the code in the Digitrax command stations didn't keep time well.
-ken c

pabender, 2018-12-10:
@devel-bobm @KenC57 

My specific use case is with a Roco Z21, which doesn't  appear to support the fast clock internally.

I was looking at the LocoNet PE Spec document last night, and it notes that early LocoNet Command Stations (the DT200 at least ) did not support this either, so it isn't just a standalone LocoNet that we need to consider, though that certainly is a valid use case.

devel-bobm, 2018-12-10:
@KenC57 - DCS210 at factory default setting responds to "get fast clock slot" with "OPC_LONG_ACK" with code "FAIL".  I don't expect this behavior, as the documentation for the DCS210 states:

> Any DT500 throttle can edit system functions like the fast clock and routes.  Both of these features can be enabled or disabled by your command station.  The DCS210 ships with fast clock and routes enabled.

Perhaps there is an OpSw which changes the DCS210 fastclock master behavior.  If so, I haven't found it yet...

@pabender, JMRI does not currently have a "command station" defined for the DT200 acting as a command station.  That could be rectified.  It would probably affect something on the order of 10 JMRI user; I don't recall seeing any JMRI user group support requests for that "command station"...

I have been working with the `LnCommandStationType` class lately for unrelated reaons.  Perhaps that class could include a new method `public boolean canBeFastClockMaster()`.  That information could be of use in JMRI's LocoNet fast clock support.  (And perhaps, for completeness, I could add the "DT200 acting as command station".)

pabender, 2018-12-10:
@devel-bobm right, we don’t have a DT200 command station defined, and I am not sure there is a high priority in creating that.

We do have command stations ( and standalone LocoNet instances ) that don’t have a master clock, so it would be good if we could get that added.  The JMRI Roco Z21 code currently treats it’s LocoNet like a stand-alone LocoNet, so adding the canBeFastClockMaster() and appropriate code to enable JMRI to respond to the queries would be beneficial for my use case.

balazsracz, 2018-12-10:
On the OpenLCB side we have both alternatives: clock master and clock
slave. I had struggled quite a bit on how to represent this in JMRI. There
is no master/slave choice in the fast clock setup dialog; also such a
choice could not be transmitted to the ClockControl via the existing API. I
have considered making more than one entry in the "select clock" dropdown,
but I was suggested that JMRI does not support at this time the ability to
have more than one ClockControl implementation. Also switching the dropdown
from one to another would not have appropriately destructed the unwanted
implementation, and having the clock master merely constructed would be
sufficient to cause trouble on the network.

In the end I added a separate button in the profile / connection tab for
OpenLCB, that opens protocol settings. The clock can be turned off / master
/ slave mode there.

KenC57, 2018-12-10:
We've come up with a number of items, not surprising. But did we find any
clock issue that should delay 4.14? I've heard lots of good things that
would be right for 4.15.*
-ken c

pabender, 2018-12-10:
> On Dec 10, 2018, at 4:05 PM, Ken Cameron <notifications@github.com> wrote:
> We've come up with a number of items, not surprising. But did we find any
> clock issue that should delay 4.14? I've heard lots of good things that
> would be right for 4.15.*

In my opinion, all of what we are talking about is for 4.15.*

BHoffman351, 2018-12-10:
> On Dec 10, 2018, at 4:05 PM, Ken Cameron <notifications@github.com> wrote:
> 
> But did we find any clock issue that should delay 4.14? I've heard lots of good things that would be right for 4.15.

Bob, Paul, and Ken summed up what I was getting at.  There is nothing specific that I was trying to point out as an issue.

Regards,
Brett

balazsracz, 2018-12-10:
On Mon, Dec 10, 2018 at 10:05 PM Ken Cameron <notifications@github.com>
wrote:

> We've come up with a number of items, not surprising. But did we find any
> clock issue that should delay 4.14?

I would like the fix for the OP be included in 4.14. This fix is ready and
should not cause a delay.
The reason I'd like it in the release is that the JMRI clock slave
functionality for OpenLCB is not a great user experience without it. I'm
working with a manufacturer to bring a standalone clock master to market,
and having support in the prod release of JMRI would be helpful.

bobjacobsen, 2018-12-11:
> I would like the fix for the OP be included in 4.14.

Which branch is that? And what is it based off?

balazsracz, 2018-12-11:
https://github.com/JMRI/JMRI/pull/6270
I rebased it before 4.13.6 was cut so it can cleanly merge into both heads.

balazsracz, 2018-12-23:
Somebody with write access should consider reopening this issue as there is a lot of interesting conversation about loconet code that folks offered help to fix in 4.15 line.

pabender, 2018-12-23:
I agree with @balazsracz this wasn’t resolved ( at least not completely) by #6270, but we probably want to break some of the issues off into separate issues.

newHeiko, 2019-04-19:
I believe there should be a kind of matrix - clock sources (Internal Hardware clock, Loconet...) out of which one master can be selected and clock slaves (Loconet...) which receive the time from the clock source (if enabled).

Clock rate could propagate from the master to the slaves, but it could also be rounded to an integer or overwritten in the process so the master can run at 1.5x speed and the slaves run at 1x or 2x and adjust their time whenever they get a new time from the master.

Correcting the master clock (in case of loconet) would be done on the time source side (pulling in the time from another time source?).

A loconet connection would provide both a time source (from loconet) and a clock slave.

I'll try and draw something up to explain in more detail what I mean, I guess this is a case for drawings vs. words...

Thanks for reading all the way to the end ;)

Heiko

closed 2021-04-14 by stale[bot]