JMRI/JMRI #6259: Fast clock setup dialog apply/cancel button is unintuitive

balazsracz, 2018-12-08:
In the Fast Clock Setup dialog there are a few things that work counter-intuitively.

1. There is an Apply button that doesn't apply anything.
It performs a "save & close" behavior, where it actually asks the user whether they want to save their configuration or configuration+panels. The text in this dialog box is also misleading at that time: it says "remember to save... and you can do that by going to [some menu]..." whereas actually you can save by clicking the save button right there.  There is a Cancel button in the same modal dialog, which does not save the configuration at that point. This cancel does not stop the closing process of the dialog though.

2. There is a Cancel button that does not undo any changes.
After pressing the cancel button, and reopening the dialog I get back the same values that I had changed, whereas my expectation would be to get my changes discarded and back to the original values before opening the dialog.

3. The Apply (effectively close) button behaves differently than clicking the X to close the window.

The root cause for this behavior seems to be that the individual components in the dialog have their action listeners be immediately saving the entered values (or checkbox/combobox state) into the Timebase model. This overwrites previous information there, making an undo / cancel operation infeasible and an apply operation meaningless.

I see two ways to improve the situation:
a) We accept the behavior and update the buttons to have labels that describe the actual behavior. That means to rename the cancel button to Close, put up the warning. Make the window X behave like close. Make sure we display the new warning dialog that has the save buttons. Update the text in the warning dialog to say that the configuration needs to be saved, and the user can use the buttons below, or if they want to save later, then go to menu XXX. Ensure that the menu we refer to actually exists.
We change the Apply button to a Save button.

b) We update the behavior to be more like a dialog box with an apply/cancel buttons.
Remove all actionlisteners that modify the model. Move all code that changes the model from the actionlisteners to the Apply button implementation. Cancel would avoid updating the model but close the dialog. There will be some amount of difficulty with the "Set" buttons that exist today in this dialog and also modify the model, and most of the time also the network state. I'm not quite sure how to deal with them.
Maybe the Apply or Save button should really be inside the "startup options" panel?

@silverailscolo what do you think?

bobjacobsen, 2018-12-09:
Sorry to be coming to this late.

One thing that I've found confusing is the lack of visual separation between the two kinds of things you can do here:  "I want to change the clock _now_" and "I want to change the settings of the clock _at_ _startup_".  

[image]

The parts in the blue boxes are immediate:  They take place now, when the arrowed buttons are pressed, and have no meaning later.  Why aren't those adjacent and enclosed in a single border?

The parts in the gold boxes are for later:  Selections there take place later, and are controlled by the two arrowed buttons at the bottom. Why aren't those controls adjacent?  Perhaps this should even be a separate window from a "Configure" menu option?  

And what does that "Set" button in the middle even do?

And what, if anything, makes the internal-vs-LocoNet selection at the very top persistent?

silverailscolo, 2018-12-10:
Thanks for the fresh views on this pane. I've found the [Set] buttons pretty unique in JMRI. I would prefer to take them out and rely on a working [Apply] button (which would make a [Cancel] possible as noticed by @balazsracz).
Rearranging the 3 parts as 2 blue/gold groups as suggested by @bobjacobsen also makes sense. A direct Save option via the dialog was missing, but of course the text should match.
BTW, the "LocoNet" string in front of Fast Clock in line 2 is replaced by any system name you happen to be using. The "Internal" Fast Clock would mean the JMRI software Fast Clock?

Some time ago I split up the hours:minutes field as all the error checking on a combined entry field was really complex and never good enough. Lots of apps ask you to enter time this way so I would leave that as it is.

Pugwash1, 2018-12-11:
This is from a Loconet perspective, so not all may apply. I pretty much agree with every thing so far, the start-up and the set now should be clearly delineated and the two set buttons to be made one apply (for loconet at least this is done together). The, the synchronize and correct should be left just as currently it makes the correct look subservient to the synchronize.
I would also like a clear selection of the role being played, reading the comments in code "Internal Clock" implies master, and  "Loconet" implies slave, but in fact neither of these options are currently correctly implemented. 
The master mode is supposed to respond to time requests _after_ the CS response ( to overrule it ) , optionally respond to change time requests, and to broadcast the time about every 60 secs. The dtxxx throttles always ask for the time during startup. What it does currently is write, not a time reply,  the new time to master on every fast clock minute if synchronize or correct is checked.
The slave mode is supposed to listen for time broadcasts and if one is not seen in 90 secs ask for it. Nothing else.
A new mode of "totally ignore loconet" would be useful. 
A master / slave module between JMRI instances, or a private fast clock time server / client.

Pugwash1, 2018-12-13:
More Loconet fast clock stuff. If JMRI is master.
If we write the CS (command station) FC (fast clock) slot with a CLK_CNTRL byte of 0x00, bit 6 off then if the CS resonds to a FC request it reply with a message that say "ithis data isnt valid", the DT402 & DT500, my only 2 test throttles ignore it.
We send our time with CLK_CNTRL = 0x40, bit 6 on indicating data is valid. DT402 & DT500 throtlles use it.
We must ignore all FC slot data where our throttle ID is in the ID1,ID2, they are our messages being echoed back, which can cause a war like escalation of data..
If JMRI is slave then we only listen.

PR #8419 (2020-04-24): Fast Clock Setup Window + Clock Javadoc
Update for Fast Clock Setup Frame layout, close window handling.

Moves Load / Store Strings to higher Bundle level.
Javadoc for various Clock related classes.

Fixes #6158
Fixes #6259

[image]
closed 2020-04-24 by bobjacobsen