JMRI/JMRI #6275: Adding an AnnouncementManager?

danielb987, 2018-12-10:
Maybe this is going over the river to get water, as we say in Sweden. But in PR #6260 where is a discussion about if announcement messages to throttles could be sent thru a StringIO.

I'm thinking if there is a reason to make a more general approach to this. For example, in issue #6238, @msunderwd and I have a discussion about developing a card for measure voltage and current after each booster. If this card has a power on/off switch controlled by a DigitalIO, it could be useful to have one DigitalIO controlling all of these measuring cards, meaning that if setCommandedState() is called on the main DigitalIO, then all the connected DigitalIO are switched on or off.

For the throttles, JMRI have support to send a LocoNet command with a text message to LocoNet throttles like DT402 and the Withrottle protocol has support for a text message and Engine Driver can display it.

My question is if it's useful to add a `AnnouncementManager` that can have DigitalIOs, AnalogIOs and StringIOs that set the state of other items. It would then be easy to use these announcement items in panels, scripts, and so on.

These announcement NamedBeans could maybe have letter 'Y' as default system letter, since that letter doesn't seem to be in [use](http://jmri.sourceforge.net/help/en/html/doc/Technical/Names.shtml).

In the example of sending announcement messages to throttles, it would mean that a global StringIO is created with a global system name, like 'YTA1' (Y=System letter, T=Throttle, A=Announcement, 1=First NamedBean), and then LocoNet would create a LocoNet specific announcement StringIO, WiThrottle would create its specific announcement StringIO, and so on. And writing to the global StringIO would then send the string to all the other connected StringIOs.

pabender, 2018-12-10:
> These announcement NamedBeans could maybe have letter 'Y' as default system letter, since that letter doesn't seem to be in [...]

This would be an Internal Manager, so use I for the system letter.

Because of the way in which we parse type letters, That needs to be unique.  

You could have IA1 as an example system name ( I for internal, A for Announcement, and anything after that is an identifier for the specific thing )

mattharris, 2018-12-10:
> You could have IA1 as an example system name ( I for internal, A for Announcement, and anything after that is an identifier for the specific thing )

Except we wouldn't be able to use `IA1` as that's already taken by Audio objects: http://jmri.org/help/en/html/doc/Technical/Names.shtml#types

pabender, 2018-12-10:
> > You could have IA1 as an example system name ( I for internal, A for Announcement, and anything after that is an identifier for the specific thing )
> 
> Except we wouldn't be able to use `IA1` as that's already taken by Audio objects: http://jmri.org/help/en/html/doc/Technical/Names.shtml#types

So that is taken.  I do see there is one duplicate on the list already.  Reporters and Routes both use R.

mattharris, 2018-12-10:
That's true...

We _could_ get away with it as all `Audio` objects have a sub-type; one of `L` Listener, `B` Buffer, `S` Source; giving names such as `IAB1`, `IAS45`, etc.

So we could have a possible way of determining the correct managers to use.

danielb987, 2018-12-10:
Maybe have a colon as sub type? That would make it easy to separate from Audio.

`IA:D<subname>` - Announcement DigitalIO
`IA:A<subname>` - Announcement AnalogIO
`IA:S<subname>` - Announcement StringIO
where `<subname>` is some characters for that NamedBean.

An Announment StringIO for throttles could for example be `IA:SThrottle0`.

bobjacobsen, 2018-12-10:
Before we go too far on name format, perhaps we should discuss name _meaning_?  

I still don't get what the name refers to.  

1. Is it a specific _announcement_, like "The station is now closed"?  

2. Or is it an "announcer" of some kind, who's meaning is "If you ask this to announce a string, it sends it to objects A, B and C"? 

3. Or is it some kind of "announcement system", where you get the Announcer then use a method call to say "Send this string to objects A, B and C"?  (This is what comes to mind from the example "An Announcement StringIO for throttles")

Only (2) looks like a bit like a NamedBean to me.  (1) doesn't need, or even benefit from, being a NamedBean. (3) is more like a PowerManager or Programmer: Something that's just accessed by type as an instance.

If (2), then the system name could be made from the list of names of objects to be told (we do this already for i.e. SE8c head names).  Something like (not a proposal) `IA:IT1:IT2:LT12`.  But notice that to create an example, I had to switch to Turnouts from Throttles, because Throttles are not accessible as NamedBeans (and I'm not sure they're the right target in any case; c.f. Paul's example of targeting an annunciator device)

I apparently don't quite understand the use case here. Sorry for my confusion.

pabender, 2018-12-10:
Sent from my iPhone

> On Dec 10, 2018, at 8:57 AM, Bob Jacobsen <notifications@github.com> wrote:
> as NamedBeans (and I'm not sure they're the right target in any case; c.f. Paul's example of targeting an annunciator device)
> 

Right, my thoughts after sleeping on this are along the lines of a station annunciator which could be some combination of a message board and a text to speech device ( for those announcements such as “next train arriving...” ).

There is another use case I though of today that would be somewhat useful on a Digitrax throttle that supports messages.  Assuming you have a way to identify the specific throttle assigned to a train in a specific block or station, you could send the throttle a message identifying the block or station.

( identification of the train comes from either a reporter or from block following, so that part of this use case is currently possible )

danielb987, 2018-12-10:
> Assuming you have a way to identify the specific throttle assigned to a train in a specific block or station

LocoNet command OPC_WR_SL_DATA has ID1 and ID2. I'm not sure, but I suppose that's the same ID that is used by `LnMessageManager.sendMessage()`. And OPC_WR_SL_DATA has the loco address too. So if there are a listener that listens to LocoNet and catches these ID:s, I think you would be able to keep a list of them together with the loco address.

And I think this would not be a big problem to implement for WiThrottle as well (for use with Engine Driver).

pabender, 2018-12-10:
Sent from my iPhone

> On Dec 10, 2018, at 9:27 AM, Daniel <notifications@github.com> wrote:
> Example usage:
> 
> Cards that measure voltage and current from a booster and reports on LocoNet, which has a power on/off switch controlled by a DigitalIO. An announce DigitalIO can be used to turn all of the cards on or off.
> Cards that controls the light in the room (ceiling light) by an AnalogIO and there an announce AnalogIO can set the value of all of these.

Personally, I can see the string use case, but I am having trouble seeing the use case here.  Also, I think those use cases are handled by the existing infrastructure.

In other words, when some event occurs that requires an action on multiple named beans, you either

A) use an aggregating structure to make multiple things behave as one.  This is what a structure like blocks does ( potential multiple sensors become one block)
B) inform multiple things of changes.  We handle this both through listeners and through things such as logix.

danielb987, 2018-12-10:
That's ok for me. I created this issue to tell my ideas to see if it was worth working on.

bobjacobsen, 2018-12-10:
> Example usage:
> 
> 	• Cards that measure voltage and current from a booster and reports on LocoNet, which has a power on/off switch controlled by a DigitalIO. An announce DigitalIO can be used to turn all of the cards on or off.
> 	• Cards that controls the light in the room (ceiling light) by an AnalogIO and there an announce AnalogIO can set the value of all of these.
> 	• Throttles that can display a text message by a StringIO. An announce StringIO can be used to send a text message to all types of throttles of all connections that supports this (LocoNet, WiThrottle).
> The reason to why we have created DigitalIO, AnalogIO and StringIO is that it allows panels, Logix, scripts, and so on to use the generic interface instead of being forced to use more specialized versions like turnout or sensor.
> In some cases, it could be of interest to tell many items the same thing. Like turning off a lot of boosters or sending a message to all throttles. This can be done using something like a PowerManager, but it would require to add support for that PowerManager.
> 
> If JMRI had support for announcement NamedBeans, you could add that NamedBean to a panel, a script or a Logix, and everything would simply work.
> 
> Lets say you have 10 boosters where each booster has a voltage/current meter with a DigitalIO that could turn on/off the power from that particular booster. And that you have a panel that shows all the boosters, their voltage and current, and if they send out power to the track or not.
> 
> It would then be nice to also have a "main button" that power on/off all these boosters. Click that button and every booster is turned off. Click again and every booster is turned on.

Definitely see that the grouping could lead to simplifications.  It’s something that Logix _can_ do, but it becomes complicated:  “Define an internal sensor (what’s that?), then a Logix (what’s that?) to listen to that sensor and drive all the outputs via an Action (What’s that?), and don’t forget to also drive them the other way…”  Something simpler for digital output already sounds like it would be well received.

But there, it’s more of a multiplexer:  Takes one input and drives N outputs.  If that’s a bit, then it sends a bit.  If it’s a string, it sends a string.  “Announcing” is just one specific case.

BHoffman351, 2018-12-10:
> On Dec 10, 2018, at 9:43 AM, Daniel <notifications@github.com> wrote:
> 
> And I think this would not be a big problem to implement for WiThrottle as well

I hear that a lot! ;)

It is not a problem to implement it in WiThrottle, but Steve was using the message format for (what I felt was) debugging. It may make sense for @mstevetodd <https://github.com/mstevetodd> and I to come to a consensus on the apps end, but I see no reason why it would not be a good addition.

And, in this case, yes it is easy for me to add it to the WiThrottle app.

Regards,
Brett

danielb987, 2018-12-11:
> And I think this would not be a big problem to implement for WiThrottle as well
> I hear that a lot! ;)

@BHoffman351 : I apologize! I wasn't referring to the WiThrottle app when I wrote it, I was referring to implementing StringIO in the WiThrottle server in JMRI. But it would be useful if you add it to the app  :)

closed 2021-04-14 by stale[bot]