JMRI/JMRI #6279: Move turnout icon on a panel doesn't work on Java 9 and above

danielb987, 2018-12-10:
Tested on:
Windows 7, Java 1.8
Windows 10, Java 9
Debian buster/sid (Debian testing), Java 11.0.1

I created a panel and added a right-hand turnout. This worked fine on both Windows 7 and Debian testing. I then tried to move the turnout by holding down right hand mouse button and drag the icon. This works fine on Windows 7 but on Debian testing this opens up the popup menu and I'm unable to move the turnout icon. If I do a single click on Windows 7, I get the popup menu, but if I hold down the mouse button I can drag. But on Debian it doesn't matter if I do a single click or if I hold down the mouse button. But I can throw the turnout with the left mouse button. Could this problem be related to that my Debian uses Java 11? I have confirmed that I have this problem also with Java 10 on Debian testing.

I have a similar problem on Windows 10, Java 9. If I hold down right mouse button on the turnout icon and moves the mouse, nothing happens, but then I release the mouse button, I get the popup menu for the panel (Background color and Add item). If I do a single click with the right mouse button, I get the turnout popup menu. Left click throws the turnout.

Java 8 works correctly, but on Java 9, Java 10 and Java 11, I'm unable to move the turnout icon on a panel.

The type of panel I tested is "Panel Editor".

I have a standard USB mouse with two mouse buttons and a mouse wheel.

bobjacobsen, 2019-01-20:
Java 9 and 10 have introduced changes to how the Swing Look&Feel implementations are done.  They've become serious about using PL&F methods for identifying clicks, pop-ups, drags, etc. Previously, these were only recommended for use, and work-arounds were possible; some of them required system-specific code.  

JMRI has a lot of places were we made the user interface "nice" by breaking away from the standard interactions.  Eventually, we're going to have to fix this.  The roll out will be a bit tricky, as in some places and for some computers the user interface will have to change.  But getting it consistent (even on a single computer!) is still a good thing.

I don't know whether it makes sense to fix the current Issue specifically or not (Note that LayoutEditor and ControlPanelEditor have code that's aimed at specific things like this:  C.f. `isMetaDown()` in LayoutEditor and the comment just before it.  Perhaps Yet Another Patch is the thing to do.

danielb987, 2019-01-20:
Yes, I fixed the LayoutEditor a while ago in PR #4246. I could do the same for PanelEditor if it's desired. I forgot about that when I created this issue.

bobjacobsen, 2019-01-20:
Well, I'm not in a position to do the work, so it's not clear my opinion matters.  

A patch like this (from LayoutEditor) is probably a fine way to solve the immediate problem, even though it's redefining the "is the Meta key down?" idea as something quite different:
```
    // The meta key was until Java 8 the right mouse button on Windows.
    // On Java 9 on Windows 10, there is no more meta key. Note that this
    // method is called both on mouse button events and mouse move events,
    // and therefore "event.getButton() == MouseEvent.BUTTON3" doesn't work.
    // event.getButton() always return 0 for MouseMoveEvent.
    private boolean isMetaDown(MouseEvent event) {
        if (SystemType.isWindows() || SystemType.isLinux()) {
            return SwingUtilities.isRightMouseButton(event);
        } else {
            return event.isMetaDown();
        }
    }
```

On the other hand, it's probably better to avoid redefining things like "meta key is down" and instead have some common (built into `JmriJFrame`?) gestures so that things aren't different in different places.  The `LayoutEditor` class alone has 17 places were `isMetaDown()` is part of logic, and they're far from consistent. Many times, `isMetaDown()` (as modified) is considered on equal terms with `isAltDown()`:

```
LayoutEditor.java:5487:                 && allControlling() && !isMetaDown(event) && !event.isAltDown() && !event.isPopupTrigger()

LayoutEditor.java:5758:         if (!isMetaDown(event) && !event.isPopupTrigger() && !event.isAltDown()

LayoutEditor.java:6548:         if ((selectedObject != null) && (isMetaDown(event) || event.isAltDown())
```

but in other cases it's not:
```
LayoutEditor.java:5045:         } else if (allControlling() && !isMetaDown(event) && !event.isPopupTrigger()

LayoutEditor.java:6559:             if ((selectedObject != null) && isMetaDown(event) && allPositionable()) {
```

or even some other modifier-combination logic appears:
```
LayoutEditor.java:6745:             } else if (selectionActive && !event.isShiftDown() && !isMetaDown(event)) {
```

If we extend to other classes in `jmri.jmrit.display`, let alone the rest of JMRI, we see even more combinations.

It's definitely confusing to users.  People are periodically confused on the user list. I don't use Layout Editor much, but I find it's macOS behavior really inconsistent with everything else. 

Again, an immediate patch is probably fine. Just document it well. 

But in the long term, having a single set of decoders for common gestures is going to save a _lot_ of code and maintenance effort, while giving a better user experience.

geowar1, 2019-01-20:
I agree that the event dispatching code in LayoutEditor’s mousePressed() method is awful; I’ve been tempted multiple times to refactor it… but it’s so convoluted (and has known bugs)… it’s going to a major PIT*! I just haven’t gotten motivated to do it… yet.

danielb987, 2019-01-21:
Should I close this issue now when PR #6470 is merged? Or keep it as a reminder that gestures should be refactored?

closed 2019-01-23 by danielb987