JMRI/JMRI #6281: managing command station features and capabilities

devel-bobm, 2018-12-10:
In general, JMRI might benefit from some formalized specification of command station features and capabilities.  Current JMRI code tends to hard-code for the known features of a given command station.  Think "what mobile decoder addresses are 'short addresses' and what ones are 'long addresses', and how can/should JMRI distinguish between the two".

It might be beneficial to provide a schema-based soluiton to defining and communicating command station "features".  This I think that it would be a useful thing to have for LocoNet-based connections, where the various supported command stations have a wide variety of features and behaviors.

We have already implemented XML-based definition of Digitrax command station configuration.  This makes it possible for the symbolic programmer to configure Digitrax command station "Op Switch" settings.  And I would prefer to use a single XML file for both command station "Op Switch" and command station "feature" information, rather than maintaining info in two separate files.  But I do not see any existing convenient way to add the "features" information within the existing `decoder.xsd` schema.  Is there a good way to extend the decoder.xsd schema to include "features" informtion?

If not, a new XML/XSD-based mechanism is appropriate.

Regardless of where the information is kept, I imagine that the mechanism would have to be somewhat "free-form", since different command stations have different care-abouts.  NCE command station care-abouts are not the same as Digitrax command station care-abouts, which are not the same as Easy-DCC command station care-abouts, ...  Where similar concepts apply, the schema could enforce commonality, such as requiring a definiton of the ranges of "short addresses" and "long addresses".  For other less-common cases, more-generic "feature" elements and attributes would provide the raw information.  Connection-specific JMRI code would need knowledge of the less-generic "features" and be know how to interpret the information.

Here are some of the "features" I'd think may be useful to pre-define for each type of Digitrax command station:
- Number of locomotive control slots
- a list of valid locomotive control slots
- highest locomotive "short address"
- lowest locomotive "long address"
- Can provide "Fast clock master" (ref #6256)
- number of supported command station "Routes"
- number of turnout commands per "route"
- supports "expanded slots"
- supports locomotive address "aliasing"
- supports programming track
- supports independant operation of programming track when main track is operable
- support for various service-mode and operations-mode programming modes
- support for analog throttle(s)
- slot(s) associated with analog throttle(s)
- supports "Query mode"
- list of supported "Query mode"-reportable data
- list of supported "Query mode"-resettable data
- supports LocoNet reporting of track output "short-circuit" events
- supports LocoNet reporting of track output "auto-reversing" events
- supports "OPC_IDLE"
- supports conservative/aggressive service-mode programming track algorithms

I suppose that this proposal really implies some sort of abstract "Command Station" object, with connection-specific command stations inheriting from the abstract object, and including fields and methods associated with the connection-specific information from the above-mentioned XML.

Thoughts?

pabender, 2018-12-11:
> On Dec 10, 2018, at 6:17 PM, Bob Milhaupt <notifications@github.com> wrote:
> In general, JMRI might benefit from some formalized specification of command station features and capabilities. Current JMRI code tends to hard-code for the known features of a given command station. Think "what mobile decoder addresses are 'short addresses' and what ones are 'long addresses', and how can/should JMRI distinguish between the two".
> 
Part of the reason most of this is coded into Java, as opposed to being coded into XML is that this isn’t something that is going to change.

For XpressNet this type of change would largely just be changing one type of hard coded file for another, and I don’t see much advantage to that.
> I suppose that this proposal really implies some sort of abstract "Command Station" object, with connection-specific command stations inheriting from the abstract object, and including fields and methods associated with the connection-specific information from the above-mentioned XML.
> 
If we move forward with this, the object can’t be called a command station.  We already have one of those ( it is the raw packet sending interface ).

Pugwash1, 2018-12-11:
Bob. If moving this forward for all then you could do it for Loconet in LnCommandStationType.
This information would be really useful for loconet.

devel-bobm, 2018-12-11:
@Pugwash1, 

I will consider making this a LocoNet-specific improvement.  It just seems silly to implement at the LocoNet level when it could be useful for other systems...

pabender, 2018-12-11:
@devel-bobm 

Once you get a PR ready, I might be able to use it, but I wil have to work out details further.

devel-bobm, 2018-12-12:
@pabender,

FYI - I do not expect to have a flushed-out implementation any time soon.

rhwood, 2018-12-12:
Coding these features into XML and providing a generic interface would be useful as it could make providing a headless, web, iPhone, or Android interface for setting up JMRI significantly easier to write.

bobjacobsen, 2018-12-12:
I agree with the comments in the original Issue paragraphs from @devel-bobm and the comment above from @rhwood:  having an XML-based (or properties-based) way to specify command station features would be good.

The existing Digitrax mechanism isn't actually all that great; we could do a lot better now using a file of key-value pairs organized by type of command station.  

```
  <type>DCS50</type>
    <name>Zephyr</name> <!-- can be I18N'd -->
    <key>slots</key><value>10</value>
  </type>
  <type>DCS240</type>
    <name>
    <key>slots</key><value>120</value>
  </type>
```
etc

bobjacobsen, 2018-12-12:
Maybe the thing to build is a `jmri.util.KeyValueStore` class that can read an XML file and return 

```
Map<String, Map<String, String>>
```
where the first/outer key can be used as the command station name, and the inner are the key and value for specific items.   That could then be used easily across the program.

A bit more elegant would replace the inner Map with a class that provide the name, Map and perhaps other context info later on.

icklesteve, 2018-12-14:
Hi Folks,

I'm looking at how to define CS capabilities for MERG connections and can see how this would be beneficial.

For CS versioning I'd be tempted to go a step further.
Eg, a query to command station returns
CS Name : CANCMD, Major version : 4, Minor version : B

If there's no pointer for a particular feature for a CANCMD 4 B , use base definition for a CANCMD 4
If there's no pointer for a particular feature for a CANCMD 4, use base definition for a CANCMD
If there's no pointer for a particular feature for a CANCMD, use a basic default definition.

As the source code is on Github for some CBUS command stations, when people customise their builds I don't want to restrict their feature set by being a completely unknown CS ( just partially unknown! ).

I can always code this minor version thing within MERG or extend the proposed class, however if big manufacturers start to offer more firmware updates may be worth thinking about from the outset?

For info, not sure if any of these worth adding system-wide?
Current care-abouts :

- Dispatch or Release when throttle closed ( Boolean? tbc. )
- Steal available Boolean
- Steal NV ( node variable setting ) 0-255
- Steal NV bit flag 0-7
- Share available Boolean
- Share NV ( node variable setting ) 0-255
- Share NV bit flag 0-7
- Force max session ( slot ) limit Boolean
- max session NV 0-255
- Current ( Power ) Refresh time NV 1-255
- Current ( Power ) Multiplier factor NV 1-255

( The NV's are pointers to which NV to query, not the actual value to use.
For many features, the availability isn't confirmed until the query response from the CS has been received. )

Steve

devel-bobm, 2018-12-14:
I wrote: 

>  I envision file access as a solution which is likely to be specific to the usage.

To be more clear, I envision that file access is handled as part of the code which makes use of the `jmri.util.KeyValueStore` class, not code implemented within the `jmri.util.KeyValueStore` class.

rhwood, 2018-12-14:
> FYI,
>
> I have started on the jmri.util.KeyValueStore class as suggested by @bobjacobsen.
>
> At this point, I am not building any sort of persistence (i.e. file) access into the class. I envision file access as a solution which is likely to be specific to the usage.
>
> For the case where I wish to encode Digitrax command station capabilities in a file which is not intended to contain any run-time-modifiable data, should such a file be placed in a new sub-directory of jmri/resources?
>
> Is there a particular filename "extension" which is traditionally used for files which contain Key/Value pair data?

Does the java.util.prefs.Preferences class not meet the requirements we have for key/value stores?

https://docs.oracle.com/javase/7/docs/api/java/util/prefs/Preferences.html

Randall

bobjacobsen, 2018-12-14:
> Is there a particular filename "extension" which is traditionally used for files which contain Key/Value pair data?

From the earlier discussion, it's not just key-value data.  People want levels of heirarchy for it, which is difficult to do with flat stores.  Hence the suggestion of a XML format (which is good at handling nesting).

More to follow later this evening...

devel-bobm, 2018-12-15:
The @bobjacobsen solution is more-elegant than my code.  I will shelve mine.

rhwood, 2018-12-15:
The MetaDataStore looks and behaves like a simpler version of java.util.prefs.Preferences (I only bring this up because the Preferences classes can use multiple types of data stores (properties files, xml files, databases, registries), and allow storing and retrieving known types (String, boolean, integer, floats, doubles, byte arrays). You might want to look at reusing that code instead of writing a new version from scratch. The only restriction the Preferences have on a key is that its not empty and contains no '/'s (this allows retrieving nested keys as 'parent/child'). We already use Preferences objects within JMRI, with a properties file backing, so that code can be reused.

devel-bobm, 2018-12-18:
@rhwood, I am new to the java.util.prefs.Preferences concept, so it is not intuitively obvious how to apply it to the use-case I'm interested in.

It seems perfectly good for _configuring_ a bunch of options, but that's not what I want to do.  I want to pre-define the set of available LocoNet command stations and their various capabilities and characteristics, and ship that information as part of a JMRI distribution.  I don't want to hard-code the capabilities and characteristics into JMRI.  I don't want the user to be able to change those characteristics  or capabilities (at least not from within JMRI).  I don't see an obvious application of Preferences to this requirement.

Would I code up something (that won't ever be part of JMRI distributions) to create the `Preferences`-based class which can be used to create the file, then ship the resulting file (not the code) as part of JMRI?  And then I would code up a class (which implements `Preferences`) that is capable of reading (but not writing!) that file object and making its "nodes" and "key/value pairs" available to other JMRI code, and ship that as a class within JMRI distributions?

KenC57, 2018-12-18:
Bob M,

I think the idea is the xml, like what the preferences reads (or like
decoder definition files), is easy for the program to read. So the edit (and
create) is outside of JMRI. So they are manually edited files that become
part of the distribution.

-ken c

rhwood, 2018-12-18:
@devel-bobm if you ignore the write part of the Preferences API it has a complete API for accessing primatives by type. I wasn’t intending to suggest that we create a read-only Preferences structure, but that it’s API be looked at for how Bob’s should be completed.

That said, if you wanted to create a backing store for a Preferences class, one of the classes in jmri.util.prefs does implement a .properties (non-xml, one key=value pair per line) backing store for Preferences. (Note that this implementation uses a special form of a Java package as the first part of a key.)

If we were concerned about readability by users, I’d think that a .properties file would be more readable than XML, if not, the existing infrastructure for handling XML should be leveraged.

devel-bobm, 2018-12-19:
@bobjacobsen,
I tried the code from your `bj-command-station-meta-store` branch, but ran into a test failure with the as-written code and XML and test.  When run, I get a null value instead of "Val1" for the test line:

`Assert.assertEquals("Val1", m.getNestedContents().get("Store 1").getItems().get("KeyA"));`

A bit of debugging reveals an empty list for `m.getNestedContents().get("Store 1").getItems()` .

I believe that the code for "nested contents" is not working as you intend.

devel-bobm, 2018-12-20:
@bobjacobsen,

If I modify to the second "for loop" in the `loadContent(Element root)` method as shown below, the test you've written will pass with the XML file you provided.

```
        for (Element c: root.getChildren("store")) {
            String key = c.getChild("key").getText();
            Element val = c.getChild("value");
            MetaDataStore value = loadContent(val);
            retval.nestedContent.put(key,value);
        }
```

Does that modification fall in-line with your intention?

bobjacobsen, 2018-12-20:
Could you please fetch the branch again?  I think you somehow didn’t get the last commit to it.

> On Dec 19, 2018, at 1:41 PM, Bob Milhaupt <notifications@github.com> wrote:
> 
> I believe that the code for "nested contents" is not working as you intend.

--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

devel-bobm, 2018-12-20:
@bobjacobsen,

I orignally did a "cut-and-paste" "manual file creation merge" via a web browser, using the "blob" links provided in your post on `MetaDataStore` code and test and sample file.  For whatever reason, the web browser showed me the original version, not the updated version.  

Today, I used the "history" button on the web page, selected the second commit, and found the newer code.  I've tried it and it seems to work for me.  I cannot begin to guess why the web browser view of the file was not showing me the most recent code...

I'm still getting my hands around what I can/cannot do with this class and the associated file format.  I think it will be able to achieve the functionality I need, but I have not implemented the full set of characteristics, so cannot yet confirm the suitability of `MetaDataStore` and its associated file format.

closed 2021-04-14 by stale[bot]