JMRI/JMRI #6283: Edit LE Icons to user files fails

Paul-T, 2018-12-11:
Release 4.13.6 appears to have fixed the ability to add a user chosen icon in Layout Editor.
Edit existing icons does not work.

Jenkins 4.15.1ish has same bug

Pugwash1, 2018-12-11:
I have just tested this every way to sunday and it works both for edit and add.
The add is select the icon and then click on the one you want to change.
The edit is drag the icon to the one you want to change. If the icon is small its very, very hard to select it for the drag, you must drag the icon not the file name, but when you get it it works.
Steve G.

Paul-T, 2018-12-11:
You are correct.
Nowhere did I see that edit uses drag & drop!!
For a novice, using two different methods seems unhelpful.  However I can live with this.

closed 2018-12-11 by Paul-T