JMRI/JMRI #6301: Cab signals

pabender, 2018-12-14:
I have been working on signaling my train club’s layout which uses a Digitrax systems.  We aren’t ready for physical signals yet, but I want to start giving operators  signal feedback as soon as possible.

As you may know, Digitrax DTxxx throttles ( at least DT400 and above, possibly the DT300, but I don’t have one to confirm ) have a 3 position semaphore cab signal ( it can also blink, so I guess that gives 7 possible aspects ).

The mechanics of getting the cab signal to work is known, and we have distributed a script with JMRI for years to make them display something ( see 
https://github.com/JMRI/JMRI/blob/master/jython/cabsignaller.py ).  This is really just a test script, but it tells us a few important things:

1) the message format ( which is just a peer transfer ) including the payload
2) the payload includes the locomotive address and the indication.

The use case for this is we want to be able to place a locomotive somewhere on the track and, once the locomotive’s initial position and it’s direction of travel has been identified to the system, the cab signals should appear on the throttle with an appropriate aspect for the current block.  ( initial position may be automatic ( using a reporter ), initial direction of travel may require entering the next block ).

Going through the thought exercise on making this work, I see the following as functions that contribute to the use case:
1) existing signal logic, since the cab signal should be reflecting conditions ahead.
2) tracking of trains between blocks after identification ( either manually or via a reporter )
3) direction of travel sensing ( possibly by watching throttle, but at least via movement between blocks )
4) transmission of the signal aspect ( which is basically just going to require converting the script to java code )

Some of the code required to make this work already exists in jmri.jmrit.dispatcher, but the dispatcher code assumes predefined trains, with defined starting and ending points, and therefore a defined direction of travel, so it can’t be used without modification.

Have I missed anything we need to tie together to make this work?

Btw, modeling this based on PRR practices, the cab signals are, basically, mobile ABS signals.  They eliminate the need for fixed wayside signals, at least for intermediate blocks.

pabender, 2018-12-14:
@KenC57 I played with the existing tool we have the other night with a DT402 ( connected to a Roco Z21 ) and it behaved.  The script only modifies 3 bytes in the message ( 2 for the locomotive address and one for the aspect ) sends.  I know the peer xfer message allows changing the destination address, and the script doesn’t change that.

I am ok with this not working for all Digitrax throttles which might be able to display the signals.  Most of the club actually uses UT4D throttles, which don’t respond to this anyway.

That said, I am really more interested in the mechanics behind how it should work than that it actually works with all LocoNet throttles that are capable of doing this.  

Getting the mechanics right will allow us to expand this to other things we have more control over ( which could be WiThrottle/EngineDriver, but it could also be something else )

icklesteve, 2018-12-14:
Hi Paul,

No dispatcher, routes, Lroutes, oBlocks, Sections, Transits needed, just blocks ( and their paths ).
See the Signal Data / Cab Data columns for an experimental feature in MERG CBUS Command station Monitor.

http://jmri.org/help/en/package/jmri/jmrix/can/cbus/swing/cbusslotmonitor/CbusSlotMonitorPane.shtml

Starts approx line 1333
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/can/cbus/swing/cbusslotmonitor/CbusSlotMonitorDataModel.java

How this is done roughly -

Get a copy of block list
Loop through list and match value with Loco ID ( or column for alternative value, eg UK headcodes )
When block value change event is the loco, get block direction and follow the path to the next signal ( only done in SML so far )
If the next block is null, query for bumper stop.
Get aspect etc. of signal
Remove changelistener from previous signal, add changelistener to new signal in case changes.
Todo - Add changelistener to path in case points change.

On change of aspect I've temp. coded a link between the BR2003 aspects to the corresponding experimental OPC values.

On value change, an updated CBUS message is sent with Loco ID as identifier which can then be picked up by the handsets / whatever.

When the CS monitor table spots the session loco ID has changed direction, the JMRI block direction is reversed 180deg and the signal lookup method is re-triggered.

Have added a "Block Button" to the loco session table,
state 1 : Lookup Block - Click to search all the block values to get started
state 2 : Reverse block direction ( if state 1 gave the wrong answer )

The blocks need to be initialised after the LE panel(s) loaded so I've added a "no blocks located" and a menu dropdown to reinitialise the block list as part of #6296 
It'd be good to do this automatically at some point ( saving any interesting block details before re-init ), though at least for now it's clear to the user.

The experimental OPC also has a speed byte, the next things I'll add are
Block max. line speed
If loco ID from roster ID is speed profiled, speed profile speed.

Also note I've added individual loco table row disable cabsig checkbox + master cabsig on / off button.
Dispose sends the experimental opc disabled message to each loco ID currently active.

Am currently preparing a CBUS command station simulator, at present you can't use throttles with the sim. MERG CBUS connections so unable to test / develop this offline.

I was planning to develop the routines further ( including as much CBUS de-coupling as possible ) before discussing moving into JMRI core, have found them in practice to be stable.
Perhaps error logging if spotted breaks in block path continuity?
It borders on Command station, Layout Editor and Blocks so not sure where it'd fit.

I wasn't even sure if it was possible to do this when I started looking at this so please don't judge too harshly on code quality!!

Steve.

pabender, 2018-12-14:
@icklesteve that looks like it does just about everything required.  Cool.

I’ll play with that later and see what I can do with it ( it has to wait unfortunately. I am in the middle of some proactive hardware replacement on my development machine ).

icklesteve, 2018-12-15:
I was quite pleased with how it turned out

[image]

You never know which block is next so attached listener on every one of them :-)

No MERG CBUS simulated command station at present to grant a throttle request so you'll struggle to play with it unless you've got a physical CBUS CS.
I should have an initial sim CBUS CS ready in the next day or so.

Please please feel free to hack it into something JMRI-wide,
You'll see how this could fit together with multiple hardware types better than I can!

another todo -  move max path to value of Warrants max path in preferences

Steve

pabender, 2018-12-15:
> The order in which the block had its TD changed and became active also came into play, I'm sure there's room for optimisation here but didn't want to risk upsetting existing code.

That will come over time.

> Please please feel free to hack it into something JMRI-wide,
> You'll see how this could fit together with multiple hardware types better than I can!

I’ve started.  Hopefully I will have something interesting to show off by the end of the weekend ( though I am also trying to finish grading for a couple of classes.  That has a due date, so it is of higher priority ).

> I'm wasn't sure on how to deal with non-JMRI throttles on a JMRI system-wide basis, ie the physical handheld controller is controlling a session on a physical CS on a non-shared basis with JMRI listening in ( via session monitor / virtual throttle? ), and then processing the blocks and sending back data . . ?

The short answer is you don’t.

The longer answer is you split the system into system independent parts and system specific parts with one or more interfaces between them.  You then let the system specific implementations decide how to deal with the real hardware.

icklesteve, 2018-12-15:
Meant to say, it's also worth having a look through http://jmri.org/jython/BlockLister.py

pabender, 2019-03-05:
I am closing this via #6414.

In addition to updating the cab signal table, the code in #6414 (now merged) will send a cab signal to a Digitrax DT402 or DT500 plugged into the network.  It actually sends something every time the specified address transitions to a new block.  On blocks without signals, this means the cab signal will be blank.

I don't have any other loconet throttles available that can display cab signals.

I've got a ticket open with Digitrax trying to figure out if it is possible to send to Duplex Wireless throttles.  That doesn't currently work. I suspect a firmware upgrade will be required.

There is still some documentation required along with a little code cleanup.  Also @icklesteve is working on MERG support using the new tool.

closed 2019-03-05 by pabender