JMRI/JMRI #6303: Connection Preferences actioned immediately

icklesteve, 2018-12-15:
Long-standing issue present in 4.12

Main JMRI Preferences - Connections
With an existing connection open, choose another connection ( either manufacturer or connection ) 

Do NOT click on Save Button in preferences.
Exit the window, Preferences has unsaved changes, save them? popup with Yes / No / Cancel
Click No.

So user has not clicked saved.
User has declined saving the changes.
The menu in JMRI has not changed from ( eg. MERG ) to new hardware type.
Connection Status in main JMRI window + Help > About does NOT change to red.
Connection Name(s) in main JMRI window + Help > About do NOT change.
User assumes preferences haven't been changed as they haven't been saved.

On then opening windows ( eg. MERG Console ), many will appear to open normally however will have a nullpointer in the JMRI console for no connection type and will not communicate to the layout.

The connection can be restored within the session by re-selecting the original connection type.

All Hardware types affected.

A few potential options on improving

- Popup advising system instability
- Disabling the Connection preference taking effect immediately, only applied on re-start
- Forced re-start on change in connection(s)

Any other options?

Steve

svatoun, 2019-03-03:
As a newbie I started to hunt this issue.

During testing, I have found something similar with Z21 but that bug is XPressNet specific; I don't support it would apply to MERG as well.

The thing is that the individual `ConnectionConfig`s register themselves with system-wide managers. As you select a different connection, the former one gets `dispose`d (removed from manager etc). As part of the `dispose`, some cleanup occurs. In Z21 (XpressNet) case a field which should never been null is cleared. I think that cleanup is not necessary; as after such a change, the entire system is in a meta-state when a `ConnectionConfig` is registered, but not `configure`d. I do not fully yet understand the startup / init process, but the `ConnectionConfig` can be fully activated only on startup, it seems. It should be shutdown soon anyway, there are more things that may get screwed IMHO.

As I understood the code structure, disabling "immediate reconfiguration" is not possible without complete overhaul of the APIs, so that's not going to happen.

---

If someone more knowledgeable reads this, please correct the above if necessary; thanks.

---

Anyway, to provide better diagnostic info, please locate the `default.lcf` log confguration file in your JMRI. Open the file (any text editor), locate the line 

> log4j.appender.A1.layout=org.apache.log4j.PatternLayout

and change it into

> log4j.appender.A1.layout=org.apache.log4j.EnhancedPatternLayout

Then locate the line starting with

> log4j.appender.A1.layout.ConversionPattern

and append `%throwable` at the end; the line should read something like

 `log4j.appender.A1.layout.ConversionPattern=%d{ISO8601} %-37.37c{2} %-5p - %m [%t]%n%throwable`

This should instruct JMRI to print stacktrace to your console, post it here.

pabender, 2019-03-03:
> During testing, I have found something similar with Z21 but that bug is XPressNet specific; I don't support it would apply to MERG as well.
>
> The thing is that the individual `ConnectionConfig`s [...]

Remember that dispose is also called during tests to clean things up (especially threads) and in the specific case of the Z21 connection, there are several threads to stop.  I'll look at the PR once you have it together and see if it is OK.

Also, It isn't quite true that connections can only be started at startup.  We don't regenerate the menus after startup, but you can start a connection.  

There are at least two cases we support where it really should be possible to start a connection after the system is running:
1) In the network connection case (such as the Z21) where there is no external hardware to connect to the computer.
2) In the case of a tunneled protocol.  Pretty much right now the only places we support tunneled protocols are in XBee connections and the Z21 case.  (In the XBee case, the tunnel is to the remote serial port on an XBee device.  In the Z21 case, the protocol tunnels XPressNet and LocoNet traffic).  

The XBee tunneled connections were originally only configurable using a script.  Recent changes have made it possible to configure those using more standard connection configuration.

The Z21 tunneled connections have no separate configuration.

svatoun, 2019-03-03:
@pabender specifically for XpressNets I found a place where a field is cleared, but never reiniitalized again - except from `configure`. And code assumes it is non-null always, so after XPressNet is deselected from the config  panel, the code could fail on several places. 
I I've found `configure`s and `configureManagers` to be called from XML deserializers (only ?). What's the entry point to initiate the connection on a running system ?

BTW - if discussion / education (thanks for that !) on code structure should be held on mailing list instead of issue/PR, I'll post a question to mailing list next time.

pabender, 2019-03-03:
> @pabender specifically for XpressNets I found a place where a field is cleared, but never reiniitalized again - except from `configure`. And code assumes it is non-null always, so after XPressNet is deselected from the config panel, the code could fail on several places.

Which field are you talking about?

After dispose is called on any object, that object is no longer viable, so the fact that there is something that is set to null that other parts of the code think should be non-null isn’t necessarily an issue.

> I I've found `configure`s and `configureManagers` to be called from XML deserializers (only ?). 
What's the entry point to initiate the connection on a running system ?

You more or less have to do this from a script.

See, for examples:
https://github.com/JMRI/JMRI/blob/master/jython/XBeeSystemAutoConnectTest.py
And
https://github.com/JMRI/JMRI/blob/master/jython/XBeeSystemConnectTest.py

> BTW - if discussion / education (thanks for that !) on code structure should be held on mailing list instead of issue/PR, I'll post a question to mailing list next time.

They can go in either place...

svatoun, 2019-03-03:
eh ;) sorry to be so vague. I'll post diff next time. See `XNetSystemConnectionMemo:301`, `xt` is null-ed, but reinitialized only from the constructor. In my case (which probably does not cover the reported bug!) causes an exception to be thrown from wherever this object is  held `getThrottleManager`  and more importantly callers of `getXNetTrafficController` assume it is non-`null`. 

You've noted that menus are not updated, so (I assume) there are dangling references to the objects which are already removed from xxxManagers are floating arround and could be used - that's why I wrote the system is in a "meta-state": live part of the GUI (prefs panel) and registries see something different than the rest of the system. 

That's rather unstable and I'd personally avoid that; such situations tend to be unstable / inconsistent - remember still speaking **newbie**-generally, requirements here in JMRI may be specific and require specific solution. We are talking about a point where previously-initialized code uses X connection while the "roots" of the data now enumerate only Y connection. But still may both appear (? I got such impression) at the same level in UI and elsewhere.

From that perspective, I am really not comfortable (but there's not much to do bcs of UI and code workflow)  with `dispose` being called right after another connection is selected in the UI: the user may change his mind and select back the former object before restart. Then there are states of the ConnectXXX systems:
- uninitialized, unregistered (the other ones in the combo selection, never used really, `adapter` null)
- initialized, disposed, unregistered (the one in efffect initially, deselected)
- uninitialized registered (the one newly selected, which never went through startup sequence and `configure`
- and worst of all: initialized, disposed, re-registered; but never went through complete and tested startup sequence again after `dispose`
From that path, I'd expect that only the state which went through the startup is the really tested path. 
It's rather hard to define modes of operation for 'active' and 'inactive' (or the like) and make the above consistent with the operation modes.

As I am writing - maybe it would be possible to re-populate the config panel's combos after the ConnectionConfig has changed. Just in order to ditch the once-disposed object and replace it with a fresh (never used) one. That depends on the desired behaviour: if the user changes to Z21 from Z21 simulator (simulator gets disposed), then changes her mind and changes back to Z21 Simulator - is the system expected to continue working "as if nothing happened" ?
If so, then selection must do just the opposite to `dispose`; if `dispose` terminates threads, "something" called from selection in UI must start them. But that's not what currently `register` does.

closed 2021-04-14 by stale[bot]