JMRI/JMRI #6310: System names for signal groups are not loaded from panel files

balazsracz, 2018-12-16:
When a panel file is loaded, signal groups are loaded slightly incorrectly.
The system name stored in the panel file is ignored, and it is always replaced with an IG:AUTO:#### value during load. (I think, but would need to confirm, that already existing IG:AUTO:#### values are also ignored and replaced with a different #### value).
User names are loaded fine.

Bug was introduced Nov 2018.

Is this something that needs to be fixed before the production release? I'm not quite sure if and how the system names for signal groups are used.
I have a candidate for a fix: https://github.com/balazsracz/JMRI/commit/cbc06a66aeb62d0dabf81a84129b544da9d87d63

bobjacobsen, 2018-12-16:
Thanks!  I think this should be fixed, so thank's for tracking it down.  I'll include it in a new 4.13.7 ASAP.

But I'm confused by what went wrong.  The fix is in java/src/jmri/managers/configurexml/DefaultSignalGroupManagerXml.java but `git blame` doesn't seem to show any changes there in November 2018.  Did something else get broken?  I.e. is `newSignalGroup(..)` broken?

bobjacobsen, 2018-12-16:
I think I see what happened.  The` newSignalGroup(String )` call is a bit of a trap, because its argument is a user name, not a system name.  This is unlike just about every other `newFoo(String)` method in JMRI.  Will look at changing that...

PR #6365 (2018-12-27): Migrate newSignalGroup method to better name
See comments in #6310: The newSignalGroup method has an error-prone name, as it doesn't do what other newBeanName methods do.  This deprecates that method (name) and creates a better one to prevent future confusion.

Closes #6310
closed 2018-12-27 by bobjacobsen