JMRI/JMRI #6313: Windows uninstaller doesn't seem to work

rm5248, 2018-12-16:
Windows 10

I have several versions of JMRI installed in different folders for testing purposes.  

C:\Program Files (x86)\JMRI
C:\Program Files (x86)\JMRI-4.10
C:\Program Files (x86)\JMRI-4.13.4

The uninstaller doesn't work on any of them.  It pops up the prompt to allow for administrator approval, however clicking 'yes' doesn't do anything.  Running the Uninstall.exe from the the command line does nothing either.

If I go to the 'Apps & Features' screen in the control panel, I see an entry for JMRI 4.13.5(which is what is installed under C:\Program Files (x86)\JMRI-4.13.4).  Clicking the 'uninstall' button on here does run the uninstaller correctly.  After the installer runs, there are still around a dozen files/folders that are left over as well; it looks like some of these may have been manually added to the installation, but at least some of them appear to be part of the normal JMRI installation.  

The following files were left over:

```
C:\Program Files (x86)\JMRI-4.13.4\help
C:\Program Files (x86)\JMRI-4.13.4\jython
C:\Program Files (x86)\JMRI-4.13.4\lib
C:\Program Files (x86)\JMRI-4.13.4\resources
C:\Program Files (x86)\JMRI-4.13.4\xml
C:\Program Files (x86)\JMRI-4.13.4\help\en
C:\Program Files (x86)\JMRI-4.13.4\help\en\html
C:\Program Files (x86)\JMRI-4.13.4\help\en\package
C:\Program Files (x86)\JMRI-4.13.4\help\en\html\hardware
C:\Program Files (x86)\JMRI-4.13.4\help\en\html\tools
C:\Program Files (x86)\JMRI-4.13.4\help\en\html\hardware\can
C:\Program Files (x86)\JMRI-4.13.4\help\en\html\hardware\can\images
C:\Program Files (x86)\JMRI-4.13.4\help\en\html\hardware\can\images\CbusEventTable_963x276.png
C:\Program Files (x86)\JMRI-4.13.4\help\en\html\hardware\can\images\CbusEventTable_Columns_123x318.png
C:\Program Files (x86)\JMRI-4.13.4\help\en\html\hardware\can\images\merg-add-light-short-588x308.jpg
C:\Program Files (x86)\JMRI-4.13.4\help\en\html\hardware\can\images\merg-add-sensor-short-646x176.jpg
C:\Program Files (x86)\JMRI-4.13.4\help\en\html\hardware\can\images\merg-add-turnout-long-622x176.jpg
C:\Program Files (x86)\JMRI-4.13.4\help\en\html\hardware\can\images\MergNodeConfig.png
C:\Program Files (x86)\JMRI-4.13.4\help\en\html\hardware\can\images\SendFrame.jpg
C:\Program Files (x86)\JMRI-4.13.4\help\en\html\tools\scripting
C:\Program Files (x86)\JMRI-4.13.4\help\en\html\tools\scripting\FAQ.shtml
C:\Program Files (x86)\JMRI-4.13.4\help\en\package\jmri
C:\Program Files (x86)\JMRI-4.13.4\help\en\package\jmri\jmrit
C:\Program Files (x86)\JMRI-4.13.4\help\en\package\jmri\jmrit\operations
C:\Program Files (x86)\JMRI-4.13.4\help\en\package\jmri\jmrit\operations\images
C:\Program Files (x86)\JMRI-4.13.4\help\en\package\jmri\jmrit\operations\images\Timetable.png
C:\Program Files (x86)\JMRI-4.13.4\jython\ImportShortSensorsV8.py
C:\Program Files (x86)\JMRI-4.13.4\jython\LnScanner47R-Fix.py
C:\Program Files (x86)\JMRI-4.13.4\jython\SetStorageTurnouts.py
C:\Program Files (x86)\JMRI-4.13.4\lib\hamcrest-core-1.3.jar
C:\Program Files (x86)\JMRI-4.13.4\lib\jcommon-1.0.23.jar
C:\Program Files (x86)\JMRI-4.13.4\lib\jfreechart-1.0.19-experimental.jar
C:\Program Files (x86)\JMRI-4.13.4\lib\jfreechart-1.0.19-swt.jar
C:\Program Files (x86)\JMRI-4.13.4\lib\jfreechart-1.0.19.jar
C:\Program Files (x86)\JMRI-4.13.4\lib\jfreesvg-2.0.jar
C:\Program Files (x86)\JMRI-4.13.4\lib\junit-4.11.jar
C:\Program Files (x86)\JMRI-4.13.4\lib\orsoncharts-1.4-eval-nofx.jar
C:\Program Files (x86)\JMRI-4.13.4\lib\orsonpdf-1.6-eval.jar
C:\Program Files (x86)\JMRI-4.13.4\lib\servlet.jar
C:\Program Files (x86)\JMRI-4.13.4\lib\swtgraphics2d.jar
C:\Program Files (x86)\JMRI-4.13.4\resources\icons
C:\Program Files (x86)\JMRI-4.13.4\resources\original_icons
C:\Program Files (x86)\JMRI-4.13.4\resources\icons\misc
C:\Program Files (x86)\JMRI-4.13.4\resources\icons\WASATL Panel.JPG
C:\Program Files (x86)\JMRI-4.13.4\resources\icons\misc\Nixie
C:\Program Files (x86)\JMRI-4.13.4\resources\icons\misc\Nixie\README.txt
C:\Program Files (x86)\JMRI-4.13.4\resources\original_icons\README.md
C:\Program Files (x86)\JMRI-4.13.4\xml\SoundTraxx_Eco_Diesel_OEM_Athearn.xml
C:\Program Files (x86)\JMRI-4.13.4\xml\SoundTraxx_Tsu2_Diesel_OEM_Genesis.xml
```

For now, I've simply gone and deleted the folders manually.

dheap, 2018-12-16:
If the uninstallers fail manually remove the directories. But you also need to do the following as it's possible you have JMRI versions installed for "This User only".

- Go to Control Panel-> "Programs and Features" and keep uninstalling JMRI, refreshing (F5) every time until all are gone or none will work.
- Then navigate to "C:\Program Files (x86)\" and delete any JMRI folders if present.
- Then navigate to "C:\Users\<username>\AppData\Local\VirtualStore\" and delete any "\Program Files (x86)\JMRI" folder if present (you'll need to enable display of hidden files/folders).
- **Don't delete** "C:\Users\<username>\JMRI\". It has your data.
- Then reinstall JMRI for "All Users".

Sent from my iPad
-- 
Dave in Australia

> On 17 Dec 2018, at 6:56 AM, Robert Middleton <notifications@github.com> wrote:
> 
> I have several versions of JMRI installed in different folders for testing purposes.
> 
> C:\Program Files (x86)\JMRI
> C:\Program Files (x86)\JMRI-4.10
> C:\Program Files (x86)\JMRI-4.13.4
> 
> The uninstaller doesn't work on any of them. It pops up the prompt to allow for administrator approval, however clicking 'yes' doesn't do anything. Running the Uninstall.exe from the the command line does nothing either.

rm5248, 2018-12-17:
Dave,

Thanks, I will check that out later to ensure that I did in fact delete everything.  The installers not running unless they're run from the add/remove programs is still confusing though.

closed 2021-04-14 by stale[bot]