JMRI/JMRI #6314: When should JUnit4's @Ignore be used?

bobjacobsen, 2018-12-16:
In JUnit4, a test method (`@Test` annotated method) won't be executed during testing if it's annotated with `@Ignore("explanation")`.

There are currently 154 of these.  You can see the [history here](http://jmri.tagadab.com/jenkins/job/Development/job/Ignored%20Test%20Scan/) and a list of their [explanatory comments here](http://jmri.tagadab.com/jenkins/job/Development/job/Ignored%20Test%20Scan/lastStableBuild/warnings7Result/) (click the "details" tab)

The question is: When should `@Ignore` be used, and for what?  (We have a bit on that in our [developer pages](help/en/html/doc/Technical/JUnit.shtml); the question is what that should say now)

There are a couple of clear cases:

- The test actually runs, and the `@Ignore` is just old or mistaken:  It seems pretty clear that `@Ignore` should be removed
- The test fails because it's not complete or a not complete test.  `@Ignore` can mark that and motivate somebody to complete it so the test coverage improves:  This seems like a good use.

The complicated case is e.g. a test which has no contents, but the method needs to exist because a super class requires an implementation.  The pattern here is that `@Override` is present, or should be, and the method body is empty.  For example, the "testStateOff()" test for the EasyDCCPowerManager looks like this:
```
    @Override
     @Test
     @Ignore("no unsolicited messages, so skip test")
     public void testStateOff() throws JmriException {
     }
```

The parent class wants a testStateOff() method, but there's nothing for it to do on this protocol.  

Should that be marked with `@Ignore` or not?

One approach is to _not_ use `@Ignore`, but rather mark the empty body with a comment that says this is normal.  
```
    @Override
     @Test
     // no unsolicited messages, so nothing to test here
     public void testStateOff() throws JmriException {
     }
```

@pabender points out that's a tiny bit inefficient, as JUnit will run the `@Before`/`@After` methods around a method that's not doing anything.  But keeping @Ignore there means that people volunteering to clean up code will keep coming back to it and discovering that there's nothing that needs to be done; that wastes their time.

So what should `@Ignore` mean in cases like this? Should be we use some other (JMRI-specific or not) annotation that that has more of a ToDo meaning?

pabender, 2018-12-17:
> In the testStateOff example, can the method be overridden, but the `@Test` annotation removed, or will this cause other issues?

I did notjnow the answer, so I did an experiment. Leaving the 
`@Test` Off, with or without the `@Ignore` Causes the test to be run.

bobjacobsen, 2018-12-17:
With regard to the changes mentioned below, this is _exactly_ why I don’t think that @Ignore should have the ambiguous meaning that you prefer, if I understand correctly.

So at least we agree on that.

As to the two changes:  That’s not how I read them, but given the possibility that they meant something else, I certainly could have read them wrong.  The “needs mock pi setup”, as I read it, wasn’t correct, because it runs fine as is, without "mock pi setup”. Ditto the 2nd comment.

Again, @Ignore should be used for tests that _fail_ when run.  These don’t.

Bob

> On Dec 16, 2018, at 8:30 PM, Paul Bender <notifications@github.com> wrote:
> 
> Also, in #6309, you had actually removed ignore comments that indicated what work was required to remove the ignore, two in particular:
> 
> From /java/test/jmri/jmrix/pi/configurexml/RaspberryPiConnectionConfigXmlTest.java
> You removed
> 
> @Ignore("needs mock pi setup")
> 
> Which states explicitly the ignored test needs more setup
> 
> And in /java/test/jmri/jmrix/roco/RocoXNetThrottleTest.java
> 
> You removed several ignores of the form
> @Ignore("parent class method creates a new throttle")
> 
> Which states that the parent class test needs to be modified OR this class needs to reimplement it.
>

pabender, 2018-12-17:
Sent from my iPad

> On Dec 17, 2018, at 3:19 AM, Bob Jacobsen <notifications@github.com> wrote:
> 
> With regard to the changes mentioned below, this is _exactly_ why I don’t think that @Ignore should have the ambiguous meaning that you prefer, if I understand correctly.

I see this as being a lot like suppressing findbugs/spotbugs warnings.  I am sure we have some of those that are present that are hiding simple fixes that the person who put the suppression in place did have time to deal with or didn’t know how to deal with at the time they were put into place.

If it makes it more clear to you, let’s add an annotation that extends ignore, but means a test is not applicable for a particular class under test, and leave the Ignore in place for tests that someone needs to come back and fill in the body for.

> As to the two changes: That’s not how I read them, but given the possibility that they meant something else, I certainly could have read them wrong. The “needs mock pi setup”, as I read it, wasn’t [...]

Those particular comments are about the tests in the parent class.  

The parent class test for the Raspberry Pi DOES fail when run.  

And for the Roco Throttle tests, the parent class test doesn’t test anything in the Roco Throttle.

In both cases, those tests still need to be implemented for those particular classes. This may also be true of the EasyDCC test, but I don’t know that code as well, so I can’t say.

rhwood, 2018-12-17:
Can offer a suggestion: if the test writer knows a test must be ignored, why not require:
- an issue in GitHub for the test that can be referred to
- the @Ignore() include a reference to the issue
- that the first line of the test be an Assert.fail() with text pointing to the issue

This would require no extension and make it clear why the test is ignored and offer a space for discussion (or offer a space for discussion if the issue is not clear).

I think this has already been done in a couple of places.

Randall

pabender, 2018-12-17:
@rhwood it would certainly remove the empty test body if tests included an Assert.fail.

I don’t think all of the ignored tests that still need implemented warrant issues.

pabender, 2018-12-17:
so it turns out that you can't extend an annotation, so we would need a not applicable annotation and an ignore annotation on the method.  

The NotApplicable becomes extra documentation.  I think that's worth adding.

pabender, 2018-12-17:
I just added PR #6320 

This PR 
1) Implements the NotApplicable annotation 
2) Adds the NotApplicable annotation to tests in jmri.jmrix.roco.RocoXNetThrottleTest that are not applicable.
3) Adds an Assert.fail to  previously empty test methods in jmri.jmrix.roco.RocoXNetThrottleTest that need to be reimplemented for the class under test.

The NotApplicable annotation currently contains a value.  I'm not sure if it should.

bobjacobsen, 2018-12-17:
Let’s back up a bit.

@pabender Why are those @Ignore instances useful? I.e. What do you want people or machines to infer from them?

pabender, 2018-12-17:
@bobjacobsen I have been looking through our current usescof ignore, and I see 4 general patterns.

1) tests that have bodies that are marked as failing or intermittently failing
2) tests without bodies that indicate a parent class test doesn’t work for the class under test ( either a change in setup is required or tweaks to the test itself are required )
3) tests without bodies that indicate a parent class test doesn’t work because some change is required to the class under test
4) tests without bodies that are not currently applicable to the class under test.

I think we can all agree that cases 1,2, and 3 are always valid reasons to use an Ignore.  The ignore in all 3 cases is a reminder that that work needs to be done

In some cases, instances of 3 and 4 are the same ( because the underlying protocol may allow something that isn’t currently implemented ).

When instances of 4 are things that will never be applicable, I don’t want the machine to waste time on the test.  If this were a cucumber test or a JUnit 5 test, I would add a tag to say that the test is not applicable and just filter tests that don’t apply.  We don’t have that option in JUnit 4, but we can approximate the effect by having an ignore and an additional annotation.

Overall, I want to increase the behavior documentary value of the tests, and I think we can accomplish that with the changes I have made that appear in #6320.

Also, I really don’t like running empty tests.  Any running test with an empty body should at least have an assert added, even if it doesn’t do anything else.

bobjacobsen, 2018-12-17:
> 
> @bobjacobsen I have been looking through our current usescof ignore, and I see 4 general patterns.
> 
> 	• tests that have bodies that are marked as failing or intermittently failing
> 	• tests without bodies that indicate a parent class test doesn’t work for the class under test ( either a change in setup is required or tweaks to the test itself are required )
> 	• tests without bodies that indicate a parent class test doesn’t work because some change is required to the class under test
> 	• tests without bodies that are not currently applicable to the class under test.
> I think we can all agree that cases 1,2, and 3 are always valid reasons to use an Ignore. The ignore in all 3 cases is a reminder that that work needs to be done

No, I don’t agree. That’s my entire point.  @Ignore is _not_ a general form of @ToDo.

> In some cases, instances of 3 and 4 are the same ( because the underlying protocol may allow something that isn’t currently implemented ).
>
> When instances of 4 are things that will never be [...]

I agree that things should have documentary value. Having a crisp “This will fail if run” meaning for @Ignore is a simplification that improves its documentary value.  Happy to add an @ToDo (or @MakeThisTestBetter or whatever) if that’s useful to you for other meanings.

I agree that empty tests should be fleshed out.  We have lots of tests with no (or no meaningful) Asserts.  Having it be easy for somebody to find one and add code to it’s fine (though it rarely happens, and I suspect you and I do more almost all of that).  Comments and/or an @ToDo or similar would be a fine way to do that.  @Ignore is _not_ because it’s ambiguous; you look, and you don’t find just that; you try to track it with a plot, and you can’t sort it out.

As to performance:  Machine time and people time need to be kept in perspective.  What have ~150 @Ignores.  Will 150 (and that’s an upper limit) setUp/tearDown times take even half a second?  I’ve probably wasted many orders of magnitude more time due to these. 

I still haven’t heard a reason why your more-expansive definition of @Ignore _itself_, which is separate from every other project that I work on, will make anything _better_ (except saving less than 0.04% of the CI elapsed time which if was really important would have been done in other ways)

Bottom line:  I spend lots of hobby time trying to clean up other people’s junk.  For the last month or so, that's been over 100% percent.  You want to make that harder. I don’t think that’s a good idea.

pabender, 2018-12-17:
Sent from my iPad

> On Dec 17, 2018, at 1:39 PM, Bob Jacobsen <notifications@github.com> wrote:
> > @bobjacobsen I have been looking through our current usescof ignore, and I see 4 general patterns.
> > 
> > • tests that have bodies that are marked as failing or intermittently failing
> > • tests without bodies that indicate a parent class test doesn’t work for the class under test ( either a change in setup is required or tweaks to the test itself are required )
> > • tests without bodies that indicate a parent class test doesn’t work because some change is required to the class under test
> > • tests without bodies that are not currently applicable to the class under test.
> > I think we can all agree that cases 1,2, and 3 are always valid reasons to use an Ignore. The ignore in all 3 cases is a reminder that that work needs to be done
> 
> No, I don’t agree. That’s my entire point. @Ignore is _not_ a general form of @ToDo.

I see those first 3 uses as ALL saying something is wrong with the test that needs to be fixed before the test can be run, which is one of the things ignore is for. ( see http://junit.sourceforge.net/javadoc/org/junit/Ignore.html ).  

These differ only in where exactly the issue lies ( the test itself, setup for a test in a parent class, the object under test, etc ).  Different development cycles can produce each of these cases.    

We have a lot of code that was written before the tests, and for some of that code the exact preconditions for the tests are sometimes difficult to decipher.  When tests are inherited, this results in ignored tests that look like the second type I listed.

Using true TDD, the test always comes first, so changes to the underlying code has to follow.  When tests are inherited, this results in ignored tests that look like the third type I listed.

The 4th type is the only one where there can be any ambiguity.  That is what we need to fix.

> I agree that things should have documentary value. Having a crisp “This will fail if run” meaning for @Ignore is a simplification that improves its documentary value.

Ok, I can see that, but only for the 4th case.

> I agree that empty tests should be fleshed out. We have lots of tests with no (or no meaningful) Asserts. Having it be easy for somebody to find one and add code to it’s fine (though it rarely [...]

Ok, so let’s add that  ToDo annotation.  I’ll add that to my open PR.  The ToDo annotation really should be added to all of the ignored tests, not just the ones with an empty body.

I still think we should use a NotApplicable annotation for the cases when a base class test isn’t EVER going to be applicable.  It would be great if we could make that an extension of ignore so we could ignore tests with the single Annotation.

I will go ahead and remove the ignore from those tests marked NotApplicable.
( Keeping the not applicable annotation hopefully will aid in transitioning to JUnit 5 style filters some day )

bobjacobsen, 2019-01-20:
PR #6335 added the consensus result to the developer documentation, following up on #6334 which updated the code.

closed 2019-01-20 by bobjacobsen