JMRI/JMRI #6323: NodeIdentity no longer guarantees a unique identity per computer

rhwood, 2018-12-18:
#5242 broke the guarantee that a NodeIdentity would be as unique as the hardware the JMRI instance is running on (by requiring that the first MAC address found is part of the NodeIdentity, this guarantee worked as long as MAC addresses were not duplicated, which is a hard requirement in a non-routed network segment).

This guarantee is broken by 5242 because that PR relies on the user's home directory not being shared or copied between computers. A simple fix would be to make the NodeIdentity be `jmri-UUID-MAC-PROFILEID`, by adding the MAC back into the identity string.

dheap, 2018-12-19:
> #5242 broke the guarantee that a NodeIdentity would be as unique as the hardware the JMRI instance is running on (by requiring that the first MAC address found is part of the NodeIdentity, this [...]

It was only broken if you copy the "nodeIdentity.xml" file between computers.

Since it was also time-based, it wasn't possible for it for another instance to create a non-unique nodeIdentity, even if the MAC was identical.
 
> This guarantee is broken by 5242 because that PR relies on the user's home directory not being shared or copied between computers. A simple fix would be to make the NodeIdentity be `jmri-UUID-MAC-PROFILEID`, by adding the MAC back into the identity string.

This wouldn't fix the problem if you copy "nodeIdentity.xml" between OS partitions or Virtual Machines on the same hardware.

 OS partitions or Virtual Machines was the situation #5242 was addressing.

dheap, 2018-12-19:
> by requiring that the first MAC address found is part of the NodeIdentity

With #5242 the UUID is based on both time and first MAC address. But its a one-time calculation.

dheap, 2018-12-19:
My memory is that including the current MAC address also caused a switch of identity every time a user with a laptop changed from wireless to wired connection and part of #5242 was to solve that as well.

dheap, 2018-12-19:
The current one-time UUID is based on a 60-bit time stamp and the first-reported MAC address at the time of creation https://en.wikipedia.org/wiki/Universally_unique_identifier#Version_1_(date-time_and_MAC_address)

The only way identical nodeIdentities could be created would be if:
1. two instances with identical reported MAC addresses were able to generate a UUID within 100 nanoseconds clock time of each other.
2. The user copies "nodeIdentity.xml" between two machines.
3. Two machines share the same actual settings: files.

I don't think (1) is likely to be a problem.

Adding the currently-reported MAC address will break the [Laptop](https://github.com/JMRI/JMRI/issues/5185#issuecomment-378103244_) case (a very common occurrence for many users) and may not fix (2), which can be solved by deleting "nodeIdentity.xml" on the copy.

I'd consider (3) to be very much a corner case.

rhwood, 2018-12-19:
The NodeIdentity was written to solve #3 because it was a real issue when that class was written, and was the original use case. I have not proposed removing UUIDs, but I *really need* to be able to meet the RFC standards for DNS naming, which is broken by both how you generate the "URL-safe" (which is *not* DNS-safe) UUIDs.

Note that the current scheme for the "laptop" use case is broken *unless* I am dual booting between OSes that JMRI recognizes are different (for example using Boot Camp between two macOS instances uses the same nodeidentity.xml file, so the current scheme doesn't help that laptop user.) if I dual-boot between two Linux or macOS or Windows machines with a shared home drive (just like how Boot Camp works between macOS and Windows), so I'll keep that in mind in allowing the original use cases to be used again.

Both your laptop use case and my use case can be solved, and arguing that some use cases that were once supported shouldn't be anymore is simply being hostile.

rhwood, 2018-12-19:
(To be completely clear the common home drive scenario occurs when a flash card is used between multiple Raspberry PIs, which I think is likely more common (at least among modular model railroaders) than creating dual boot laptops).

dheap, 2018-12-19:
The previous scheme was broken for:
1. A single-boot laptop with two interfaces (WiFi and Ethernet). Many (the majority in my experience over here) model railroaders use these.
2. A dual-boot laptop.
3. Multiple OS Partitions (real or virtual) on a single computer.

All of these were everyday occurrences for me and created real problems for me. All were definitely fixed by the current scheme.

I understand your last-described use-case and the need to do something about it. But please don't break any of the above cases.

There are also existing methods to translate my URL-safe UUIDs back to the underlying standard Type 1 UUID (and hence retrieve the underlying MAC address at the time of creation). The sole purpose of the "URL-safe" version was simply to reduce path length of the resulting profile folders without sacrificing uniqueness (by packing into the minimum number of required bits).

dheap, 2018-12-19:
Note that none of my described multiple OS use-cases used the (optional) shared home drive. All used unique home drives.

In the case of the single-boot laptop (very common use-case), the first MAC address found isn't consistently the same, so the user is seemingly randomly swapped between two identities, with disparate settings.

I hope you can find a way to handle this maze of cases.

rhwood, 2018-12-19:
I extensively tested and developed the NodeIdentity on a quad NIC system and never had that break on me. The only laptop scenario I can think of that having been broken on is one where a hardware switch (not a software switch) is used to disable the WiFi NIC, in which case, the NodeIdentity was *working as designed* since removing the recognized NIC is changing the network identity of the laptop, which is the use case for which the NodeIdenty was written to solve. If you have other problems to address then please address them using other implementations that don’t hijack this use case.

rhwood, 2018-12-19:
I also don’t understand you why you keep harping on first NIC, since all NICs on the computer are enumerated to find the identity. If Java is not reporting all NICs on some OS, that bug needs to be reported to the Java developers, because that means that Java is itself fundamentally broken at this layer and we can’t rely on any mechanism to uniquely identify computers.

dheap, 2018-12-19:
My memory of the previous behaviour was that when multiple NICs were found it created multiple identities and seemingly randomly swapped around between them.

If you code to avoid that happening with your "added current MAC" proposal, that would be great. I believe we still need the generated-UUID part to differentiate between non-shared home folders on the same hardware and it shouldn't affect your shared home folders.

dheap, 2018-12-19:
@pabender It seems I didn't explain clearly. My expectation is that:

Case 1: Should always have the same nodeIdentity.

Case 2: Should always have differing nodeIdentities.

Case 3: Should always have differing nodeIdentities.

That's agreeing with you, not disagreeing.

I take @rhwood 's statement about the RaspberryPi. We need to solve that.

One quite simple way would be to extract the original MAC address that seeded the UUID (easily done) and if it can't be found on the current hardware, declare the UUID invalid (replicating the first-use case for all users when my code was merged. This then creates a new nodeIdentity and takes care of copying from the previous identify.

But it should be messy if you regularly swap that particular JMRI instance on an SD card between different RaspberryPi hardware as it would create a new copy of the identify folder. You could prevent that by scanning previous identities for one based on a matching NIC and reverting to it.

The seed NIC can always be extracted from any the name of nodeIdentity created under my existing code.

dheap, 2018-12-19:
But it's very late (early in the morning) here, I'm typing on an old iPad where editing is difficult and I need to get some sleep...

dheap, 2018-12-19:
Ecuse all the typos...

KenC57, 2018-12-19:
If the pi image has the right file(s) removed, will JMRI correctly populate
a new value from the pi it is plugged into? I've got an image around, what
should I look to 'not' find that this isn't an issue? Or what can I tell
from the value on a running pi system that it was generated for that pi?
-ken c

bobjacobsen, 2018-12-19:
> I also don’t understand you why you keep harping on first NIC, since all NICs on the computer are enumerated to find the identity. If Java is not reporting all NICs on some OS, that bug needs to be [...]

Not following this full thread, sorry, but the NICs on my laptop are _not_ constant.  They come and go depending on location, etc.  But perhaps I’ve misunderstood….

pabender, 2018-12-19:
Sent from my iPad

> On Dec 19, 2018, at 12:51 PM, Bob Jacobsen <notifications@github.com> wrote:
> Not following this full thread, sorry, but the NICs on my laptop are _not_ constant. They come and go depending on location, etc. But perhaps I’ve misunderstood….

@bobjacobsen,  are you saying the Media Access Control (MAC) address of the WiFi or Ethernet Network Interface Card (NIC ) changes based on location?  This is a MacBook of some kind right?

The MAC address is built into the hardware, so it really shouldn’t change, though some NICs do allow you to assume the address of another device ( this feature is primarily used for high availability servers and routers.  I have never seen it used on devices intended for everyday computing ).

Networks do sometimes come and go with location, but that doesn’t mean the MAC address changes.

mattharris, 2018-12-19:
On my laptop, I get additional MAC addresses depending on which docking station I'm plugged into at the time. In addition, depending on the location, there are virtual NICs created to tunnel out via proxy servers to the internet.

All of this provides a less than stable NIC situation on my machine...

bobjacobsen, 2018-12-19:
> @bobjacobsen, are you saying the Media Access Control (MAC) address of the WiFi or Ethernet Network Interface Card (NIC ) changes based on location? This is a MacBook of some kind right?

Yes, it’s a MacBook Pro.

At various locations, I plug into hardware that provides additional network interfaces. 

Also, in some cases a virtual NIC appears that provides “local” connection to my lab network.  (Typically when on unsecure WiFi)

pabender, 2018-12-19:
Sent from my iPad

> On Dec 19, 2018, at 1:50 PM, Bob Jacobsen <notifications@github.com> wrote:
> 
> 
> > @bobjacobsen, are you saying the Media Access Control (MAC) address of the WiFi or Ethernet Network Interface Card (NIC ) changes based on location? This is a MacBook of some kind right?
> 
> Yes, it’s a MacBook Pro.
> 
> At various locations, I plug into hardware that provides additional network interfaces. 
> 
> Also, in some cases a virtual NIC appears that provides “local” connection to my lab network. (Typically when on unsecure WiFi)

Ok, that is different than the situation I thought you were describing.

In both of these cases you are adding additional network interfaces, and therefore additional NICs ( with their own MAC addresses.)  The hardware inside your laptop doesn’t actually change.

bobjacobsen, 2018-12-20:
> In both of these cases you are adding additional network interfaces, and therefore additional NICs ( with their own MAC addresses.) The hardware inside your laptop doesn’t actually change.

The virtual NIC case can happen while the laptop isn’t connected to additional hardware.  For example, this one is a virtual connection that’ll probably disappear when I get off the insecure WiFi I’m now using:

en7: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	ether 78:4f:43:98:c9:83 
	inet6 fe80::10a5:3b31:5fa5:b853%en0 prefixlen 64 secured scopeid 0xa 
	inet6 2607:f140:400:a006:1490:7600:ec4f:126 prefixlen 64 autoconf secured 
	inet6 2607:f140:400:a006:b971:c3eb:46d3:8d31 prefixlen 64 autoconf temporary 
	inet 10.142.27.64 netmask 0xfffffc00 broadcast 10.142.27.255
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect
	status: active

On the other hand, JMRI doesn’t have a way to know what’s “additional” hardware (i.e. is external now, but might go away later) and what’s inside the computer.  They all look pretty much the same. For example, which of these is inside my laptop?

en6: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	ether ac:de:48:00:11:22 
	inet6 fe80::aede:48ff:fe00:1122%en6 prefixlen 64 scopeid 0x9 
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect
	status: active

en9: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=10b<RXCSUM,TXCSUM,VLAN_HWTAGGING,AV>
	ether 0c:4d:e9:ae:c7:22 
	inet6 fe80::1810:986f:c5a0:b29c%en9 prefixlen 64 secured scopeid 0x17 
	inet 128.32.235.106 netmask 0xffffff80 broadcast 128.32.235.127
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect (1000baseT <full-duplex,flow-control,energy-efficient-ethernet>)
	status: active

I think the more general point is that “same computer” != “same set of network hardware”; network hardware can change with location, at the very least.

dheap, 2018-12-20:
I'm trying to do some testing of this PR, but have been handicapped by family responsibilities and a balky VM. Family responsibilities will be heavy the next week and I'll also have somewhat limited (intermittent) access to my main Dev machine.

pabender, 2018-12-20:
> In both of these cases you are adding additional network interfaces, and therefore additional NICs ( with their own MAC addresses.) The hardware inside your laptop doesn’t actually change.
> The [...]

Right, virtual networks are defined completely in software.  

> I think the more general point is that “same computer” != “same set of network hardware”; network hardware can change with location, at the very least.

I never made that claim.  It’s pretty simple to demonstrate that with USB NICs.  

My concern, and I believe @rhwood’s as well, is that there appears to be a case where different computers ( with different MAC addresses ) can end up with the same UUID because the current UUID code doesn’t regenerate UUIDs when it should.

bobjacobsen, 2018-12-20:
> I never made that claim. It’s pretty simple to demonstrate that with USB NICs.

My initial post was a reply to this by another author:

> I also don’t understand you why you keep harping on first NIC, since all NICs on the computer are enumerated to find the identity. If Java is not reporting all NICs on some OS, that bug needs to be [...]

You can’t rely on using _all_ NICs to uniquely identify computers, because it'll think one machine is a different one at different times.

You can’t rely on a _single_ NIC because one piece of hardware might be connected to different machines (your USB example).

On a Mac, the kern.uuid parameter is unique and persistent:

kern.uuid: 5D53F7E4-472A-369D-97D8-4DD877A4BDFF

But getting it from Java would involve running 'sysctl kern.uuid’ in a shell script AFAIK.

bobjacobsen, 2018-12-20:
Maybe I don't understand how this is supposed to work.

Below are the profile directories in four profiles that I use.  They're used in two hardware-related locations, plus on airplanes, etc.  The first three have hardware information in them (C/MRI and/or NCE connections); the 4th is simulator only.

These four profiles have a _lot_ of different IDs in them.  Yet it's only one 'computer', at least as I understand the intention of the NodeID feature.  What's going wrong? Do I understand the intention wrong?

```
profiles/DecoderPro_on_site/profile:
jmri-F8BC1264EF35-3f71df8d
jmri-pacUab9nWhNiaa8lVszo6--3f71df8d

profiles/Shasta_Ops/profile:
jmri-F8BC1264EF35-3e1cd780
jmri-pacUab9nWhNiaa8lVszo6--3e1cd780

profiles/Shasta_RS422/profile:
jmri-784F4398C983-3f0a6c35
jmri-irOPFcs4GhNiaaDe7dLmLd-3f0a6c35

profiles/Shasta_Simulator/profile:
jmri-6E709F079A4C-3e3bd73b
jmri-784F4398C983-3e3bd73b
jmri-92FD61C1C87D-3e3bd73b
jmri-C82A145014E4-3e3bd73b
jmri-d6HikdZ-GhNiaaDe7dLmLd-3e3bd73b
jmri-hLIolb_WbhNiaaDe7dLmLd-3e3bd73b
jmri-irOPFcs4GhNiaaDe7dLmLd-3e3bd73b
jmri-klnw8bR4bhNiaaDe7dLmLd-3e3bd73b
jmri-pejzCbR1bhNiaaDe7dLmLd-3e3bd73b
```

dheap, 2018-12-20:
I'm not sure where this leaves us.
1.  Does this mean you are abandoning #6311 so further testing is unwarranted? My testing so far with my use cases was positive.
2. If you are going to revert #5242 so Shared Profiles are broken for dual-boot and VM systems without providing a fix, I would be very concerned about that. It would have a big impact on my development workflow.

dheap, 2018-12-20:
>@dheap It leaves us with more than one class for uniquely identifying a computer, using different techniques, because the use cases are, I think, incompatible.

@rhwood I may be slow this morning, but I can't interpret your intentions from this. Are you saying that:
- You will return the NodeIdentity class to be NIC/network identify  only.
- #6311 will be withdrawn, or significantly changed.
- You will work on a "computer" identify so "per computer" preferences will be moved from NodeIdentity to a new "ComputerIdentity" class (or possibly multiple classes.

Is this correct? If so, I won't spend any more time testing #6311. I intended to do so today.

rhwood, 2018-12-20:
I have not described how I may or may not accomplish this. #6311 *will not* be withdrawn--it contains important fixes, but is broken by this.

Pugwash1, 2018-12-20:
To save energy, why don't we just use the hostname and tell people that if they need to save preferences by machine, the machines must have different hostnames.
If you do develope the perfect solution, don't tell us, sell it to the highest bidder..... It's worth a fortune.

rhwood, 2019-02-17:
@dheap Can you test the current state of #6311?

dheap, 2019-02-17:
@rhwood I'll try to take a look at it tomorrow"

dheap, 2019-02-18:
@rhwood I did some pretty extensive testing this last year. I've made a start redoing the tests with the current state of #6311.

It'll take a few days as I have family and personal commitments.

rhwood, 2019-02-18:
@dheap Thank you. Just so you are aware, the code in for identities in #6311 is different than it was last year.

dheap, 2019-02-18:
> Just so you are aware, the code in for identities in #6311 is different than it was last year.

Yes, I very quickly worked that out. It's looking good with my tests so far...

dheap, 2019-02-21:
@rhwood Testing completed, see https://github.com/JMRI/JMRI/pull/6311#issuecomment-465922606

dheap, 2019-03-23:
@rhwood Does #6311 resolve this?

rhwood, 2019-03-23:
Yes, It does.

closed 2019-03-23 by rhwood