JMRI/JMRI #6337: Error opening profile created in PanelPro 4.10

rhwood, 2018-12-21:


bobjacobsen, 2018-12-22:
That file says it was written by JMRI 4.0 with schema layout-2-9-6.xsd 

multipleChoice is defined in schema xml/schema//auxiliary-configuration/class-preferences-4-3-5.xsd which postdates JMRI 4.0, probably when the jmri.managers.JmriUserPreferencesManage class took over the duties of handling this.

In JMRI 4.0 (as also referenced in that file), this `<UserMessagePreferences>` information was read and written by the jmri.managers.configurexml.DefaultUserMessagePreferencesXml class

So it looks like something went wrong in the schema migration.

bobjacobsen, 2018-12-22:
Or maybe we didn't use to validate profile information.

rhwood, 2018-12-22:
It is an older profile. Wonder of we removed some code later that used to handle this, since this code is in the pre-per-computer UI Preferences format.

rhwood, 2018-12-22:
I've traced this down to a scenario where a migration from old-style preferences to per-computer preferences for the user interface is occurring.

closed 2021-04-14 by stale[bot]