JMRI/JMRI #6338: Messages in Error opening profile dialog cannot be cut and pasted.

rhwood, 2018-12-21:
In JMRI 4.10, the error messages opening a profile could be selected and copied to the clipboard (albeit with keyboard shortcuts only), adhered to the various Human Interface Guidelines (by using error indicators and being a dialog instead of a modal application window), and showed error hierarchies (can't connect is the error, it is caused by the indented can't find serial port):
[image]

In 4.12 and current master, the messages can no longer be selected or copied, the dialog does not adhere to those guidelines (its not a dialog as presented by the system and has no error indicator), and error messages are all centered, so no relationship between the two errors is available (although the button choice is an improvement):
[image]

rhwood, 2018-12-21:
Also "Start with new profile" should probably be "Change profile" and lead to the profile selector, which has a "New profile" button on it.

danielb987, 2018-12-22:
@rhwood 
Can you check if #6340 fix the first part?

I have not done "Change profile" yet. That would need some more work so I think it's good to merge #6340 first.

danielb987, 2018-12-22:
> could be selected and copied to the clipboard (albeit with keyboard shortcuts only)

Would it be useful with an extra button "Copy to clipboard"?

rhwood, 2018-12-22:
No. A usable context menu would be useful.

danielb987, 2018-12-22:
What menu items would that menu have?

I do `JOptionPane pane = new JOptionPane(...)` and then create a JDialog from that so I might add a context menu to that pane.

rhwood, 2018-12-22:
A context menu is a right click menu on a JTwxtField. I’d seriously go back to the old panel, including the table (or list) of error messages instead and add the extra buttons. It already allowed the selection and copy, but needed the (per selectable item) context menu.

rhwood, 2018-12-22:
> The code that was before I added the option to edit connections was this:
>
> JOptionPane.showMessageDialog(null,
> 	new Object[]{
> 	    (list instanceof JList) ? Bundle.getMessage("InitExMessageListHeader") : null,
> 	    list,
> 	    "<html><br></html>", // Add a visual break between list of errors and notes // NOI18N
> 	    Bundle.getMessage("InitExMessageLogs"), // NOI18N
> 	    Bundle.getMessage("InitExMessagePrefs"), // NOI18N
> 	},
> 	Bundle.getMessage("InitExMessageTitle", Application.getApplicationName()), // NOI18N
> 	JOptionPane.ERROR_MESSAGE);
> (new TabbedPreferencesAction()).actionPerformed();
> That code still exists and is used if there is another error than a connection error. It's more or less that code that I use now. I don't see how I can change that code to add a context menu.

Add it to the JList named `list`. See https://docs.oracle.com/javase/7/docs/api/javax/swing/JComponent.html#setComponentPopupMenu(javax.swing.JPopupMenu)

danielb987, 2018-12-22:
OK. What PopupMenu should I add? Is there any menu I can use, or should I create a new one and if so, with what menu items?

bobjacobsen, 2019-01-08:
See PR #6340. Should that close this?

rhwood, 2019-01-08:
Yes, it should close this.

closed 2019-01-09 by bobjacobsen