JMRI/JMRI #6347: LocoNet fast clock should fully implement master mode

pabender, 2018-12-23:
@pabender,
Part of the desired functionality is supported already.  JMRI can "publish" fast time to LocoNet.  

But it does _not_ reply to any "LocoNet request for current fast time".  In other words, when a "LocoNet Fast Clock" sends its periodic "what time is it now" (slot 0x7B read) message, JMRI does not reply.

I tested the above via a "LocoNet Simulator connection".

In the JMRI "Tools" -> "Clocks" -> "Setup fast clock..." window,  I set "Time Source" to "Internal Computer Clock", checked "Synchronize Internal Fast Clock and LocoNet Fast Clock", then set a time and rate, and activated the "Apply" pushbutton.  Then the LocoNet Monitor shows frequent writes to the Fast Clock slot, representing fast clock changes every fast minute.

I then used the "Send LocoNet Packet" tool to send a slot 0x7b read request.  JMRI did not reply with a "fast clock report" (0xe7 0x0e 0x7b ...) message.

Perhaps JMRI's LocoNet features should use the combination of "Internal Computer Clock" and "Synchronize Internal Fast Clock and LocoNet Fast Clock" to _imply_ that JMRI is the LocoNet fast clock master.  Perhaps JMRI LocoNet code should perhaps be made sensitive to Fast Clock slot reads when that definition of JMRI as LocoNet fast clock master applies.  Perhaps JMRI LocoNet code should respond to those Fast Clock slot reads.  

Perhaps those optimizations should apply _only_ when JMRI is configured for a command station which _does not_ act as a fast clock master.  Since a command station will, by definition, reply faster than JMRI can, if both a fast-clock capable command station and JMRI are acting as "fast clock master", then there will be conflicting LocoNet messages regarding fast clock time.  This can potentially cause user-visible jumps in a fast clock display.

_Originally posted by @devel-bobm in https://github.com/JMRI/JMRI/issues/6256#issuecomment-445834221_

pabender, 2018-12-23:
See also ( from #6256 )
https://github.com/JMRI/JMRI/issues/6256#issuecomment-445850387
https://github.com/JMRI/JMRI/issues/6256#issuecomment-445872699
https://github.com/JMRI/JMRI/issues/6256#issuecomment-445955165

bobjacobsen, 2018-12-23:
> Perhaps those optimizations should apply only when JMRI is configured for a command station which does not act as a fast clock master. 

This is the key point.

1) When JMRI is used with a time-providing command station, it can’t be the master.  JMRI doesn’t know how to disable the responder in the command station.  So it instead works pretty hard to keep that clock accurate, then lets it respond.  AFAIK, this works fine.

2) When there isn’t a command station, or the command station doesn’t provide time responses, then JMRI could provide them instead.   That code isn’t here AFAIK.

Since JMRI already has the command station type configured in, that could be used to select between these two modes.

Pugwash1, 2018-12-24:
See my last comments on #6259 for how to implement master and slave for Jmri fast clock on loconet.
My dcs240 works fine for fast clock master.

pabender, 2018-12-24:
My use case is for what is, effectively a standalone LocoNet network.

IF it is easy to implement that for the standalone LocoNet, then getting this to work for other systems is just a matter of deciding when it should be allowed.

danielb987, 2018-12-26:
>  But that means that JMRI may have to be a bit "crafty" in order to figure if/when to act as a Fast Clock master.

Is it possible for JMRI to send "LocoNet request for current fast time" and see if it gets a reply? If it does, there is a fast clock. If not, JMRI must support the fast clock. JMRI could send a couple of messages, for example reading slot 0 to check that the master is alive before decide whether to be a clock master or not.

devel-bobm, 2018-12-26:
@Pugwash1,  I am curious to know your DCS240 OpSw settings.  I wonder if an OpSw can control DCS240 fast clock mastership, and I wonder if I've corrupted my OpSw settings, since my DCS240 does not act as master, even though it is set for "factory default" settings.  A JMRI "Roster" entry for the DCS240 might be the easiest way to get all of the data.

devel-bobm, 2018-12-26:
@danielb987, JMRI probably can request the current fast time to see if a master replies.  It would be my preference to avoid doing so, if possible.  With some 3rd-party fast clocks, requesting the current fast time can result in what I'll call "aggressive clock advance" to the next minute (regardless of the "sub-minute" counter).
I'm curious about your suggestion of reading slot 0.  Is that operation simply looking to see if there is  a "LocoNet Master" connected to LocoNet, or is there something specific about fast clock mastering available in slot 0 read data?  If the former, remember that a functioning LocoNet may have a "Fast Clock Master" even though it does not have a functioning "LocoNet Master".

danielb987, 2018-12-26:
> I'm curious about your suggestion of reading slot 0.

JMRI could be connected to the layout without being able to connect to the Master. If I disconnect the LocoBufferUSB, JMRI will not be able to open the serial port. If I connect the LocoBufferUSB to the computer, JMRI will see the serial port and be able to "connect" to the layout. But the Master could still be offline, for example if I had disconnected the power to the Digitrax Zephyr that I use. Reading slot 0 could be one way to actually verify that there is a master connected.

If I start JMRI first, and then power up the Master, JMRI may think there is no clock master since it made its decision too soon.

The Master could be there, but currently not available (possible turned off).

Pugwash1, 2018-12-26:
@devel-bobm DCS240 HWver 0.0 SWver 0.3. Serial 488 
08:35:21.666: [BB 7F 00 3B]  Request Command Station OpSwitches (or DCS210/DCS240 check for multiple command stations on LocoNet).
08:35:21.676: [E7 0E 7F 00 03 38 02 47 01 00 00 1C 68 62]  Report of current Command Station OpSw values:
	OpSw1=Thrown, OpSw2=Thrown, OpSw3=Thrown, OpSw4=Thrown, OpSw5=Thrown, OpSw6=Thrown, OpSw7=Thrown, OpSw8=Thrown,
	OpSw9=Closed, OpSw10=Closed, OpSw11=Thrown, OpSw12=Thrown, OpSw13=Thrown, OpSw14=Thrown, OpSw15=Thrown, OpSw16=Thrown,
	OpSw17=Thrown, OpSw18=Thrown, OpSw19=Thrown, OpSw20=Closed, OpSw21=Closed, OpSw22=Closed, OpSw23=Thrown, OpSw24=Thrown,
	OpSw25=Thrown, OpSw26=Closed, OpSw27=Thrown, OpSw28=Thrown, OpSw29=Thrown, OpSw30=Thrown, OpSw31=Thrown, OpSw32=Thrown,
	OpSw33=Closed, OpSw34=Thrown, OpSw35=Thrown, OpSw36=Thrown, OpSw37=Thrown, OpSw38=Thrown, OpSw39=Thrown, OpSw40=Thrown,
	OpSw41=Thrown, OpSw42=Thrown, OpSw43=Thrown, OpSw44=Thrown, OpSw45=Thrown, OpSw46=Thrown, OpSw47=Thrown, OpSw48=Thrown,
	OpSw49=Thrown, OpSw50=Thrown, OpSw51=Thrown, OpSw52=Thrown, OpSw53=Thrown, OpSw54=Thrown, OpSw55=Thrown, OpSw56=Thrown,
	OpSw57=Thrown, OpSw58=Thrown, OpSw59=Closed, OpSw60=Closed, OpSw61=Closed, OpSw62=Thrown, OpSw63=Thrown, OpSw64=Thrown.
08:35:35.040: [BB 7B 00 3F]  Request Fast Clock information.
08:35:35.051: [E7 0E 7B 04 68 7D 4E 47 75 18 70 68 03 03]  Response Fast Clock is  Running, rate is 4:1. Day 24, 13:11. Last set by ID 0x03 0x68 (488).
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available.
08:35:36.928: [BB 00 00 44]  Request data/status for slot 0.
08:35:36.938: [E7 0E 00 00 00 00 00 47 00 00 00 00 00 51]  Report of slot 0 information:
	Loco 0 (short) is Not Consisted, Free, operating in 28 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x00 0x00 (0).

devel-bobm, 2018-12-26:
@Pugwash1, thanks for the OpSw1-OpSw63 info.  Can you provide OpSw64-OpSw128, too?  That comes from a read of slot 0x7E, which can be gotten by sending "BB 7E 00 3A".

devel-bobm, 2018-12-26:
@danielb987, reading slot 0 would solve the problem for the case where the command station is fast clock master and turned off when JMRI is started.  But it would not solve the problem where the command station is not the fast clock master and where the actual fast clock master is not currently available.

If we have user-configurable settings which allow JMRI to act as the LocoNet fast clock master, then JMRI should do what it is told, regardless of presence of any other fast clock master on LocoNet.  

Perhaps JMRI sould attempt to identify the existence of any additional LocoNet fast clock masters (beyond its own) on LocoNet and inform the user when identified.  This would allow the user to make the necessary configuration changes.

KenC57, 2018-12-26:
Might the simple way be to have JMRI do what it is configured for, but to
log/alert if a conflict is found? Similar to the command station beeps when
it thinks there is a second command station on the network. Trying to do too
much to 'test first' can significantly slow down startup and as pointed out,
doesn't do well when everything isn't online at JMRI startup.
-ken c

Pugwash1, 2018-12-26:
@devel-bobm  Request Extended Command Station OpSwitches (DCS210/DCS240 only).
Report of current Extended Command Station OpSw values:
	OpSw65=Closed, OpSw66=Thrown, OpSw67=Thrown, OpSw68=Thrown, OpSw69=Closed, OpSw70=Thrown, OpSw71=Thrown, OpSw72=Thrown,
	OpSw73=Thrown, OpSw74=Thrown, OpSw75=Thrown, OpSw76=Thrown, OpSw77=Thrown, OpSw78=Thrown, OpSw79=Thrown, OpSw80=Thrown,
	OpSw81=Thrown, OpSw82=Closed, OpSw83=Thrown, OpSw84=Thrown, OpSw85=Thrown, OpSw86=Closed, OpSw87=Thrown, OpSw88=Thrown,
	OpSw89=Thrown, OpSw90=Thrown, OpSw91=Thrown, OpSw92=Thrown, OpSw93=Thrown, OpSw94=Thrown, OpSw95=Thrown, OpSw96=Thrown,
	OpSw97=Closed, OpSw98=Closed, OpSw99=Thrown, OpSw100=Thrown, OpSw101=Closed, OpSw102=Closed, OpSw103=Thrown, OpSw104=Thrown,
	OpSw105=Thrown, OpSw106=Thrown, OpSw107=Thrown, OpSw108=Thrown, OpSw109=Thrown, OpSw110=Thrown, OpSw111=Thrown, OpSw112=Thrown,
	OpSw113=Thrown, OpSw114=Thrown, OpSw115=Closed, OpSw116=Thrown, OpSw117=Thrown, OpSw118=Thrown, OpSw119=Closed, OpSw120=Thrown,
	OpSw121=Thrown, OpSw122=Thrown, OpSw123=Thrown, OpSw124=Thrown, OpSw125=Thrown, OpSw126=Thrown, OpSw127=Thrown, OpSw128=Thrown.

Pugwash1, 2018-12-28:
OK, this is as far as I have got.
I can use the internal clock and ignore the rest of the world. (Status Quo)
I can use the internal clock and push corrections out to the CS on every new FCmin. (Status Quo)
I can use the LocoNet Fast Clock, CS or 3rd party as master, and accept it as gospel.
I can be Master of the LocoNet Fast Clock and stomp out or trump other fast clocks if they exist.
Is it important that a DT xxx throttle be able to change the clock when I am Master? Currently I ignore.
The current set-up GUI doesn't make it easy to set up and the interface isn't Loconet senistive.
The current set-up GUI starts with clock on if you ask for it to start stopped and vice versa.
Just cleaning up code, should have something in the next couple of days.
@devel-bobm Have you found a sure fire way of killing the CS fast clock? I get very mixed results across the DCS50/51/240.

pabender, 2018-12-28:
> OK, this is as far as I have got.
> I can use the internal clock and ignore the rest of the world. (Status Quo)
> I can use the internal clock and push corrections out to the CS on every new FCmin. [...]

Great, that will do what I really need.

> Is it important that a DT xxx throttle be able to change the clock when I am Master? Currently I ignore.

I think eventually we should respond appropriately when another device sets the time.  Consider, for example, a user that has a standalone LocoNet with one or more Logic Rail fast clock(s) ( https://www.logicrailtech.com/lnfc.htm ). That user should be able to adjust the rate and time.

That said, that isn’t as high a priority as giving JMRI the ability to act as the fast clock master, so it isn’t something that should hold back the code.

devel-bobm, 2018-12-28:
@Pugwash1,

No, I have not found any way to control whether a command station acts as a fast clock master.

devel-bobm, 2018-12-28:
If I recall correctly, one problem I have seen (more than 10 years ago) with Logic Rail Technology "LocoNet Fast Clock" devices is that when JMRI sets the fast clock time, they tend to be off by one minute.  

I recall thinking that this was simply a problem where the LNFC device improperly handles the "sub-minute" information in the fast clock write LocoNet message.  Or it was related to the layout owner's _requirement_ that one LNFC must be configured to act as "fast clock master" even though the DCS100 has a "fast clock master" that could not be disabled.  This "opertational feature" was so annoying that I stopped using JMRI to set or display fast clock time, since accurate time is a _requirement_ when running under timetable/train order.  I never bothered to "run the issue to ground".

Pugwash1, 2018-12-28:
@devel-bobm I don't have a Logic Rail Technology "LocoNet Fast Clock" (LRTFC) , so I cant say.
Using multiple dt40x, dt500's the jmri fast clock and the throttle fast clocks stay  in agreement with each other regardless of who is the master of the LocoNet clock. Using the "Correct" feature can cause drift as that uses an 0xEF message to set the command station clock, but the DTs ignore 0xEF and wait for the CS / Master to send an 0xE7, then they both agree again. Any change in the fast clock settings takes up to 2 minutes on my test setup, to reach all DT clocks without JMRI and up to 55 seconds using the "new" JMRI FC. Which should be ready for a spin this afternoon.
There is a note in the code about not sending updates at zero minute because it upsets LRTFCs, but not sending it  results in the CS taking control again and the CS sends the zero minute! So I don't suppress the zero minute send.
In the new code just having the JMRI fast clock blindly track the LRTFC master will work.

devel-bobm, 2018-12-29:
I have confirmed that JMRI/head code for loconet fast clock generates fast clock messages that result in a LRT-LNFC display that is one minute behind throttles and JMRI.  I believe this is a problem with the LRT-LNFC.  I have not been able to investigate if there is a way to "cheat" the message so that the LRT-LNFC displays consistently with throttles and JMRI.

KenC57, 2018-12-29:
Bob M: Is that offset on a standalone LocoNet? IMO trying to make JMRI a
master when there is any Digitrax command station (which we can keep in sync
with) isn't meaningful work.
-ken c

Pugwash1, 2018-12-29:
> ---------- Original Message ----------

Ken:
Actually making JMRI a LocoNet master clock even with a CS on the LocoNet is
meaningful work.
How you take control is to broadcast every 50 sec (less than 70 sec according to
Hoyle), this suppresses queries ,  and respond to certain requests.
Even if a master responds, we respond straight after and everything resets to
us.
Its the same code for standalone or with CS attached so its no extra work.
Steve G.

devel-bobm, 2018-12-29:
@KenC57 , I was working with a Command Station on the LocoNet when I saw the LRT-LNFC being off by one minute.  I do not _recall_ seeing any command-station-generated LocoNet messages in response to the ones sent by JMRI.  (I wasn't looking carefully and didn't log my experiments).  I will be able to repeat my experiment, and to check operation on a "standalone" loconet with LRT-LNFC in a couple of days.

@Pugwash1,
Your statement about "overriding the command station" is only correct if the fast-clock implementation does not run a mechanical clock (i.e. Q80 "analog" clock mechanism).  Conflicting "time setting" or "time reporting" messages could cause chaos for such a mechanical fast-clock.

pabender, 2018-12-29:
> On Dec 29, 2018, at 8:40 AM, Bob Milhaupt <notifications@github.com> wrote:
> I have confirmed that JMRI/head code for loconet fast clock generates fast clock messages that result in a LRT-LNFC display that is one minute behind throttles and JMRI. I believe this is a problem with the LRT-LNFC. I have not been able to investigate if there is a way to "cheat" the message so that the LRT-LNFC displays consistently with throttles and JMRI.
> 
So this is with the CS acting as command station right?

Does the LRT-LNFC do the same thing with JMRI acting as master ( code in PR #6371 )?

Pugwash1, 2018-12-29:
@devel-bobm @pabender 
After an afternoon of testing we observed that the DT40x, Dt500 throttles were always behind their time by the amount of the fracMin. In the github code for loconet android it notes that digitrax throttles were sometimes off by a minute. Now as far as we could tell  the digitrax thottles reset their fracMins to 0 (zero) when they received an E7 message, not to the fractional minute, as documented at the bottom of page 14 in the LNPE1, "Upon seeing a SYNC read, all devices should reset their local sub-minute phase counter". In other words this is a whole minute message not a sub minute message. So I tried just that and it worked, JMRI in sync with loconet, loconet insync with JMRI. The key is ignore the frac minutes, so when we are master, broadcast the time at the whole minute fast clock boundary. That also keeps home grown stuff in sync. whether or not they process the fracMin.
The code I will push up tomorrow will have this in, and hopefully a fix for how the clock starts: stopped, running or current time.

devel-bobm, 2018-12-31:
I've done a little investigation this am with a "true standalone" LocoNet and found some confusing behaviors.

- the "Correct hardware clock" item is back on JMRI "Fast Clock Setup".  I haven't done a pull since my previous experiments.  Does this GUI item perhaps disappear for connections without LocoNet hardware?

- "Who's sending periodic fast clock slot read report messages on LocoNet when JMRI is _not_ configured for 'synchronize internal fast clock and LocoNet fast clock".  It is not JMRI.  There is no command station.  The LRT-LNFC is configured as a "slave".  So far as I know, the throttle cannot broadcast fast clock slot read reports.  Is the PR3 perhaps sending those messages?  The throttle?  The LRT-LNFC?

- And "a while ago, the LRT-LNFC was behind by a fast-minute; now it is in sync".  Why?  What changed?

- Does the act of allowing JMRI to "set" the fast clock  (by checking the "Synchronize" checkbox then unchecking it) cause the LRT-LNFC to be one minute behind?  Why?

- If I check "Synchronize", I get two identical LocoNet fast clock slot write messages.  JMRI sends both of these.  Why two identical messages?  One for "rate" and one for "time"?  If I uncheck it, I do not get any LocoNet fast clock slot messages.

- Does the occurrance of two identical LocoNet fast clock write messages specifically cause the LRT-LNFC to display fast time that is one minute behind the time displayed by the DT400R?  **No.**  Sending a single message (instead of two), using JMRI's "Send LocoNet Packet" tool,  still results in a one-minute offset on the LRT-LNFC display.

Pugwash1, 2018-12-31:
> the "Correct hardware clock" item is back on JMRI "Fast Clock Setup". I haven't done a pull since my previous experiments. Does this GUI item perhaps disappear for connections without LocoNet hardware?
Possibly, Its there if you have MRC or loconet.

> "Who's sending periodic fast clock slot read report messages on LocoNet when JMRI is not configured for 'synchronize internal fast clock and LocoNet fast clock". It is not JMRI. There is no command [...]

If the LRT-LNFC is a slave it will query the fast slot  every so often if it hasn't seen the e7 message recently. DTxxx throttles also do this. 

> If I check "Synchronize", I get two identical LocoNet fast clock slot write messages. JMRI sends both of these. Why two identical messages? One for "rate" and one for "time"? If I uncheck it, I do not get any LocoNet fast clock slot messages.

This is because one of a number programming issues.

> 
Does the occurrance of two identical LocoNet fast clock write messages specifically cause the LRT-LNFC to display fast time that is one minute behind the time displayed by the DT400R? No. Sending a single message (instead of two), using JMRI's "Send LocoNet Packet" tool, still results in a one-minute offset on the LRT-LNFC display.

Does the time on a dtxxx throttle agree or disagree with the LRT-LNFC, that peice of information will narrow down where the problem lies?

devel-bobm, 2018-12-31:
More data here.  There are a variety of interesting findings, highlighted in **bold** text.

My current setup:
- No command station.
- UP90 (front jack) providing power to RailSync (to power the throttle)
- PR3 in "MS100 Interface with termination" mode (to "pull up" LocoNet data lines)
- LRT-LNFC (configured as fast clock slave)
- DT400R

Findings:
- If JMRI is off-line, and I plug in a DT400R (without battery!), and I set the DT400R fast clock, the LRT-LNFC takes the identical time as the DT400R once I hit the throttle "EDIT" button.  If I then set the "rate" to 0 and hit the throttle "EDIT" button, the LRT-LNFC continues to display the same time as was previously set, and stays "paused".  I then hit the DT400R "EXIT" button to get out of "set the alarm time" mode.  **The DT400R appears to remain paused, at the time I set, with the LRT-LNFC matching that time.**  Everything checks out "as expected".

- Still without JMRI running, I wait a few minutes to ensure that both the LRT-LNFC and DT400R are "stopped".  Everything checks out "as expected", with the DT400R and LRT-LNFC showing matching times.

- I then use the DT400R to set the fast clock time and rate to new time and rate.  When I set the new time and press "Edit", **the LRT-LNFC shows the correct time as entered on the DT400R**, and remains paused (a partial answer to a @Pugwash1 question, above).  When I set the rate to be "1:1" via the throttle, the LRT-LNFC becomes un-paused.  When I hit "Exit" to get out of the "alarm" feature, **I find that the DT400R has advanced to one minute beyond what I had entered** (another part of an answer to a @Pugwash1 question, above).  I wait a fast minute and see the LRT-LNFC change minutes just a smidgen before the throttle does so.  They remain offset by one fast minute.

- **The DT400R is transmitting a "fast clock slot read report" message (almost) every "real" minute (I have seen it occasionally skip a message, though).**  I know this is coming from the throttle because if I disconnect the DT400R from LocoNet, the reports no longer appear on LocoNet.

- When the DT400R transmits its "fast clock slot read report" message, JMRI decodes the message with the same time shown on the DT400R display.  The LRT-LNFC, however, shows a time that is one fast minute "behind" the time shown on the DT400R.

- Transmitting a "fast clock slot read request" message does not result in a "fast clock slot read report" with this hardware configuration.  There isn't any real "fast clock master", even though the DT400R is sending occasional "fast clock slot read report" messages.  If another throttle plugs into LocoNet and requests the current fast time, it won't get any info, unless the periodic DT400R "fast clock slot read report" happens to show up close (in time) to its request.  **This seems like the DT400R is acting as a "partial" fast clock master.**

- If I use the throttle to set the time to a new time, and set the rate to a new rate, as soon as I set the new time, the LRT-LNFC accepts the new time (with no offset from the time dialed-up on the throttle).  As soon as I set the new rate and exit back to fast clock display, the DT400R advances its clock by one minute, and is one minute advanced versus the LRT-LNFC.

- If I start-up JMRI and show the "Fast Clock Setup", with Time Source set to "internal computer clock",  and I click "synchronize internal clock and LocoNet fast clock", the LRT-LNFC chirps and jumps to the new time (wall-clock time), at a rate of 1:1.  **Both JMRI and the DT400R throttle show a time which is one minute in advance of _both_ the LRT-LNFC _and_ the computer's clock.**  To me, this behavior implies that the DT400R is misbehaving, and JMRI is modeled to match the misbehavior of the DT400R.

- If I watch for a few minutes, I see that JMRI is transmitting "fast clock slot write" messages every "fast" minute (which at 1:1 equals real-time a minute).

- If I un-check "synchronize internal clock and LocoNet fast clock", JMRI stops sending its periodic "fast clock slot write" messages.

- If I wait a while, looking for periodic "fast clock slot read request"s, I do not see any.  **The LRT-LNFC is not sending the periodic "fast clock slot read request"s.**  It's been over eight minutes so far...  (This answers another @Pugwash1 question from above.)

Additional investigation, with respect to sub-minute values, is in order...

devel-bobm, 2019-01-02:
After further review, all findings reported above on behavior on LRT-LNFC must be considered suspect and likely inaccurate.  I've determined that my LocoBuffer-USB "Rev O" does not properly communicate with the LRT-LNFC, even though the DT400R receives the messages successfully.

My PR4, however, seems to communicate with the LRT-LNFC successfully.  I will need to repeat my earlier experiments using the PR4.

I do not know the exact failure-mode, just that the failure exists.  I suspected that the LocoBuffer-USB "Rev O" might not have enough power so plugged it into the front jack on a powered UP90.  This did not resolve the issue.  I will shelve investigation of this problem in favor of PR4-based experimentations with the LRT-LNFC.

Pugwash1, 2019-01-04:
@devel-bobm Good News I hope.
I know why the LRT-LNFC goes one minute out from the get go.
Digitrax command stations have different speeds. If you know where the end is but not the beginning how do you set the time to a whole minute. What you do  is set it so that 1 tick causes it to go to the next minute. If you set the  time using a dtxxx the only thing it knows is that the last tick is 00 7F (0x7F00) so any value greater than that causes the CS to jump one minute on the next tick. The LRT-LNFC doesnt seem to know that and will remain one fast minute behind. To find the magic zero number send a e7/F command with with the minfrac 00 7F (0x7F00) , immediately (within a sec) after read the FCslot  (bb 7B 00 00 ) on a dcs50 it returns the minfrac xx 79. To set the fracmin to zero use  00 79 (0x7900)  and the clock wont jump forward 1 minute. For a dcs240 the magic number is 00 69 (0x6900). I will see if I can incorporate that in LnClock.

devel-bobm, 2019-01-05:
@Pugwash1, 

> I know why the LRT-LNFC goes one minute out from the get go.
Digitrax command stations have different speeds. If you know where the end is but not the beginning how do you set the time to a whole minute. 

I am not convinced that this is the case.  I believe that it is at least partly because the LocoNet "specification" for fast clocks, as seen in the LocoNet Personal Edition 1.0 spec, says one thing, but the DTxxx throttles seem to do something else.  (I _assume_ that any non-disclosure-agreement version of LocoNet spec will mirror what's said in the LocoNet PE version. )

I believe LRT-LNFC attempts to implement a reasonable interpretation of <256-MINS_60>.  But, as seen in Alex Shepherd's work, the DTxxx throttles tend to interpret the byte as <255-MINS_60>.  That, to me, seems to be the source of the discrepancy.  And that, to me, seems to be a good reason to believe that it may be impossible to get the two types of clock to "agree" about the current time, without some non-trivial changes on someone's part.

devel-bobm, 2019-01-05:
@Pugwash1 ,
Have you noticed that the DTxxx throttles tend to generate an odd-looking LocoNet message when setting the clock to x:00 ?  LocoNet Monitor sees the LocoNet message with <256-MINS_60> as 0x7F and sees <256-HRS_24> as (x-1), so LocoNet Monitor reports it as (x-1):00.

I'm thinking that the monitor should be updated to reflect this odd behavior, where the special-case "forces" the hours back to (x) and zeros-out the minutes.  This, of course, may be inappropriate when a "more-well-behaved" device comes along, though, such as JMRI as a "true" fast-clock master, or when an LRT-LNFC sets the fast clock, or when a Z21 device sets the fast clock or ...

Does your experimentation give you an opinion on whether this proposed change would be a good one?

bobjacobsen, 2019-01-05:
> it may be impossible to get the two types of clock to "agree" about the current time, without some non-trivial changes on someone's part.

LocoNets are cheap now. JMRI could have connection specific option(s) about how clocks behave, so that people who need more than on behavior just create a second LocoNet for the clocks with different behavior. 

If you can’t make everybody happy with one thing, offer two things & sort them out.

Pugwash1, 2019-01-05:
Bob
I have noticed the odd behavior around x:00. I have no inspiration as to why it behaves like that, it may have to be an option to program round it.

closed 2021-04-14 by stale[bot]