JMRI/JMRI #6359: /json/time schema changed in 4.14?

preble, 2018-12-26:
In 4.12+Rb6a9bb1 `GET /json/time` returns:

```
{"type":"time","data":{"time":"2018-12-26T15:35:34Z","rate":1.0,"state":2}}
```

The same endpoint in 4.14, however, returns the expected JSON object wrapped in an array:
```
[{"type":"time","data":{"time":"2018-12-26T14:57:20Z","rate":1.0,"state":4}}]
```

Is this intentional? I wasn't able to find mention of this in the [release notes](http://jmri.org/releasenotes/jmri4.14.shtml), unless this bit is somehow related?

> Most JSON WebSocket lists now send individual and list changes

rhwood, 2018-12-27:
Apparently it was intentional (see https://github.com/JMRI/JMRI/commit/c5abd8e31660b39a7efda518e98a1c60121bf403), since the WebSocket always returns an array of JSON objects, someone thought that all JSON endpoints should, although I would not have done that, since the JSON RESTful protocol is explicitly documented as not returning an array for this. I'd call this a bug that should be reverted, since clients can check that a message is a JSON object or array and behave appropriately.

mstevetodd, 2018-12-27:
Thanks Randall. I was not aware there was a documented standard for this. Can you provide a link?

rhwood, 2018-12-27:
http://jmri.org/help/en/html/web/JsonServlet.shtml documents that http://<server>:<port>/json/<type>/<name> will return a single item, however, some types (time and other types that are single instances within JMRI) respond to empty names, since, being single instances, they have no name. I'm reading "time" being listed as "n/a" in the table of unnamed types means it should not respond with a list.

Basically, the "list" request format response always being an array is only meaningful for listable items, and non-listable, non-namable items may respond to a null name without providing an array.

Note that with your change in place, an array of objects (say turnouts) will be returned wrapped in an array (so the returned response would be `[[{"type": "turnout", "data": ...} ...]]` instead of the documented `[{"type": "turnout", "data": ...} ...]`.

http://jmri.org/JavaDoc/doc/jmri/server/json/package-summary.html also documents that non-array responses are valid responses.

pabender, 2018-12-27:
> http://jmri.org/help/en/html/web/JsonServlet.shtml documents that http://:/json// will return a single item, however, some types (time and other types that are single instances within JMRI) respond to empty names, since, being single instances, they have no name. I'm reading "time" being listed as "n/a" in the table of unnamed types means it should not respond with a list.
> 

Is it still a valid assumption that time is a single item?

@balazsracz has been working on OpenLCB fastclovk support.  OpenLCB supports the notion of multiple independent fastclocks.

rhwood, 2018-12-27:
If the JSON time service provides an array of time sources, then it will; the code change that caused this change in behavior was not a code change in the time service, it was a code change to wrap all JSON responses in an array of objects or arrays, regardless of what the service returned.

So the change in 4.14 did not have the effect of allowing multiple time sources to function, but only had the effect of always wrapping the single time source response in an array.

rhwood, 2018-12-27:
The time question is also wrapped up in which connection is the default time source. Can we set that using defaults, or is there some other UI for specifying which time source I get from the InstanceManager when using getDefault(...)? As of now, the JSON time source only handles the default; unless that is changed, it will not and should not be an array.

mstevetodd, 2018-12-27:
I see three "single instance" types: time, power and railroad. I agree that my change broke those and I will fix shortly. Apologies to @preble and any others inconvenienced by that.
I still contend that the "listable" types, that happen to only have one element defined, should still return an array, but I'm not sure this is the proper place to re-hash that.

balazsracz, 2018-12-27:
The singletons I know of are time and power. These behave differently:
- power manager is coming from the InstanceManager; each connection can register a PowerManager, and the Defaults selector exists for it in the profile preferences.
- clocks are different: there is a singleton object called Timebase, which all of the JMRI user interfaces are using. This can be reached from the InstanceManager but it is not connection specific. The individual connections can register a ClockControl object with the InstanceManager, but there is no concept of Default for ClockControl and it does not appear in the defaults selector. The only place where ClockControl objects are fetched from the instance manager and operated upon is the single implementation of the Timebase class (i think SimpleTimebase). There is a menu entry (tools->fast clock->setup fast clock...) that opens a specific dialog for setting up clock, which has a dropdown for all the registered clockcontrol implementations, including the internal fast clock. The selection here only influences what the Timebase implementation is doing behind the scenes, i.e. whether it synchronizes the clock visible on the Timebase interface with the hardware clock provided by the connection. The settings are saved into the panel.xml instead of the connection preferences xml.
- I know nothing about 'railroad' and don't know where to set that up in JMRI.

Back to clocks: There are a lot of todos in the code for adding support for multiple fast clocks. This can be done in two different ways:
1) JMRI keeps having a single timebase, but it can be configured to be synchronizing with multiple different hardware connection's fast clocks. Most of the todos seem like this. The UI does not need to change, i.e. there is still just one clock face that JMRI can show. THere is no need to report hardware clocks individually to clients.
2) The UI gets extended to support more than one independent clock, and all the UIs become selectable to which clocks they display; in this case reporting multiple clocks' state becomes necessary. Synchronization between clocks, or correcting clocks that drift would become a crossbar setting. I think this is a huge complexity explosion with very little benefit to the user and quite likely limited demand as well. I don't think it makes sense to invest in this direction (I certainly will not spend time on it even though OpenLCB is a hardware protocol that would allow multiple independent clocks).

preble, 2018-12-27:
Thanks @mstevetodd and everybody else for the quick attention on this issue!

closed 2018-12-27 by preble

mstevetodd, 2018-12-27:
@preble, what sort of client are you using for this? The original change was made 5 months ago and no one else reported the issue.

commit 20b09c54 (2019-02-24): Fix #6359

Other attempted fixes made invalid presumtions about returned values for certain message types. For example: there being only a single power object is not guaranteed.