JMRI/JMRI #6368: Programmer "311" needs description in help

devel-bobm, 2018-12-28:
"Communications Error Between PC and command station (311)" is not described in the help file, while the other "3xx" programmer error messages are described.  

This omission affects several different pages of the english help files:
- help/en/html/setup/Errors.shtml
- help/en/html/tools/programmer/messages.shtml
- help/en/manual/DecoderPro/Error.shtml
- help/en/manual/DecoderPro3/Error.shtml

A couple of pages in the french help files would need updates, too:
- help/fr/html/setup/Errors.shtml
- help/fr/html/tools/programmer/messages.shtml

This message appears to be used only by `XNetProgrammer`, `EliteXNetProgrammer`, and `LI100XNetProgrammer`.  

@pabender, is this (i.e. english pages) something you can work?

pabender, 2019-01-03:
@devel-bobm See #6391

I also updated the description of the 310 error to include an example of how that can happen without there actually being anything wrong.

My written french is very rusty, so I won't try to update those pages (I can still read what's there at least....)

devel-bobm, 2019-01-03:
That works for me!  

I am closing this issue, under the assumption that he P/R will be merged.

closed 2019-01-03 by devel-bobm