JMRI/JMRI #6369: Add support for bit-mode programming operations for LocoNet

devel-bobm, 2018-12-28:
Most Digitrax command station models support bit-mode programming, but JMRI does not (currently) implement the code to support it and does not allow the user to select Bit-mode programming operations.

Bit-mode reads can be significantly faster than other service-mode CV read types.

@bobjacobsen, @pabender, all:
Is anyone aware of a reason why JMRI does not support bit-mode programming for LocoNet-based command stations?

dheap, 2018-12-28:
@devel-bobm All command stations that support Direct Mode as per S9.2.3 the majority these days) are required to internally implement the following Instruction Types:

- Bit Manipulation (Bit Read & Bit Verify).
- Verify Byte.
- Write Byte.

Many (if not the majority of) command stations that support Direct Mode don't expose the individual instruction types to a computer interface but simply provide two instructions:

- Direct Read - internally (to the command station) uses a sequence of Verify Bit and Verify Byte to read a CV using the minimum possible number of programming packets.**
- Direct Write - uses the Write Byte instruction to do it with one programming packet type.

I can't speak for LocoNet systems programming implementation in JMRI, but LocoNet Protocol seems to expose:
1. Direct mode byteRead/Write on Service Track
2. Direct mode bit Read/Write on Service Track 

Mode 1 would have to use a command-station sequence of Verify Bit and Verify Byte (as per illustration below (as there is no such NMRA packet as "Direct Byte Read"^^). Having JMRI manually send a sequence of Mode 2 instructions (as per example below) via LocoNet Protocol to read an entire CV bit-by-bit would have to be slower than using Mode 1 (one LocoNet instruction, command station does the sequencing).

** A typical sequence is:
- Is bit 0 of CVnnnn set?
- Is bit 1 of CVnnnn set?
- Is bit 2 of CVnnnn set?
- Is bit 3 of CVnnnn set?
- Is bit 4 of CVnnnn set?
- Is bit 5 of CVnnnn set?
- Is bit 6 of CVnnnn set?
- Does CV nnnn have value 1stuvxyz (given the replies above)? If success, return 1stuvxyz.
- Does CV nnnn have value 0stuvxyz (given the replies above)? If no, return Read Fail.
You can't read an entire CV much faster than that...

^^ The only reverse communications from a decoder to a programmer  in S9.2.3 Service Mode are:
- No reply, usually taken as an implicit NO.
- Reply (a single pulse), taken as a YES.

Dave

bobjacobsen, 2018-12-28:
Originally, JMRI used the LocoNet programming control bits (from the LocoNet PE, pages 13 and 14 in the edition I have) to offer three service mode choices: Paged, Direct Bit, and Direct Byte.  It also provided ops-mode bit and byte choices.  That's what the DCS100 provided, sort - of, but none of the three Direct modes worked the way that various NMRA people thought they should. And there was much pointless discussion....

Eventually, a consensus emerged that the bit modes didn't really work in the DCS100, and they certainly didn't work in (early) DCS50's, where "work" is defined as "program non-Digitrax decoders reliably".  Various people had various opinions on why, there was much discussion, but I never put a 'scope on it to check.

But they were enough faster that people will still recommend trying them, and there were people who did but discovered they were unreliable so they were unhappy, while others had various incantations (My favorite, IIRC:  Lean MRC decoder-equipped locos off the tracks for the first couple seconds, then drop them back down. That actually seemed to work...)

It was a mess.  So in late 2004, the bit modes were suppressed in the user interface.  You can still see some signs of them in `LnConstants` and `SlotManager`.

If we want to revisit this, there are a couple questions:

- Do they work now? If so, with what hardware? 
- Is the LocoNet PE the right protocol to address them so that they work?  (I.e. do they work because they're requested some other way?)

dheap, 2018-12-28:
I've done some Program Track packet sniffing with several DCC systems I have access to and they use minor variations on my typical example.

But I don't have regular access to a LocoNet or Digitrax system.

devel-bobm, 2018-12-28:
@bobjacobsen,

In my recent experiments with DCS210 and DCS240, I find that I cannot reliably perform a "read entire sheet" of the CV pane with a hard-wired-to-the-command-station Digitrax SDH164D sound decoder.  It does not matter what programming mode I use, including a hacked-in "bit-mode", I _always_ have at least one "red" CV in the list of perhaps 50 CVs.

If that is any indication of "does bit mode work", then the answer is "no".  But then each supported programming mode is also having problems reading the decoder, so apparantly something else is going on...

I will try to repeat this experiment with my DCS100 (with oldish firmware) and DB150 and DCS51...

danielb987, 2018-12-28:
@devel-bobm 
On a web forum, I got the tip that if you have a Bachmann decoder, it may need a resistor parallell to the decoder in order to program the decoder. Maybe you could test to have a 1K ohm resistor or even a 150 ohm resistor parallell to the decoder to see if that helps.

bobjacobsen, 2018-12-30:
> I've done some Program Track packet sniffing with several DCC systems I have access to and they use minor variations on my typical example.

I went back through some old notes to try to understand this.

The problem seems to have been a combination of philosophical and practical.  The (early) Digitrax bit mode reading checked the eight bits with eight operations, but didn’t check with a final byte read nor check the complement of each bit when no response was returned. This is a touch faster, but requires a lot from the reliability of the connection and decoder: People thought that it was too often giving reads marked OK but with incorrect values.

I have no stake in this either way.  it would be easy to add ProgrammingMode.DIRECTBITMODE and/or ProgrammingMode.OPSBITMODE to the Digitrax-supported modes.  People could select or not select them as they see fit.  If a controversy emerged again, we could figure out how to deal with it either via pulling the code and/or acts of moderation.

dheap, 2018-12-31:
This helps explain:
> The problem seems to have been a combination of philosophical and practical.  The (early) Digitrax bit mode reading checked the eight bits with eight operations,

Eight operations is the bare minimum.

> but didn’t check with a final byte read

Which may succeed within the eight, but may require a ninth to differentiate between OK or Read Fail.

> nor check the complement of each bit when no response was returned.

That way can be expensive, possibly up to 16 operations. Did/does any system actually do it this way?

> eight bits with eight operations ... is a touch faster, but requires a lot from the reliability of the connection and decoder: People thought that it was too often giving reads marked OK but with incorrect values.

Personally, I would consider the "eight bits with eight operations but no check with a final byte read" to be too great a risk in any circumstances.

Is anyone in the position of being able to Packet Analyse current LocoNet-based systems to see the behaviour in the available modes?

dheap, 2018-12-31:
Agreed that can be a problem.

I use the NCE Packet Analyser (neither it nor Pricom are available to purchase new any more). It normally derives power from an external source (usually the NCE Cab Bus line). With the DCC systems I have access to:
- NCE Power Cab always has track power applied. It just switches between full and restricted power modes.
- SPROG in Programmer Mode can be configured to have track power on prior to entering service mode (although it does drop it shortly after programming, with no adverse effect on packet sniffing).
- NCE Power Pro with PTB-100 powers the programming track all the time you are in service mode with the throttle. With JMRI you can use Send Binary Command to manually enable service mode and send service mode commands.

> devel-bobm Can you suggest any particular hardware? Can the Pricom pocket tester be made to sniff the service-mode programming track on a system which cuts power to the decoder between service-mode operations?

dheap, 2018-12-31:
Interesting:

## Service Mode Packet Sniffing:

### SPROG read CV19 of ESU:
78 07 EF 90 ++
78 07 EE 91 +++++++
78 07 ED 92 +++++++
78 07 EC 93 ++
78 07 EB 94 +++++++
78 07 EA 95 ++
78 07 E9 96 ++
78 07 E8 97

_Either SPROG doesn't to a byte verify or I'm missing the last packets due to power-off. That's possible._ 

@crosland may be able to comment on this.

### NCE Power Pro read CV19 of ESU:
78 07 EF 90 ++
78 07 EE 91 ++++
78 07 ED 92 ++++
78 07 EC 93 ++
78 07 EB 94 ++++
78 07 EA 95 ++
78 07 E9 96 ++
78 07 E8 97 ++
74 07 97 E4

It appears the NCE Power Pro may always do nine operations (eight bit verifies plus a byte verify). But I haven't confirmed that...

I didn't bother with Power Cab at present as it's not currently set up. Can do so if needed...

devel-bobm, 2018-12-31:
(Thread drift:) Perhaps we should consider adding a JMRI "programming mode" (perhaps as a sort of "facade") where it can perform "bit-mode" reads (with a protective "byte-mode read value verify" for those programming devices which need it added) and "byte-mode" writes.  This would seem to be, on average, the fastest combination of setting to read/program CVs.  

This would apply only to those programming devices which support both bit-mode reads and byte-mode writes.

crosland, 2018-12-31:
The SPROG is not strictly compliant with the NMRA specs. It uses a pragmatic approach developed during it's early life followed (mostly) by "if it ain't broke, don't fix it".

It does not do an initial bit verify sequence to determine if the decoder supports direct bit mode.

Direct mode writes are always direct byte. This is faster and I found QSI decoders did not handle direct bit writes (this may well be fixed now).

Prior to SPROG IIv3 the SPROG implemented direct bit reads by testing for 0 in each bit. It then did a byte verify to determine if no ACK from all bits meant a value of 0xFF or a problem.

SPROG IIv3 onward (Inc. SPROG 3 and Pi SPROG) verify all bits against 1. If no ACK from all bits (could be 0x00 or a problem) then it will verify all bits against 0. If there is an ACK from every bit then it's a genuine read of 0x00, otherwise it reports NO ACK.

Unfortunately, I don't recall why I broke my own rule and "fixed" the behaviour this way.

It seems Dave's SPROG read of CV19 read a non-zero value during the first pass.

pabender, 2018-12-31:
Sent from my iPhone

> On Dec 31, 2018, at 10:09 AM, Bob Milhaupt <notifications@github.com> wrote:
> 
> (Thread drift:) Perhaps we should consider adding a JMRI "programming mode" (perhaps as a sort of "facade") where it can perform "bit-mode" reads (with a protective "byte-mode read value verify" for those programming devices which need it added) and "byte-mode" writes. This would seem to be, on average, the fastest combination of setting to read/program CVs.
> 

I would hesitate greatly to add a JMRI specific programming mode labeled as such.   I think that would only confuse users.

I think it would be fine to add a facade that does this without the users knowledge.

On the XpressNet side of things, Lenz says they support both bit mode and byte mode direct programming operations, but there is only one service mode command for direct mode reads and another for writes.

Paul

crosland, 2018-12-31:
That would only make sense if a system can perform individual bit reads. 
Does any system allow that?

I think the generally understood meaning of direct bit is that the 
command station/programmer performs 8 (or more) individual bit 
operations but only handles bytes at the user interface level.

Andrew

<mention@noreply.github.com>
Sent: 31/12/2018 15:09:03
for LocoNet (#6369)

bobjacobsen, 2018-12-31:
> (Thread drift:) Perhaps we should consider adding a JMRI "programming mode" (perhaps as a sort of "facade") where it can perform "bit-mode" reads (with a protective "byte-mode read value verify" for [...]

That all assumes a level of control that JMRI doesn’t have with any DCC hardware (AFAIK)

To return to the original issue:  Let’s go ahead and make “Direct Bit Mode” visible for Digitrax users and see what happens.

Who wants to code it?

dheap, 2018-12-31:
I've corrected https://github.com/JMRI/JMRI/issues/6369#issuecomment-450607381.

Just realised it had a significant typographical error. I'd started testing CV19 and then switched to CV8 to avoid a "contents zero" special  case, but didn't correct my draft.

dheap, 2018-12-31:
Also added Power Cab result to 
https://github.com/JMRI/JMRI/issues/6369#issuecomment-450607381.

Seems to use same algorithm as Power Pro.

devel-bobm, 2018-12-31:
@bobjacobsen, 
I've got "direct bit" mode coded for LocoNet.  It's a "plain" implementation - it does not attempt to do anything more than the programmer-requested read or write operation.  It is noticably faster than "direct byte" mode for reads, and noticably slower than "direct byte" mode for writes.

I can probably clean-up my code and submit within about a day's time.  I may be able to code-up some testcases in that time-frame...

devel-bobm, 2019-01-01:
Pull request #6381 is a first attempt.  

It is _not_ working well for me with the one decoder I can easily test with.  
- Writes do not do anything like what I thought they would.
- Reads are haphazard, at best, with many unexplained rejections by the command station immediately upon read request, and some "incorrect" reads.  When an "incorrect" read occurs in the "speed table" CVs, it is very obvious, because JMRI insists on re-calculating values and marking them as "edited".

These behaviors are seen both with DCS51 and DCS210 command stations.

devel-bobm, 2019-01-01:
@bobjacobsen mentioned above that bit read and write operations hadn't been very reliable.  I can confirm this statement with modern hardware.

Here's a LocoNet monitor log when I manually requested 10 reads of a single CV using "direct bit mode", showing three different read values, with incorrect read values **bolded**:
20:28:06.897: [EF 0E 7C 0B 00 00 00 07 00 45 00 7F 7F 2B]  Bit Read in Direct Mode on Service Track: CV70.
20:28:06.897: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
20:28:08.659: [E7 0E 7C 0B 00 00 02 47 00 45 17 7F 7F 76]  Programming Response: Read Bit in Direct Mode on Service Track Was Successful: CV70 value **23** (0x17, 00010111b).
20:28:09.190: [EF 0E 7C 0B 00 00 00 07 00 45 00 7F 7F 2B]  Bit Read in Direct Mode on Service Track: CV70.
20:28:09.205: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
20:28:10.906: [E7 0E 7C 0B 00 00 02 47 00 45 1F 7F 7F 7E]  Programming Response: Read Bit in Direct Mode on Service Track Was Successful: CV70 value 31 (0x1F, 00011111b).
20:28:11.935: [EF 0E 7C 0B 00 00 00 07 00 45 00 7F 7F 2B]  Bit Read in Direct Mode on Service Track: CV70.
20:28:11.935: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
20:28:13.698: [E7 0E 7C 0B 00 00 02 47 00 45 1B 7F 7F 7A]  Programming Response: Read Bit in Direct Mode on Service Track Was Successful: CV70 value **27** (0x1B, 00011011b).
20:28:14.182: [EF 0E 7C 0B 00 00 00 07 00 45 00 7F 7F 2B]  Bit Read in Direct Mode on Service Track: CV70.
20:28:14.197: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
20:28:15.945: [E7 0E 7C 0B 00 00 02 47 00 45 17 7F 7F 76]  Programming Response: Read Bit in Direct Mode on Service Track Was Successful: CV70 value **23** (0x17, 00010111b).
20:28:17.333: [EF 0E 7C 0B 00 00 00 07 00 45 00 7F 7F 2B]  Bit Read in Direct Mode on Service Track: CV70.
20:28:17.333: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
20:28:19.049: [E7 0E 7C 0B 00 00 02 47 00 45 1F 7F 7F 7E]  Programming Response: Read Bit in Direct Mode on Service Track Was Successful: CV70 value 31 (0x1F, 00011111b).
20:28:20.094: [EF 0E 7C 0B 00 00 00 07 00 45 00 7F 7F 2B]  Bit Read in Direct Mode on Service Track: CV70.
20:28:20.110: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
20:28:21.810: [E7 0E 7C 0B 00 00 02 47 00 45 1F 7F 7F 7E]  Programming Response: Read Bit in Direct Mode on Service Track Was Successful: CV70 value 31 (0x1F, 00011111b).
20:28:22.325: [EF 0E 7C 0B 00 00 00 07 00 45 00 7F 7F 2B]  Bit Read in Direct Mode on Service Track: CV70.
20:28:22.341: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
20:28:24.057: [E7 0E 7C 0B 00 00 02 47 00 45 1F 7F 7F 7E]  Programming Response: Read Bit in Direct Mode on Service Track Was Successful: CV70 value 31 (0x1F, 00011111b).
20:28:24.571: [EF 0E 7C 0B 00 00 00 07 00 45 00 7F 7F 2B]  Bit Read in Direct Mode on Service Track: CV70.
20:28:24.571: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
20:28:26.319: [E7 0E 7C 0B 00 00 02 47 00 45 17 7F 7F 76]  Programming Response: Read Bit in Direct Mode on Service Track Was Successful: CV70 value **23** (0x17, 00010111b).
20:28:26.943: [EF 0E 7C 0B 00 00 00 07 00 45 00 7F 7F 2B]  Bit Read in Direct Mode on Service Track: CV70.
20:28:26.943: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
20:28:28.659: [E7 0E 7C 0B 00 00 02 47 00 45 1F 7F 7F 7E]  Programming Response: Read Bit in Direct Mode on Service Track Was Successful: CV70 value 31 (0x1F, 00011111b).
20:28:29.236: [EF 0E 7C 0B 00 00 00 07 00 45 00 7F 7F 2B]  Bit Read in Direct Mode on Service Track: CV70.
20:28:29.236: [B4 6F 01 25]  LONG_ACK: The Slot Write command was accepted.
20:28:30.983: [E7 0E 7C 0B 00 00 02 47 00 45 17 7F 7F 76]  Programming Response: Read Bit in Direct Mode on Service Track Was Successful: CV70 value **23** (0x17, 00010111b).

Five of those reads returned erroneous values (50% error rate), with two different erroneous values.

Note that the same physical setup produced a sequence of ten identical read values from that CV when using "Direct Byte" reads, for a 0% error rate.  So this is not simply a "bad connection" problem, it is a real problem in whatever algorithm or timings that Digitrax is using.

I got similar failure-rate results with the DCS51.  I cannot check the DCS210 because I seem to have killed its ability to sense acknowledgment pulses.

Due to the low reliability of the hardware, I will be closing P/R #6381.

devel-bobm, 2019-01-01:
As mentioned above, funcitonal reliability of actual hardware makes this mode of operation highly unreliable.  This issue must be shelved until Digitrax produces hardware where the mode may be reliably used.

dheap, 2019-01-01:
Further Service Mode Packet Sniffing:

### SPROG read CV19 of ESU; contents 0:
78 12 EF 85 +++++++
78 12 EE 84 +++++++
78 12 ED 87 +++++++
78 12 EC 86 +++++++
78 12 EB 81 +++++++
78 12 EA 80 +++++++
78 12 E9 83 +++++++
78 12 E8 82 +++++++
78 12 E7 8D ++
78 12 E6 8C ++
78 12 E5 8F ++
78 12 E4 8E ++
78 12 E3 89 ++
78 12 E2 88 ++
78 12 E1 8B ++
78 12 E0 8A

@crosland This confirms your "16 Reads needed if contents 0" solution.

### NCE Power Pro read CV19 of ESU; contents 0:
78 12 EF 85 ++++
78 12 EE 84 ++++
78 12 ED 87 ++++
78 12 EC 86 ++++
78 12 EB 81 ++++
78 12 EA 80 ++++
78 12 E9 83 ++++
78 12 E8 82 ++++
74 12 00 66

### NCE Power Cab read CV19 of ESU; contents 0:
78 12 EF 85 +++++
78 12 EE 84 +++++
78 12 ED 87 +++++
78 12 EC 86 +++++
78 12 EB 81 +++++
78 12 EA 80 +++++
78 12 E9 83 +++++
78 12 E8 82 +++++
74 12 00 66

dheap, 2019-01-01:
@crosland Seems the "minimum operations; 8 or at most 9 needed for answer" I described in https://github.com/JMRI/JMRI/issues/6369#issuecomment-450297783 is not fully implemented in any of the systems I tested.

Pity, as it returns a high degree of confidence in minimum operations; only succeeds if a Full Byte Verify returns an ACK, otherwise reports Read Fail, with worst-case requiring 9 operations in total.

dheap, 2019-01-01:
On the other hand, the NCE (always 9 operations) system provides greater confidence, the 8 individual bit reads must agree with the single byte verify check, probably a small price to pay.

devel-bobm, 2019-08-30:
No activity in 8 months, and no data that contradicts my findings of erratic behavior of bit-mode CV reads and writes on service mode track.  Re-closing this issue.

closed 2019-08-30 by devel-bobm