JMRI/JMRI #6380: Unable to control turnout using DCC++ digital pin outputs

Morlack, 2018-12-30:
I am currently trying a setup with  the following:

- JMRI (4.14)
- ESP32-WROOM with DCC++
- LED connected to pin 12
- Output mapped in Base Station using ID 7 referencing pin 12

When sendinga DCC++ command using the packet tool in JMRI the following commands work:
- `Z 7 1`
-` Z 7 0`

However, when adding a turnout to the switch table according to the documentation (http://jmri.sourceforge.net/help/en/html/hardware/dccpp/Turnouts.shtml) it only works rarely. It seems to send the wrong commands and the Traffic monitor returns a lot of unrecognized replies:

The logging looks like this:
```
Output Cmd: 
	Output ID: 7
	State: HIGH
 Unregonized reply: Z 7 1
	vals: Z   7   1
 Output Command Reply: 
	Output Number: 7
	OutputState: LOW
 Status Cmd 
 Unregonized reply: s
	vals: s
 Base Station Status: 
	Version: 1.1.2
	Build: Dec 30 2018 14:45:13
 Power Status: 
	Name:MAIN	Status:OFF
 Power Status: 
	Name:PROG	Status:OFF
 Output Command Reply: 
	Output Number: 7
	OutputState: 12
	OutputState: 0
	OutputState: 0
 Comm Type Reply Type: 1 Port: 192.168.1.17
 Output Cmd: 
	Output ID: 7
	State: HIGH
 Unregonized reply: Z 7 1
	vals: Z   7   1
 Output Command Reply: 
	Output Number: 7
	OutputState: LOW
 Status Cmd 
 Unregonized reply: s
	vals: s
 Base Station Status: 
	Version: 1.1.2
	Build: Dec 30 2018 14:45:13
 Power Status: 
	Name:MAIN	Status:OFF
 Power Status: 
	Name:PROG	Status:OFF
 Output Command Reply: 
	Output Number: 7
	OutputState: 12
	OutputState: 0
	OutputState: 0
```

There is no logging in the console output of the JMRI application itself that is related to the issue. I'm not sure what steps to do next. I've tried the following:

- I've edited pin number and IDs, since I saw the numbers split in the log. But single-digit numbers also fail
- Clicking multiple times helps, but unreliably. Clicking the switch 3 times might switch the state, or switch and switch back again
- Issuing the commands myself, which work
- Trying rocrail to verify the communication, but ran into issues due to little documentation about dcc++ and rocrail together (JMRI seems to be the first choice)

Could this be a bug in JMRI or have I overlooked something entirely obvious? I'm hoping you could help.

Thanks in advance!

Morlack, 2019-01-04:
Update: The turnout works with rocrail

atanisoft, 2019-02-08:
@Morlack can you share some more details on your setup? This sounds like the ESP32 CS is not receiving the commands or it is not processing them correctly. Are you using my [DCC++ESP32](https://github.com/atanisoft/DCCppESP32) code? If so please connect a serial console to the ESP32 and lets confirm that it is in fact receiving and processing the commands you are sending it.

closed 2021-04-14 by stale[bot]