JMRI/JMRI #6394: Saving preferences behave strange

danielb987, 2019-01-04:
I open _Edit/Preferences..._, changes _Command station type_ and then click _Save_.

I get a dialog that tells me that I must restart PanelPro for the changes to take effect and I click _Later_.

I then click _Quit_ in the main PanelPro window.

I get a dialog _Preferences has unsaved changes. Do you want to save them?_ I click _No_.

I start PanelPro again and open Preferences. The command station has now changed.

This seems like a bug to me. Either the preferences are already saved and then _Quit_ should not warn me about unsaved changes. Or the changes should not be saved since I clicked _No_.

bobjacobsen, 2019-02-21:
There's probably a bug here where some preference either isn't marked as not-dirty when saved. Was the _only_ thing you changed the command station?  Which connection type and subtype(s)?

danielb987, 2019-02-22:
Yes. The only change I do is that I change the `Command station type`.

I have also found another bug. If I change the command station, click "Save" and then "Later", and then click "Quit" to exit PanelPro, and then click "Cancel", PanelPro still quits. Either the quit process should be aborted when I click "Cancel", or the "Cancel" button should be removed.

bobjacobsen, 2019-02-22:
Which connection type?

danielb987, 2019-02-22:
LocoNet LocoBuffer-USB

closed 2021-04-14 by stale[bot]