JMRI/JMRI #6396: MRC system ProgrammerException

Prosyne, 2019-01-04:
I just installed JMRI on an old ACER Laptop running VISTA to run my MRC Express N-scale layout using the MRC serial dongle.

JMRI started, and looked OK, but anything I tried to do resulted in something like:

"Exeption during CV write: jmri.ProgrammerException: programmer in use"

I then tried many hardware substitutions, finally installing JMRI on my new(ish) Win10 laptop, running through an MRC Prodigy Advance, hooked up to a single piece of track, with a single locomotive.

No difference. Tried different locos. Tried rebooting, restarting. Always the same result.

I can run the MRC throttle from inside JMRI and the loco responds fine.  I just can't do any programming.

Anyone got an idea what I'm doing wrong ?

dheap, 2019-01-05:
This sort of error is usually caused by starting an operation and then starting it again or starting something else before the first has finished.

Example would be double-clicking on a button (Read, Write, Identify) rather than single-clicking.

Also try going to Preferences->Roster->Programmer and uncheck the "Cache..." and "Confirm..." options. The "Show CV numbers in tooltip" is fine to have on.
> JMRI started, and looked OK, but anything I tried to do resulted in something like:
> 
> "Exeption during CV write: jmri.ProgrammerException: programmer in use"

Prosyne, 2019-01-05:
Thanks for the quick response Dave

Unfortunately, your suggestions didn't help:

I fired up JMRI, put a loco on the Program Track, and clicked the Program Track radio button. I selected "New Loco" and made sure to single-click on the "Read type from Decoder" box: The response was:  Read MFG ID - CV 8
Then: Nothing happened.  Clicking the box again 10 - 15 minutes later brought up the usual Error.

If I bypass "Read type from Decoder" and manually pick a Decoder to create a Loco profile, then I can't write anything to the Decoder, single clicking or not, without getting that Error.

I checked my Java install, and it's the correct version. Note that this is repeatable on both my old ACER and newer HP laptops - even simplifying the system down to: a laptop, base station, MRC Dongle and a piece of track.

Any other ideas?

KenC57, 2019-01-05:
Did you try both the 'Direct' and 'Paged' programming modes? Depending on
the age of the decoder, some respond better one way or the other.

You might also try the 'Single CV Programmer' as a test. Just try reading CV
1 or 29 for example. Also trying to use the handheld to ask the MRC system
to read a CV, instead of asking JMRI to do it. If the handheld can't get a
successful read, then JMRI won't either.

We need to confirm that the decoder can be read. Some of the MRC decoder,
usually lower than 1800 series, don't read easily. So mention the type of
decoder, if you know it. That might give other clues.

Just tossing these out incase Dave didn't.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

dheap, 2019-01-05:
@Prosyne Your report indicates that JMRI jams up reading the first CV, possibly due to command station or decoder issues.

In that case, the error when you click again is expected behaviour.

Please follow up Ken's suggestions.

It will help us hugely if you go to the Help->System Console menu item, click Copy to Clipboard and paste that into a message for us all to see.

Alternatively, go to the Help->Locations menu item, click on Open Log Files Location, find the session.log file and attach to this issue.

.

Prosyne, 2019-01-05:
Sorry, I don't know what 'Direct' and 'Paged' programming modes are, or how to access them.  I could not find anything in JMRI's dropdowns about this.

MRC handhelds can read and write CV's; MRC's Software can write speed tables to the CV's.  I can run the MRC throttle inside JMRI.  Honestly, I only installed JMRI for the speed tables, but MRC come pretty close with theirs.

JMRI won't read or write anything to an NCE D13SR or two types of LaisDCC decoder.

I'm attaching the console logfile

[JMRI console.txt](https://github.com/JMRI/JMRI/files/2729353/JMRI.console.txt)

KenC57, 2019-01-05:
Well when you are in any of the programmers, there is a pulldown that shows
which 'programming mode' it is going to use. This has to do with the exact
commands used to try to talk to the decoder. Usual modes are: Paged, Direct,
and Register. Register is the oldest and most restrictive, hardly ever used
any more. Paged and Direct are the frequent ones and Direct is usually
faster.

Looking at the log file you supplied, I have to ask if you loaded the MRC
drivers for the device? While you are opening COM5, if it didn't have the
right driver (Microsoft tends to load a default driver if you don't supply
another driver).

To confirm the connection works, have you tried the throttle in JMRI? If you
can run a loco via the JMRI throttle, that would say we are talking to the
command station just fine. If it doesn't work, that says we are not talking
to the command station correctly. That would point to a connection or driver
issue.

Last thing is knowing which 'throttle id' is set for JMRI and what throttle
id's are used on your layout. JMRI has to be set to something not in use on
the layout. Just one other thought that has caused issues for folk in the
past.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

thenoel1, 2020-06-20:
Any update on this?  I am having the exact same issue.  MRC, Programming mode "Automatic" using Win 10

danielb987, 2020-08-19:
I'm closing this issue since there has been no comment on the original problem for ove a year.

@thenoel1 
If you still have the problem, please open a new issue. You can refer to this issue if you want by writing it's number in the new issue: #6396.

closed 2020-08-19 by danielb987

p10tyr, 2024-04-29:
I have been having this problem. Not sure what is going on.
I had a nano programmed with dcc-ex 4.0 and used to work just fine. 
I then installed the latest version of dcc-ex 5 and used a raspberry pi with pi-jmri image.

i kept getting the smae cv-9  -- then locked up with in use
i also then installed jmri on windows again and the exact same issue

i also downhraded dcc-ex to 4.5 something with the same issues.

I dont have the paged option just direct.

p10tyr, 2024-04-29:
Found the problem. I disabled programming support after reading it should be disabled on the nano.. and didn't realise what that meant- i then blindly copied to config file to aurduino and kept uploading the code with that config
[image]

I only realised it must be something n00b because i couldn't even do a manual <w 8 4> to reset my lias .... doh!

Wonder if ex-dcc could return a message like.. programming disabled.

danielb987, 2024-04-29:
@p10tyr 
> Wonder if ex-dcc could return a message like.. programming disabled.

If you want that feature to be added, you need to open a new issue on GitHub. This issue is five years old so it's unlikely that anyone will take notice. The only reason I saw your message was because I have replied to this issue before and therefore got emails with your replies.

p10tyr, 2024-04-30:
> @p10tyr
>
>
> If you want that feature to be added, you need to open a new issue on GitHub. This issue is five years old so it's unlikely that anyone will take notice. The only reason I saw your message [...]

Thanks 
https://github.com/JMRI/JMRI/issues/13066