JMRI/JMRI #6397: JMRI Roster Media image rendering does not utilise EXIF Orientation Tag

dheap, 2019-01-05:
JMRI Roster Media image rendering does not utilise the EXIF Orientation tag automatically created by most modern digital camera devices.

As a result, Roster Images are often displayed in an incorrect orientation.

See this jmriusers topic: [Roster Images Upside Down](https://groups.io/g/jmriusers/topic/28909847).

dheap, 2019-01-05:
I can provide test images if required.

dheap, 2019-01-05:
Sample Images. All display correctly in all Mac apps, including Adobe Photoshop.
Don't display well in Chrome web browser I'm using to add them.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

dheap, 2019-01-05:
ExifTool output for above images:
[open_5YLg8fCF.txt](https://github.com/JMRI/JMRI/files/2729185/open_5YLg8fCF.txt)
[open_AaS6NSSh.txt](https://github.com/JMRI/JMRI/files/2729186/open_AaS6NSSh.txt)
[open_bNEAPYBt.txt](https://github.com/JMRI/JMRI/files/2729187/open_bNEAPYBt.txt)
[open_grELXECT.txt](https://github.com/JMRI/JMRI/files/2729188/open_grELXECT.txt)
[open_jhe6guYL.txt](https://github.com/JMRI/JMRI/files/2729189/open_jhe6guYL.txt)
[open_zoOrUyKc.txt](https://github.com/JMRI/JMRI/files/2729190/open_zoOrUyKc.txt)
[open_qwb2Efdf.txt](https://github.com/JMRI/JMRI/files/2729191/open_qwb2Efdf.txt)

dheap, 2019-01-05:
All render correctly in Gmail on Safari and in MacOS Mail app.
But not GitHub on Safari.

bobjacobsen, 2019-01-05:
Perhaps useful: https://stackoverflow.com/questions/47657302/does-imageio-read-take-exif-orientation-metadata-into-account

danielb987, 2019-01-10:
If anyone wants to implement an Exif reader, [JPEG Metadata Format Specification and Usage Notes](https://docs.oracle.com/javase/8/docs/api/javax/imageio/metadata/doc-files/jpeg_metadata.html) seems to have information on how the information is stored.

danielb987, 2019-01-11:
Thanks.

> I read the license as saying we have to include the license file if we redistribute that. I believe we do that for other libraries.

Where do I put the license file?

bobjacobsen, 2019-01-11:
This would be simpler if there was a jar file available somewhere (i.e. that we had not created). Then it’s clear that we could distribute it in our usual way: jar in lib, link to source in lib/README, and Pom.xml entry. 

Is there such a jar file?

bobjacobsen, 2019-01-11:
Looks like the jar is available here: https://github.com/coobird/thumbnailator/wiki/Downloads

danielb987, 2019-01-11:
Yes, the JAR file is available and works fine. My question is there I should put the license info about this package. Is it enough to add a link to the developers GitHub repository? Or where should I put the license info?

danielb987, 2019-01-11:
> Is it enough to add a link to the developers GitHub repository?

In the lib/README.md file.

bobjacobsen, 2019-01-11:
Since we’ll be distributing a pre-made binary, its sufficient to have the link in the lib/README and pom.xml is sufficient. 

Bob

danielb987, 2019-01-11:
PR  #6426 tries to fix this. But it may be too slow. Can anyone test the PR to see if it's fast enough?

PR #6426 (2019-01-12): Read exif information in roster images
Fix #6397.

I get the impression that this PR is slow but I'm not sure if it's my computer that is slow due to too many programs running. It would be good if anyone could check if this PR is too slow.
closed 2019-01-12 by bobjacobsen