JMRI/JMRI #6407: Unexpected mask values in decoder definitions probably represent errors

bobjacobsen, 2019-01-07:
The DecoderPro code only knows how to handle a single, contiguous field in a CV, which it represents with a mask like XXXVVVXX.

A number of files have masks that either aren't contiguous, or don't specify any bits.  These are probably not working as expected.

xml/decoders/Kuehn_N45.xml line 169 Value 'XXVVXXVV' 
xml/decoders/TamValleyDepot_QuadLn_S_11.xml line 148 Value 'XXXXXXXX' 
xml/decoders/Zimo_MX620_v7-8.xml line 1297 Value 'XVXXXVXX' 
xml/decoders/MERG_12A.xml line 310 Value 'XXXXXXXX' 
xml/decoders/zimo/CV100-CV152.xml line 366 Value 'XVXXXVXX' 
xml/decoders/zimo/CV161-CV185Servo.xml line 37 Value 'VVVVVXVV' 
xml/decoders/zimo/CV100-CV152version30.xml line 944 Value 'XVXXXVXX' 
xml/decoders/TamValleyDepot_QuadLn.xml line 36 Value 'XXXXXXXX' 
xml/decoders/Zimo_MX620_v6.xml line 1299 Value 'XVXXXVXX' 
xml/decoders/Zimo_MX63_MX64H_v22-30.xml line 1399 Value 'XVXXXVXX' 
xml/decoders/TD_SC8.xml line 941 Value 'VXXXXXVX' 
xml/decoders/Zimo_MX69MX690.xml line 1403 Value 'XVXXXVXX' 
xml/decoders/Zimo_MX620_v5.xml line 1305 Value 'XVXXXVXX' 
xml/decoders/Zimo_MX62_v22+.xml line 1304 Value 'XVXXXVXX' 
xml/decoders/Zimo_MX620_v9dot12.xml line 1253 Value 'XVXXXVXX' 
xml/decoders/Zimo_MX620_v9.xml line 1296 Value 'XVXXXVXX' 
xml/decoders/Zimo_MX66-2000-11.xml line 364 Value 'XXVXVXXX' 
xml/decoders/esu/v4standardCVs.xml line 352 Value 'VXXXVXXX' 
xml/decoders/Zimo_MX620_v1-3.xml line 1342 Value 'XVXXXVXX' 
xml/decoders/Zimo_MX640_v4.xml line 1401 Value 'XVXXXVXX' 
xml/decoders/TD_SMC4.xml line 638 Value 'VXXXXXVV' 
xml/decoders/Zimo_MX640_v4dot17.xml line 1385 Value 'XVXXXVXX' 
xml/decoders/Zimo_MX640_v3.xml line 1543 Value 'XVXXXVXX' 
xml/decoders/Zimo_MX640_v1-2.xml line 1539 Value 'XVXXXVXX' 

(Note only the 1st find in each file is indicated; there may be more)

AlanUS, 2019-01-09:
I looked at
`xml/decoders/Kuehn_N45.xml line 169 Value 'XXVVXXVV'`

It works as coded with a enum having the following value set (0, 1, 2, 3, 16, 32) that is actually matching the mask. This is an unusual way of doing, but I think that it is supported by the way the enum mask is coded.

This being said, the implementation is not in line with the Kuehn manual for this decoder; I will fix it.

Someone will need to look at the Zimo files, as this non-contiguous scheme  is heavily used according to the list (I'm not familiar with Zimo).

Have not checked either the ones with masks like 'XXXXXXXX', but yes more than likely not to work (or always 0).

AlanUS, 2019-01-09:
`xml/decoders/TamValleyDepot_QuadLn_S_11.xml line 148 Value 'XXXXXXXX'`
This is for a variable named `One Choice Enum` for CV1, not used for display, that effectively only have one value (zero). Guess an unused but mandatory CV1 for short address.

AlanUS, 2019-01-09:
`xml/decoders/MERG_12A.xml line 310 Value 'XXXXXXXX'`
Clearly an error to be fixed.
MERG DCC documentation [here](https://www.merg.org.uk/merg_resources/dcc_download.php)

AlanUS, 2019-01-09:
Only did a quick sampling on the Zimo files, but seems to be a (single) error that has been propagated in several files after many copy and paste...

AlanUS, 2019-01-10:
The issue with the Zimo files is that one information is coded on 2 non contiguous bits of CV124, knowing that the other bits are also in use for other purposes, with more usual continuous bits however.
Therefore a mask with non contiguous bits is useful in this case.

AlanUS, 2019-01-10:
Zimo: 
I have fixed one file (`xml/decoders/Zimo_MX620_v7-8.xml)` as sample (see #6419), by adding the proper value for the third value of the enum, to match the non-contiguous mask. It works now as specified in the manual, with a single information coded overs bits 2 and 6 for CV124.
It remains to be performed in all other Zimo files listed above.

AlanUS, 2019-01-10:
@bobjacobsen 
The code actually accepts non contiguous bits in a mask. However, in order to work properly, the enum values need to be aligned with the mask, i.e. have to take specific values to be indicated explicitly beyond the first group of bits of the mask.
The above check was useful and allowed to find true errors. However, the construct with non contiguous bits mask can be used and is needed in this very rare case (Zimo), though with precautions.

bobjacobsen, 2019-01-10:
@AlanUS thanks for your work on this!  Agree that the non-contiguous case can be useful, but tricky. I’ll update docs & schema to that effect.

AlanUS, 2019-01-10:
Zimo: fixed `xml/decoders/Zimo_MX66-2000-11.xml` : this one was an error. #6419

AlanUS, 2019-01-11:
Zimo: Fixed all occurrences of the 2 non-contiguous bits in CV124 #6419

bobjacobsen, 2019-01-11:
PR #6422 updates the schema to allow non-contiguous bits (i.e. XXVVXXVV) and adds some documentation.  Thanks @ALanUS for figuring that out!

AlanUS, 2019-01-11:
Zimo: fixed `xml/decoders/zimo/CV161-CV185Servo.xml ` and others with similar issues (CV161 and CV149)

AlanUS, 2019-01-11:
> (Note only the 1st find in each file is indicated; there may be more)
Actually there were multiple mask errors in the some of the Zimo files listed above. There could be others.

AlanUS, 2019-01-11:
Team Digital `xml/decoders/TD_SC8.xml` and `xml/decoders/TD_SMC4.xml` - fixed #6419 
- SC8: CV82 to 84
- SCM4: CV72 to 74 - and add new enum values CV71 to 72

AlanUS, 2019-01-11:
Tam Valley Depot `xml/decoders/TamValleyDepot_QuadLn.xml` 
I don't understand the logic of setting `splitVal `in this case: to be revisited by the author. Doing nothing.

AlanUS, 2019-01-11:
@bobjacobsen 
I'm done with this one. All fixed or OK, but the last one (see previous comment).

n3ix, 2019-01-12:
Alain,

 

This template has been in use and working for quite along time.  Could you please explain what the “issue” is that now needs to be revisited?  Thanks.

 

Robin

 

Robin Becker

San Diego CA

bobjacobsen, 2019-01-13:
@n3ix Having a mask with no bits set (XXXXXXXX) isn't something that the code actually handles well:  There are places that assume that changing the value of a variable _has_ _to_ change the value of the CV so that things get notified to change color, get written or read later, etc.

But if that CV is working, then it's fine.  I'll just add a comment to the file to that effect.

AlanUS, 2019-01-13:
@n3ix : In addition to @bobjacobsen comment. Good that it works as you expect; these are your products after all.
However as mask like 'XXXXXXXX' will result in the CV being **always** set to 0

Therefore a statement like:
```
      <variable item="Servo Base Block Lo Change" CV="254" mask="XXXXXXXX" default="0">
        <splitVal highCV="2" upperMask="VVVVVVVV"/>
        <label>Servo Base Block Lo Change</label>
      </variable>
```
will result in CV254 being always set to 0 and all its bits being ignored. Therefore it is equivalent to assigning the value of CV2 in "`Servo Base Block Lo Change`", which I guess is the original intention for debug purposes.

n3ix, 2019-01-13:
@AlanUS: thanks for the feedback.  Hard to remember back that far :) but yes I think that the purpose of that var was just informational.  @bobjacobsen also listed a line in the QuadLN_S_11.xml file.  I think that var was "One Choice Enum".  As I recall that var was part of a hack to get labels to line up properly and it is used in the associated .xsl file.

AlanUS, 2019-01-14:
@n3ix  I have simplified the code for the QuadLn decoder definition, so the 'XXXXXXXX' masks have been removed. I also fixed default values to 0 for the various addresses (like the other QuadLn files), topped the Servo addresses to 2041 (i.e. CV2=255 and CV3=always 0), and fixed the labels of the Hi part (was shown Lo for both Lo and Hi).
This does not lead to any functional change.

@bobjacobsen With #6444 you can close this issue, unless you want to run a second pass of the test to check if all occurrences of this have been effectively fixed (or justified).

n3ix, 2019-01-15:
Changing the servo address max value is not something I want to have happen.  There is already enough confusion with the servo address bit mapping.   In jmri the stationary interrogate turnouts map to addresses 11017-1021 for instance.  This product has been out since 2012 and everything works fine from the user standpoint.  From a product support standpoint I'd much rather live with the current state of the defaults and labels, etc since I get virtually no support requests related to them.  My request is that you _please_ not make any changes that are not absolutely necessary.

AlanUS, 2019-01-15:
@n3ix Sorry I meant, I changed the max value of the Servo start address in the ++tooltip++, to be consistent with what is coded, i.e. 2041 (CV max value = 255, with factor 8 and offset 1).
The tooltip was stating max 2033 (that is CV max value of 254) that is NOT enforced by the code.
Besides the manual is stating a max of 2045, which can only be a typo, as this is not possible given the way this is coded with a factor 8.
Again, I repeat, no functional change, just aligning what is displayed with what is coded.
Now if the max value is effectively 2033, then we need to change to code (if it is possible to impose a max value to a splitVal).
I have added in the tooltip the warning related to addresses 1017-1021, as per the manual.

n3ix, 2019-01-15:
@AlanUS  Ok, thanks for the explanation, sorry I misunderstood your earlier post.  Appreciate that you read the manual, wish more of our customers did that! :)   

Think it is great that you took a close look at the decoder files for errors.   The QuadLN product family are my only model railroad products so I guess I am pretty protective.  (I did the design, doc and provide the tech support.)  In the future if there are any functional changes to the decoder file I would really appreciate getting a chance to review them (via a PR) before they get committed.  Hope that is a reasonable request.  Thanks.

AlanUS, 2019-01-15:
@n3ix The changes in the code are visible in the commits of PR #6444, that is still not merged yet. So you can check if you want, and comment.
However, again these are technical changes fixing errors or confusing patterns like these 'XXXXXXXX' masks; there are many such technical changes happening over time that none of the original authors will ever notice or be notified for.
In this particular case, I notified you because I was not necessarily understanding the original intention behind the code.
I adding these files into JMRI you accepted that the community takes care of them.

n3ix, 2019-01-15:
@AlanUS  Sure I understand how JMRI works, having been a contributor since the early days :)  And I greatly appreciate all the work done by the contributors to this project.  But there have been a couple occasions in the past where a well meaning "technical change" has resulted in a broken decoder file getting out in the wild, so I'm sensitive to this.  Of course you will do what you must,.  FWIW whenever asked I have always been glad to make whatever changes are required to these decoder files.  Thanks.

AlanUS, 2019-01-16:
@n3ix Are you OK with the changes and in particular with my understanding of the max value for the Servo Start address? If yes, then the decoder definition will then work exactly as expected and specified by you, and documentation in tooltip will match reality.
In short, can we close this subject?

n3ix, 2019-01-16:
@AlanUS  Appreciate your patience.  Only comment is that the default values for all the derived address vars was changed to 0.  They were previously 1,2,...8, which are the values that result after the OFFSET was applied.  It probably doesn't matter but wanted to mention that.   Otherwise looks OK to me.  Thanks.

AlanUS, 2019-01-16:
@n3ix The (previous) default values of 1, 2 ,..., 8 eventually resulted in setting CV2 to 8 (last statement of the sequence), therefore having 65 as default value displayed as Servo start address. The "default" qualifier initializes the CV, not the displayed value of the CV that is modified with factor and offset. Now this is as per the original intention - I suppose - to display 1 as default start address, and then display 2, 3, ..., 8 for the following servo addresses.
In summary, only one single CV that is used to display 8 different values (thanks to factor and offset).
File `TamValleyDepot_QuadLn11.xml` for your more recent model, is properly coded with 0 as default value everywhere; so now both are consistent.

n3ix, 2019-01-16:
@AlanUS Thanks for the explanation!  Makes sense, guess I figured it out when I did the QuadLN_S but never went back to fix it in the old QuadLN.  All good.

AlanUS, 2019-01-17:
@bobjacobsen I think that now all is resolved, so that this issue can be closed.

bobjacobsen, 2019-01-19:
@AlanUS  Thank you!

closed 2019-01-19 by bobjacobsen