JMRI/JMRI #6448: 3 Turnout RBG signal head xml fails verify

KenC57, 2019-01-16:
I'm testing this with 4.14 due to a user's question. I created a 3 turnout head and it writes it out ok, but the read fails with:
[image]

I suspect the issue is related to some of the work that @bobjacobsen was doing with enums or similar. I don't have 4.15.2 at hand to test with. Figured first I'd verify for the user. See more in user group "RGB signal 4.14 parse error".
-ken c

bobjacobsen, 2019-01-16:
Here's a little file that demonstrates it.

[RYG-RGB-test.xml.txt](https://github.com/JMRI/JMRI/files/2762636/RYG-RGB-test.xml.txt)

bobjacobsen, 2019-01-16:
#6451 fixes this

PR #6451 (2019-01-16): Fix schema  for RGB SignalHeads
update schema to add "blue" for RGB SignalHeads. Add a test.  See Issue #6448 for background. Closes #6448
closed 2019-01-16 by bobjacobsen