JMRI/JMRI #6469: NEW DECODER definition needed

Hansschaap2, 2019-01-20:
Can anyone help me create a new definition for the Piko Sound decoder that is used in the Dutch 2412 model? I'm trying to make a new entry but it didn't work and I'm no programmer. This sound decoder is version 39 and Piko is manufacturer 162

silverailscolo, 2019-01-21:
For your model Piko.de brought up a link to the "PIKO SmartDecoder 4.1 Sound Next18 für Rh 2400" specs. These look quite similar to the specs for the IntelliSound4 decoders that I added to JMRI fall 2018.
Could you try _reading_ some sound functions from your loco using the Uhlenbrock 76560 decoder definition and report the results?

Hansschaap2, 2019-01-26:
I've tried the Uhlenbrock 76560 decoder. but then I found no sound options. So I tried the 35650 decoder and I can see sound definitions tabs. The throttle and F buttons are reacting but I can't change the sounds under the F buttons( There are 25 sound functions). I'll keep trying or is there an easier way to change them(under https://www.piko-shop.de/index.php?vw_type=artikel&vw_id=25658&vw_name=detail you can find the CV definitions and the sound definitions)

AlanUS, 2019-01-26:
As I wrote earlier, I suspect this decoder to be the same as the latest generation (4.1) of Uhlenbrock decoder [reference 73235](https://uhlenbrock.de/de_DE/produkte/decoder/I26ABAB1-001.htm) for which the manual can be[ downloaded here](https://uhlenbrock.de/de_DE/service/download/handbook/de/Bes73235_gesamt.pdf).
The definition for this newest generation of Uhlenbrock decoders still needs to be designed for JMRI.

It is most probably coupled to an Uhlenbrock Sound Module (via the SUSI interface). So probably is it required to go through 2 different decoder definition depending whether you want to address the Sound CVs or the others.
The most recent Sound Modules ([32410/4](https://uhlenbrock.de/de_DE/service/download/handbook/de/Bes32410_32415.pdf) and [32415/6](https://uhlenbrock.de/de_DE/service/download/handbook/de/Bes32410_32415.pdf)) are not yet in JMRI either.

silverailscolo, 2019-01-26:
Hi Hans,
Did you mean to write/test Uhlenbrock 36560 (instead of 35650, which I can’t find in DecoderPro, perhaps a typo)?
Once open, please look at the rightmost tab, called Sound Fn Map. It is supposed to allow you changing the functions for each Function button on your command station.
Simple mapping is on the Function Map tab.

Egbert

> Op 26 jan. 2019, om 09:37 heeft Hansschaap2 <notifications@github.com> het volgende geschreven:
> I've tried the Uhlenbrock 76560 decoder. but then I found no sound options. So I tried the 35650 decoder and I can see sound definitions tabs. The throttle and F buttons are reacting but I can't change the sounds under the F buttons( There are 25 sound functions). I'll keep trying or is there an easier way to change them(under https://www.piko-shop.de/index.php?vw_type=artikel&vw_id=25658&vw_name=detail <https://www.piko-shop.de/index.php?vw_type=artikel&vw_id=25658&vw_name=detail> you can find the CV definitions and the sound definitions)
>

silverailscolo, 2019-01-27:
Hi Hans, the names of the sounds (like “Horn Hoch” for F3) might be fixed in the decoder definition, and different from what Piko put in that spot. Will see how I can include the Fn sound names for your decoder following this table from the Piko 4.1 Rh2400 specs:

Until that’s working please note that these names are just labels, so you could experiment picking eg. F3 or F14 without worrying too much about the name.
Egbert

> Op 26 jan. 2019, om 19:18 heeft Hansschaap2 <notifications@github.com> het volgende geschreven:
> Hi Egbert, indeed it was a typo. But still I can't change the F keys. Must be something I do wrong but I cannot find it. The tab Sound Fn Map is there, but the list of sounds is not the same as the Piko decoder definition states. I am now reading the decoder with the NMRA raw CV's.See if that will do something.
>

silverailscolo, 2019-01-31:
The Piko 4.1 sheet corresponds to most of the Uhlenbrock 732x specs. Quite a change from the 365x CVs. Will start to create a new Uhlenbrock/Piko combined definition in February, unless someone has already picked this up.

AlanUS, 2019-01-31:
> 
> 
> The Piko 4.1 sheet corresponds to most of the Uhlenbrock 732x specs. Quite a change from the 365x CVs. Will start to create a new Uhlenbrock/Piko combined definition in February, unless someone has already picked this up.

This is in my to-do list, but will low priority, so please go ahead.

Hansschaap2, 2019-02-02:
Thanks all for trying and make the definition of the Piko sound decoder soon

silverailscolo, 2019-02-20:
First development version is in PR #6585. Could you give it a try @Hansschaap2 ?

Hansschaap2, 2019-03-03:
Hello Egbert,

Though I tried to import the new definitions it didn’t work.

How can I import the new WIP xml?

 

Deze e-mail is gecontroleerd op virussen door AVG.
http://www.avg.com

silverailscolo, 2019-03-03:
The decoder definition needs some files from the decoders/uhlenbrock/ folder as well.
Do you have GitHub running, with your own branch of JMRI?

Still have to add some extra stuff, will try to look at it the coming evenings.

Egbert

silverailscolo, 2019-03-09:
@Hansschaap2 the definition for your decoder, and related 2018 IntelliDrive2 models, was merged with PR #6585 and will be in the next JMRI test release. Look under Uhlenbrock as Manufacturer, where you will see the PIKORH2400. Can you do some testing? Not all vars are in place yet, and the Sound vars might show odd values as their documentation is limited. (I suggest not writing whole sheets initially, reading would be OK and can learn us the default values that I can fix)

Hansschaap2, 2019-04-08:
Hello All and especially Egbert,
I've tried to change the function buttons and change the  sounds settings. It didn't work and afterall in JMRI the loco on the programmer track  just gave the 301 message. So I contacted the dutch importer and he is sending the loco back to Piko for a reset. As a service he also wanted a list of which button needed to steer which sound. He also mentioned that he wished the new loco decoder from Piko was already on the market so that all shops that have one could heave done this(PIKO Loc-Programmer ). So when I've got my loco back inclusive of the new sound order I will read the values in JMRI and will send them to you so that you could finish the loco definition in the xml

Hansschaap2, 2019-04-16:
Egbert I finally got an answer from piko about the questions you asked them
Ich habe eine Frage betrifft die Sound Variablen CV 900-931.
Die Zugang mit Hilfe von CV 1021 ist (in DecoderPro) kein Problem. Aber die gültige Werte jedes CVs sind im ausgebreiteten Datenblatt nicht abgedruckt, z.B.
905.2 (Bank 2, Minimale Zufallszeit) -> **ab welcher Mindestzeit ein Zufallsgeräusch abgespielt werden soll.**
906.2 (…) -> bis zu welcher Zeit ein Zufallsgeräusch spätestens abgespielt werden soll.
CV 905.2 und 906.2 **bestimmen also, wie häufig ein Zufallsgeräusch (z.B. Lüfter) abgespielt werden soll.**
907.2 -> 
908.2 -> 

910.2 (ein Zahl von 0 - 28 für F0-F28??) -> **richtig, FunctionMapping für Soundfader**
911.2 (Idem) -> wie 910.2, **nur für Lautstärkeregelung**
 
Bank 3: vielleicht ein Ganzzahl von 0-28 für F0-F28? -> **nein, hier können die Fahrstufenschwellen für z.B. das Getriebeschalten eines Schienenbusses, oder die Schützschaltung für eine Altbau E-Lok eingestellt werden.**

Bank 4: 0 bis maximal ?? -> **Hier können die Lautstärken der einzelgeräusche eingestellt werden**

Für Bank 8 meldet die Dokumentation CVs 99-931 aber ich bin überzeugt das sei einen Eintragsfehler und soll gleich wie die andere Bank 900.8-931.8 lauten. Das betrifft also 32 Sound Slots, Wert 0-28 für F0-F28? -> **richtig, das muss natürlich CV 900 – 931 heißen. Das ist das Function Mapping für die einzelenen Soundslots (Einzelgeräusche)**

Da es sich bei der Anleitung um eine allgemein gültige Anleitung handelt, können die Einzelgeräusche und die speziellen Bedeutungen für diese Lok nicht näher genannt werden.

silverailscolo, 2019-04-16:
Thanks, Hans.
I will have to go back and look at the meaning of these answers, but at first glance this response from the manufacturer Piko looks really helpful to complete this decoder definition.

silverailscolo, 2019-04-20:
PR #6822 implements the info quoted above _and_ provides early access to part of the existing Uhlenbrock 732x definition for use with new Piko decoder models.
For the longer term it would be good to:

- [ ] split out the core CV code in , used in both sound and non-sound decoder definitions, to a series of includes to present both types under a more recognizable model name without allowing too much code duplication that is realy hard to maintain or i18n

- [ ] make Piko into a separate manufacturer for easier user access

closed 2021-02-23 by silverailscolo