JMRI/JMRI #6471: Rapido sw1200rs

187king, 2019-01-21:
I have a 1200 sound unit and both sound tabs are blank.  I am using Lenz 3.6 with powerpax and LI-USB, decoderpro 4.14, win10.  I get a lot of 303 errors when reading cv's.
Suggestions please, Sorry new post from new user
Greg

pabender, 2019-01-21:
I don’t know anything about the decoder you are trying to use, but I do know about Lenz systems, so I want to see if we can figure out what is going on when you are reading CVs.

That error 303 you are getting is in response to a command station message that says the operation you are trying to do isn’t supported, so we know basic communication is working.
Do here are some questions:
1) what command station are you using
2) what version of the software is in the command station ( you said 3.6, but is that the command station or the computer interface ).  You can use the System Information tool in the XpressNet menu to find this information.
3) do you have trouble programming any other decoders?

pabender, 2019-01-21:
@187king the attachment didn't come through.

Can you log into github and post it to the issue you created  (rather than replying to the e-mail).

187king, 2019-01-21:
Screen print
[image]

pabender, 2019-01-22:
What programming mode are you using?

187king, 2019-01-22:
Paul, I just tried paged and it read all cv’s but sound tabs still blank.  No more 303 errors.  I mentioned the decoder from the manual before but the closest I could find was ESU lokpilot V4.0 DCC.  Could that be the problem?
I have update java to latest, updated decoderpro to v4.14. 

Not sure what else to do?

187king, 2019-01-23:
First, I would like to offer a huge thanks to the group for getting me figured out.  Because the Rapido  manual indicated the decoder was a lokpilot, I selected one that I thought would be close.  Turned out I picked a non-sound decoder, so of course the tabs would be blank (my bad).

I do have a couple more questions:

In the past when I have updated versions of decoderpro, it always adds to the menu at start up (ie 4.8, 4.12, etc)  When I added 4.14 it shows active profile 4.8 as picture shows in bottom right.

[image]

 
When I show about it indicates it is 4.14
[image]

Dave asked me if the indent new loco does not uniquely id the decoder, to go into system console -  look for Found mfgid, version, etc – there is no such wording?

Once again thanks for helping the obviously uninformed.   

Greg

bobjacobsen, 2019-01-23:
There are two separate things here:  JMRI's own version (which is 4.14+Rd060e0b in this case) and what the profile directory is called, currently DecoderPro 4.8.   

JMRI controls its version name. That changes when you update to a new one.

The profile directory name is whatever it is; it could have been called "My Profile" or something else.  It doesn't change when JMRI is updated.  It doesn't control the JMRI version.

dheap, 2019-01-23:
> @187king Dave asked me if the indent new loco does not uniquely id the decoder, to go into system console - look for Found mfgid, version, etc – there is no such wording?

You need to use this button:
[image]

Followed by this button:
[image]

Then go to:
[image]

You should see something like this:
[image]
Or as text:
`2019-01-24 10:25:30,004 symbolicprog.CombinedLocoSelTreePane  WARN  - Found mfg 123 (Massoth Elektronik, GmbH) version 123; no such decoder defined [AWT-EventQueue-0]
`

We need to know that text...

dheap, 2019-01-23:
Sorry, just realised you need JMR V4.15.1 or later to always get a log entry. Earlier versions only logged that line under some circumstances.

Please update to V4.15.2 if you don't see the line.

187king, 2019-01-24:
I updated to 4.15.2 and here is a partial screen print from the log.  It appears not supported (303)

[image]

dheap, 2019-01-24:
That indicates a command station issue, not a decoder issue. What Programming Mode options are available in the Create New Loco Window?
See screen grab at https://github.com/JMRI/JMRI/issues/6471#issuecomment-457008926.

If any Direct Mode is available, try that. Don't use Register Mode under any circumstances

This may be something for a Lenz expert...

It would help greatly if you used the "Copy to Clipboard" button at the bottom of the console log window and pasted the whole log into a comment for us to see. Alternatively, go to Help->Locations->Open Log Files Location and attach the session.log file to a comment.

187king, 2019-01-26:
Here are three log files.  First direct byte:
[session.log](https://github.com/JMRI/JMRI/files/2799562/session.log)

Here is the direct bit log:
[session.log](https://github.com/JMRI/JMRI/files/2799564/session.log)

Here is the paged log:
[session.log](https://github.com/JMRI/JMRI/files/2799566/session.log)

I can see the device reading the CV's because the ground lights on the engine come on and the cv numbers at the bottom of the screen change and the data light on the USB connection comes on.
Thanks again for any assistance.

dheap, 2019-01-27:
@187king Thanks for the log files.I think I know what is going on and I'm looking into the code

Watch this space...

dheap, 2019-01-27:
@187king I've decided that the avenue I was following is most likely a blind alley.

CV261y research on the [Rapido Site](https://www.rapidotrains.com/ho-gmd-sw1200rs-support/) indicates a LokSound Select decoder, so you should be able to Read CV261 and other high CVs (>256).

Please check that your Programming Preferences match the screen below exactly:
[image]

If they don't match, make changes and try again.

If no success, I'm suspecting a problem in the Lenz area...

dheap, 2019-01-27:
I'm not familiar with Lenz systems, but my feeling is that `jmri.​jmrix.​lenz.​XNetProgrammer.getCanWrite("261")` is returning `false` for reasons justifiable/unjustifiable and that this is causing `jmri.​jmrix.​lenz.​XNetProgrammer.readCV("261")` to notify a 303 (not implemented) error.

My reading of the code and the System Information provided by @187king should mean that a read of CV261 is possible. CVs>256 are used extensively by all ESU LokSound/Lokpilot decoders.

@pabender, @bobjacobsen Could you take a look at this please?

pabender, 2019-01-27:
Sent from my iPad

> On Jan 26, 2019, at 8:32 PM, Dave Heap <notifications@github.com> wrote:
> @187king I've decided that the avenue I was following is most likely a blind alley.
> 
> CV261y research on the Rapido Site indicates a LokSound Select decoder, so you should be able to Read CV261 and other high CVs (>256).
> 
<snip>
> If they don't match, make changes and try again.
> 
> If no success, I'm suspecting a problem in the Lenz area...
> 

Version 3.6 Lenz system can only read/write CVs above 255 using direct mode.

None of the JMRI code involved in generating the direct mode reads and write has changed in 4 years, and I have used it since then to program an Intermountain SD40-2 with sound.

But yet the 303 code being seen is generated because the command station replied the operation was not permitted.

I will pull out a decoder that actually has high CVs and verify operation.

More later tonight.

pabender, 2019-01-27:
> I'm not familiar with Lenz systems, but my feeling is that jmri.​jmrix.​lenz.​XNetProgrammer.getCanWrite("261") is returning false for reasons justifiable/unjustifiable and that this is causing jmri.​jmrix.​lenz.​XNetProgrammer.readCV("261") to notify a 303 (not implemented) error.
The 303 is triggered by the command station.  On a version 3.6 Lenz system, getCanWrite and getCanRead will return true for CV261 in direct mode.
> My reading of the code and the System Information provided by @187king should mean that a read of CV261 is possible. CVs>256 are used extensively by all ESU LokSound/Lokpilot decoders.
> 
> @pabender, @bobjacobsen Could you take a look at this please?

I am looking into it.

dheap, 2019-01-27:
> Version 3.6 Lenz system can only read/write CVs above 255 using direct mode.

Which the code indicates. But there is some version checking code there & I wonder if that's where @187king's case is coming unstuck. 

> None of the JMRI code involved in generating the direct mode reads and write has changed in 4 years, and I have used it since then to program an Intermountain SD40-2 with sound.

Which may point to a specific circumstance with @187king's setup... 

> But yet the 303 code being seen is generated because the command station replied the operation was not permitted.

My assumption is that the 303 is being generated at lines 252-254 of XNetProgrammer because `getCanRead(cv) ` is returning false.

> I will pull out a decoder that actually has high CVs and verify operation.

Thanks.

dheap, 2019-01-27:
@pabender Sorry, our posts crossed. I will leave you to it...

pabender, 2019-01-27:
Here is the results of reading a few CVs (not all sheets or even one sheet, that takes too long)

First, CV1, just to see that the decoder responds correctly.  

```
22:07:37.315: [packet:22 15 01 36]   Service Mode Request: Read CV 1 in Direct Mode
22:07:37.322: [61 02 63]   Broadcast: Service Mode Entry
22:07:37.358: [61 02 63]   Broadcast: Service Mode Entry
22:07:38.329: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
22:07:38.331: [packet:21 10 31]   REQUEST: Service Mode Result
22:07:38.387: [63 14 01 03 75]   Service Mode: Direct Programming Mode Response: CV:1 Value:3
22:07:38.807: [packet:21 81 A0]   REQUEST: Normal Operations Resumed
22:07:38.808: [61 01 60]   Broadcast: Normal Operations Resumed
22:07:38.826: [61 01 60]   Broadcast: Normal Operations Resumed
22:07:38.831: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
```

And then reading CV16.2.261
```
22:08:41.583: [packet:23 16 1F 10 3A]   Service Mode Request: Write 16 to CV 31 in Direct Mode
22:08:41.583: [61 02 63]   Broadcast: Service Mode Entry
22:08:41.601: [61 02 63]   Broadcast: Service Mode Entry
22:08:42.036: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
22:08:42.045: [packet:21 10 31]   REQUEST: Service Mode Result
22:08:42.079: [63 14 1F 10 78]   Service Mode: Direct Programming Mode Response: CV:31 Value:16
22:08:42.094: [packet:23 16 20 02 17]   Service Mode Request: Write 2 to CV 32 in Direct Mode
22:08:42.580: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
22:08:42.587: [packet:21 10 31]   REQUEST: Service Mode Result
22:08:42.612: [63 14 20 02 55]   Service Mode: Direct Programming Mode Response: CV:32 Value:2
22:08:42.647: [packet:22 19 05 3E]   Service Mode Request (V3.6): Read CV 261 in Direct Mode
22:08:43.635: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
22:08:43.642: [packet:21 10 31]   REQUEST: Service Mode Result
22:08:43.660: [63 15 05 00 73]   Service Mode: Direct Programming Mode Response: CV:261 Value:0
22:08:45.123: [packet:21 81 A0]   REQUEST: Normal Operations Resumed
22:08:45.124: [61 01 60]   Broadcast: Normal Operations Resumed
22:08:45.143: [61 01 60]   Broadcast: Normal Operations Resumed
22:08:45.154: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
```
And then writing the 0 back to CV16.2.261:

```
22:17:20.857: [packet:23 16 1F 10 3A]   Service Mode Request: Write 16 to CV 31 in Direct Mode
22:17:20.857: [61 02 63]   Broadcast: Service Mode Entry
22:17:20.861: [61 02 63]   Broadcast: Service Mode Entry
22:17:21.123: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
22:17:21.126: [packet:21 10 31]   REQUEST: Service Mode Result
22:17:21.146: [63 14 1F 10 78]   Service Mode: Direct Programming Mode Response: CV:31 Value:16
22:17:21.148: [packet:23 16 20 02 17]   Service Mode Request: Write 2 to CV 32 in Direct Mode
22:17:21.627: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
22:17:21.629: [packet:21 10 31]   REQUEST: Service Mode Result
22:17:21.635: [63 14 20 02 55]   Service Mode: Direct Programming Mode Response: CV:32 Value:2
22:17:21.636: [packet:23 1D 05 00 3B]   Service Mode Request (V3.6): Write 0 to CV 261 in Direct Mode
22:17:22.117: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
22:17:22.119: [packet:21 10 31]   REQUEST: Service Mode Result
22:17:22.126: [63 15 05 00 73]   Service Mode: Direct Programming Mode Response: CV:261 Value:0
22:17:22.401: [packet:21 81 A0]   REQUEST: Normal Operations Resumed
22:17:22.408: [61 00 61]   Broadcast: Emergency Off (short circuit)
22:17:22.413: [61 00 61]   Broadcast: Emergency Off (short circuit)
22:17:22.418: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
```

But now I also see what happened.

After the short circuit, I attempted a read (I didn't capture it the first time, so had to do it again, hence the timestamp differences):

```
22:26:33.515: [packet:23 16 1F 10 3A]   Service Mode Request: Write 16 to CV 31 in Direct Mode
22:26:34.057: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
22:26:34.063: [packet:21 10 31]   REQUEST: Service Mode Result
22:26:34.266: [61 82 E3]   XpressNet Instruction not supported by Command Station
[... 2 lines cut ...]
It's that last message that causes the 303 error.

How we got to this state was that the command station issued a short circuit broadcast instead of the expected normal operation resumed (see the end of the two reads above).  

This isn't the expected behavior and could indicate there was a short detected (momentary or otherwise) on the layout or on the programming track.  We (JMRI) really doesn't know why the short circuit broadcast occurred, and I am not going to speculate a reason.

It looks like the command station is in a strange state at this point.  It isn't in normal operation mode, but it isn't in programming mode either.  The reply to the read request (the 01 04 05 message) indicates the command station processed the read request and processed it.  This is normal behavior for an LZV100/LZ100 command station in service mode.  

Since JMRI didn't see the expected broadcast message saying normal operations was resumed, JMRI also still believes we are in service mode, so that 01 04 05 message is enough to trigger the request for service mode result [... cut ...]
The 90 second delay between the read request and the request to resume normal operations is specified by Lenz as the required delay to recover from an error in service mode.

At this point, you can issue read and write requests again.  Depending on exactly how the read/write was triggered, JMRI may make a second attempt at reading the CV.

I won't speculate on the exact reason why the short circuit broadcast is sent, but the current actions JMRI takes to recover are appropriate given the messages received.

pabender, 2019-01-27:
> > But yet the 303 code being seen is generated because the command station replied the operation was not permitted.
> 
> My assumption is that the 303 is being generated at lines 252-254 of XNetProgrammer because `getCanRead(cv) ` is returning false.
> 

No, the getCanRead and getCanWrite returning false should prevent the attempt to read/write the CV, so you would never see the 303.

dheap, 2019-01-27:
> > My assumption is that the 303 is being generated at lines 252-254 of XNetProgrammer because `getCanRead(cv) ` is returning false.
> 
> No, the getCanRead and getCanWrite returning false should prevent the attempt to read/write the CV, so you would never see the 303.

My bad there. I was mis-observing. The actual code was `getCanRead() ` not `getCanRead(cv) `. A big difference...

dheap, 2019-01-27:
@pabender Thanks. But where does this leave @187king in terms of his attempt to identify and read/write his decoder which seems to be an ESU, from the IdentifyDecoder path taken (the only time CV261 is read) and almost certainly a LokSound Select (from the Rapido site) using his Lenz system?

To date I've never seen reports of a similar problem with ESU decoders...

@pabender Was the decoder you tested an ESU decoder? If so, what does attempting "Read Type from Decoder" do?

I noted @187king  is using a PowerPax program track booster.  I've seen reports that no booster (or a SoundTraxx PTB-100) gives better results with ESU decoders. ESU says no booster is needed for current ESU decoders.

pabender, 2019-01-27:
Sent from my iPhone

> On Jan 26, 2019, at 11:55 PM, Dave Heap <notifications@github.com> wrote:
> @pabender Thanks. But where does this leave @187king in terms of his attempt to identify and read/write his decoder which seems to be an ESU, from the IdentifyDecoder path taken (the only time CV261 is read) and almost certainly a LokSound Select (from the Rapido site) using his Lenz system?
> 
The IdentityDecoder method needs to be smart enough to retry the read on a failure.  If it is, then the process should complete normally after error recovery.  ( you just have to remember that if an error occurs, there may be a 90 second delay ).
> To date I've never seen reports of a similar problem with ESU decoders...
> 
> @pabender Was the decoder you tested an ESU decoder? If so, what does attempting "Read Type from Decoder" do?
> 
It is a factory installed ESU LokSound in an Intermountain n-scale SD40-2. 

I won’t be able to check what read type from decoder does until tomorrow, but I don’t recall there being any issue when I put it into the roster in the first place ( 2 years ago ).
> I noted @187king is using a PowerPax program track booster. I've seen reports that no booster (or a SoundTraxx PTB-100) gives better results with ESU decoders. ESU says no booster is needed for current ESU decoders.
> 
The Lenz system @187king is using should never need a programming track booster.  That has always been Lenz’s recommendation.

dheap, 2019-01-27:
> @pabender The IdentityDecoder method needs to be smart enough to retry the read on a failure. If it is, then the process should complete normally after error recovery. ( you just have to remember that if an error occurs, there may be a 90 second delay ).

The AbstractIdentify code implements jmri.ProgListener and currently makes 3 attempts to read/write every CV (only after receiving a non-OK status reply from the relevant programmer via programmingOpReply). After the third try it switches to Paged Mode (after checking getSupportedModes to see it is available, it currently doesn't check getCanRead(cv) or getCanWrite(cv) ) and makes 3 more attempts before aborting (and restoring the original mode).

We make the assumption that once programmingOpReply has been received, the programmer is ready for the next command.

My opinion would be that to inflict a 90 second delay before every one of 3 to 6 retries for every DCC system would be an unfair burden on non-XpressNet users.

Can anyone suggest a way forward?

I suspect the same problem will  apply to decoder Read/Write Sheet as well.

pabender, 2019-01-27:
Ok, so I couldn't sleep, and I wanted to see what the read type from decoder did with that locomotive I was reading earlier.  The decoder was identified as a LokSound Select direct/micro. No errors were observed during identification.  XpressNet log follows:

```
02:56:26.880: [packet:22 15 08 3F]   Service Mode Request: Read CV 8 in Direct Mode
02:56:26.881: [61 02 63]   Broadcast: Service Mode Entry
02:56:26.887: [61 02 63]   Broadcast: Service Mode Entry
02:56:27.850: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
02:56:27.852: [packet:21 10 31]   REQUEST: Service Mode Result
02:56:27.879: [63 14 08 97 E8]   Service Mode: Direct Programming Mode Response: CV:8 Value:151
02:56:27.885: [packet:22 15 07 30]   Service Mode Request: Read CV 7 in Direct Mode
02:56:28.867: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
02:56:28.868: [packet:21 10 31]   REQUEST: Service Mode Result
02:56:28.879: [63 14 07 FF 8F]   Service Mode: Direct Programming Mode Response: CV:7 Value:255
02:56:28.883: [packet:23 16 1F 00 2A]   Service Mode Request: Write 0 to CV 31 in Direct Mode
02:56:29.373: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
02:56:29.375: [packet:21 10 31]   REQUEST: Service Mode Result
02:56:29.379: [63 14 1F 00 68]   Service Mode: Direct Programming Mode Response: CV:31 Value:0
02:56:29.384: [packet:23 16 20 FF EA]   Service Mode Request: Write 255 to CV 32 in Direct Mode
02:56:29.866: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
02:56:29.868: [packet:21 10 31]   REQUEST: Service Mode Result
02:56:29.882: [63 14 20 FF A8]   Service Mode: Direct Programming Mode Response: CV:32 Value:255
02:56:29.896: [packet:22 19 05 3E]   Service Mode Request (V3.6): Read CV 261 in Direct Mode
02:56:30.900: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
02:56:30.907: [packet:21 10 31]   REQUEST: Service Mode Result
02:56:30.971: [63 15 05 80 F3]   Service Mode: Direct Programming Mode Response: CV:261 Value:128
02:56:30.979: [packet:22 19 06 3D]   Service Mode Request (V3.6): Read CV 262 in Direct Mode
02:56:31.996: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
02:56:31.997: [packet:21 10 31]   REQUEST: Service Mode Result
02:56:32.011: [63 15 06 00 70]   Service Mode: Direct Programming Mode Response: CV:262 Value:0
02:56:32.017: [packet:22 19 07 3C]   Service Mode Request (V3.6): Read CV 263 in Direct Mode
02:56:32.989: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
02:56:32.990: [packet:21 10 31]   REQUEST: Service Mode Result
02:56:33.001: [63 15 07 00 71]   Service Mode: Direct Programming Mode Response: CV:263 Value:0
02:56:33.007: [packet:22 19 08 33]   Service Mode Request (V3.6): Read CV 264 in Direct Mode
02:56:33.984: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
02:56:33.986: [packet:21 10 31]   REQUEST: Service Mode Result
02:56:33.997: [63 15 08 02 7C]   Service Mode: Direct Programming Mode Response: CV:264 Value:2
02:56:37.890: [packet:22 15 1D 2A]   Service Mode Request: Read CV 29 in Direct Mode
02:56:38.741: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
02:56:38.743: [packet:21 10 31]   REQUEST: Service Mode Result
02:56:38.754: [63 14 1D 2A 40]   Service Mode: Direct Programming Mode Response: CV:29 Value:42
02:56:38.759: [packet:22 15 01 36]   Service Mode Request: Read CV 1 in Direct Mode
02:56:39.782: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
02:56:39.784: [packet:21 10 31]   REQUEST: Service Mode Result
02:56:39.810: [63 14 01 03 75]   Service Mode: Direct Programming Mode Response: CV:1 Value:3
02:56:39.816: [packet:22 15 11 26]   Service Mode Request: Read CV 17 in Direct Mode
02:56:40.802: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
02:56:40.808: [packet:21 10 31]   REQUEST: Service Mode Result
02:56:40.814: [63 14 11 D8 BE]   Service Mode: Direct Programming Mode Response: CV:17 Value:216
02:56:40.821: [packet:22 15 12 25]   Service Mode Request: Read CV 18 in Direct Mode
02:56:41.807: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
02:56:41.808: [packet:21 10 31]   REQUEST: Service Mode Result
02:56:41.822: [63 14 12 B8 DD]   Service Mode: Direct Programming Mode Response: CV:18 Value:184
02:56:41.824: [packet:22 15 13 24]   Service Mode Request: Read CV 19 in Direct Mode
02:56:42.825: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
02:56:42.827: [packet:21 10 31]   REQUEST: Service Mode Result
02:56:42.837: [63 14 13 00 64]   Service Mode: Direct Programming Mode Response: CV:19 Value:0
02:56:42.945: [packet:21 81 A0]   REQUEST: Normal Operations Resumed
02:56:43.098: [61 01 60]   Broadcast: Normal Operations Resumed
02:56:43.104: [61 01 60]   Broadcast: Normal Operations Resumed
02:56:43.120: [01 04 05]   Command Successfully Sent/Normal Operations Resumed after timeout
```

dheap, 2019-01-27:
Storms here so no access  to code.

>That process definitely needs to be modified to check getCanRead(cv) or getCanWrite(cv).
That's not hard to add.

> The XPressNet programmer was mostly written before we could program CVs over 256, and so when you ask the XpressNet programmer to read/write a CV in paged mode, it just sends the least significant byte of the CV to the command station. A paged mode read of CV261 on a Lenz system is really reading CV5
> We have a similar situation with NCE Power Pro Service  Mode  writes. We trap that using a Facade when using ESU definitions and kick in an ESU-provided low-CV workaround. But we don't have access to that when doing IdentifyDecoder.

> I suspect it isn't the only programmer that behaves similarly.

The Single CV Programmer comes to mind...

> In the example I showed traces from, the programmer will actually show as busy until the timeout expires (that is the programmerBusy() method will return true) and there shouldn't be a result passed to the programmer listeners until after the timeout. We shouldn't have any new programming requests issued as long as that is true.

I can look into that tomorrow...

> What I need right now is for @187king to give us a monitor log of a session that includes the 303 on his system. I am not sure I will see it on mine.

If we could talk him through putting a custom defaults.lcf file in "Settings Location" that enables debugging on IdentifyDecoder and AbstractIdentify we could get a lot more I formation about retries and the state of the identify machine.

I doubt I've got the resources on this old iPad to create a custom file...

dheap, 2019-01-27:
Storms here so no access  to code.

>That process definitely needs to be modified to check getCanRead(cv) or getCanWrite(cv).
That's not hard to add.

> The XPressNet programmer was mostly written before we could program CVs over 256, and so when you ask the XpressNet programmer to read/write a CV in paged mode, it just sends the least significant [...]

> I suspect it isn't the only programmer that behaves similarly.

The Single CV Programmer comes to mind...

> In the example I showed traces from, the programmer will actually show as busy until the timeout expires (that is the programmerBusy() method will return true) and there shouldn't be a result passed [...]

I can look into that tomorrow...

> What I need right now is for @187king to give us a monitor log of a session that includes the 303 on his system. I am not sure I will see it on mine.

If we could talk him through putting a custom defaults.lcf file in "Settings Location" that enables debugging on IdentifyDecoder and AbstractIdentify we could get a lot more I formation about retries and the state of the identify machine.

I doubt I've got the resources on this old iPad to create a custom file...

dheap, 2019-01-27:
Can't get upload into comment to work on this old iPad. Here's a link if someone has time to upload to comment and explain how to put in "Settings Location". Even typing into a comment is painfully slow... 

https://www.dropbox.com/s/8eduqq0iicv64pc/default.zip?dl=0

Otherwise I'll do it tomorrow my time.

pabender, 2019-01-27:
> If we could talk him through putting a custom defaults.lcf file in "Settings Location" that enables debugging on IdentifyDecoder and AbstractIdentify we could get a lot more I formation about [...]

I really do want to see the XpressNet message sequence.  There is no need to modify default.lcf for that.

http://jmri.org/help/en/html/hardware/XPressNet/CreateXNetLogFile.shtml

187king, 2019-01-27:
Here is a copy of the log you suggested

[RapidoLog.txt](https://github.com/JMRI/JMRI/files/2800152/RapidoLog.txt)

pabender, 2019-01-27:
The log shows something we can’t fix.

Your command station reports that it is version 3.6, but by rejecting the high cv programming commands,  it is acting like it is a version 3.5 command station.

I have been working with Lenz equipment for a very long time and I have never seen that occur.

Can you generate a log showing what happens when you press the getvsystem info in the system information tool.  I need to make sure that matches.

187king, 2019-01-27:
Not sure. . . .  here are a couple of files
[image]
 

[image]

pabender, 2019-01-27:
It definitely is reporting 3.6 then.

I only have two things to suggest.

First, reset your command station.  Something could be corrupted.  Then try the identification process again.

Second, if that still doesn’t work, verify you can read and write CV261 with a handheld controller.

If that doesn’t work, there is something wrong with your system. You will need to contact Lenz, as there isn’t anything JMRI can do to resolve the issue.

pabender, 2019-01-27:
Reset instructions are in the manual.  You can also reset from the LZV100 configuration tool in JMRI.

187king, 2019-01-27:
I reset command station.  With handheld I can read cv 261 - returns value 137 but when I try and write a new value,  I get a err02.

pabender, 2019-01-27:
What happens with the read type from decoder after the reset?

I am not worried about the ERR02 on the write, that may be a read only CV.

187king, 2019-01-27:
After reset, no differences
Will approach vendor in Ontario where I purchased and had V3.6 update completed and see if he has any suggestions.
Greatly appreciate all efforts trying to solve my issues.  If I hear or get any updates from my fooling around or from the Vendor, will try and provide update. 
Greg

pabender, 2019-01-27:
This shouldn’t make a difference, but have you tried this without the powerpax installed?

187king, 2019-01-27:
No will try in about an hour
Greg

187king, 2019-01-27:
No powerpax installed

[image]

pabender, 2019-01-27:
Hmmm. Nterestung that it doesn’t even see CV 8.  I guess you really do need something.
 
Section 5.3 of the L.zV100 manual mentions using a 10 ohm resistor to help with inrush currents.

I really don’t think this is the issue however.

After that, I am all out of ideas.

dheap, 2019-01-27:
> I reset command station. With handheld I can read cv 261 - returns value 137 but when I try and write a new value, I get a err02.

As @pabender suggested, CV0.255.261 is a readonly CV. Its one byte of the 4 byte ProductID.

closed 2021-04-14 by stale[bot]