JMRI/JMRI #6473: What information does a LocoNet transponding report carry?

pabender, 2019-01-21:
I'm working on issues related to reporters (specifically those raised in PR #6463 and issue #1834) and I am currently looking at the LocoNet reporter.

Amy I correct that the only information really contained in the Transponding report is the address.  

I know there are bits for "enter" and "exit", but really that just tells us if the reported address should be displayed by the reporter as the current report or not.  At least some of the non-Digitrax implementations (I'm using a Z21 for testing) don't handle the exit correctly anyway... (the Z21 always sends "0 exits").

If no other information is carried in the report, then I am going to propose changing the LocoNet reporters so they report IdTag objects with the TagID set to the address reported in the report.   The "enter" reports will be used to update the location of the tag and the current report.  The exit reports (if they are sent) will be used to clear the current report.

bobjacobsen, 2019-01-21:
The LISSY messages on LocoNet can say whether  the observed loco is "northbound" or "southbound". See java/src/jmri/jmrix/loconet/LnReporter.java particularly `lissyReport(..)`   

LISSY was an Uhlenbrock innovation.  http://www.rjftrains.com/intellibox/lissy-manual.pdf  That's a small, but avid, community.

There's support in RocRail which might provide some info: https://wiki.rocrail.net/doku.php?id=lissy-programming-en

bobjacobsen, 2019-01-21:
Looking at this further, I think we may be going in the wrong direction for Reporters.

It's real important that Reporter continue to report an Object.  That's fundamental to what it does: Provide a way to bring _any_ kind of information from the layout into JMRI, not just binary information.
- The only thing that the Reporter infrastructure assumes in common is that `toString` represents the _report_:  That's what's shown for the value in e.g. the ReporterTable.
- By allowing the report type to vary, you can _also_ allow more specific code to do more specific things:  A report from a VoltageReporter might have additional info, as might a TrainSpeedReporter.  

So if you want to make the LnReporter report something more specific than Object or String, that's fine with me.  Particularly if the `toString` of that is the same, so it doesn't break all those scripts.

But I think there's a problem with making it report IdTag:
```
IdTag represents a tag that might be attached to a specific piece of rolling stock to uniquely identify it
```

That's not a report.  It has some independent existence, hence I can see why it's a NamedBean, but it's not a report.

```
RailCom represents a RailCom enabled decoder that might be fitted to a specific piece of rolling stock to uniquely identify it
```

Ditto.  In fact, IdTag and RailCom objects may _be_ Reporters.  That might be the resolution to all of this.

pabender, 2019-01-21:
Let me address a couple of points....

Sent from my iPhone

> On Jan 21, 2019, at 11:12 AM, Bob Jacobsen <notifications@github.com> wrote:

> The only thing that the Reporter infrastructure assumes in common is that toString represents the report: That's what's shown for the value in e.g. the ReporterTable.
The recent changes made to add the reportable interface change that a little.  They still use the toString with with non-reportable objects, but toReportString is used for reportable items.
> By allowing the report type to vary, you can also allow more specific code to do more specific things: A report from a VoltageReporter might have additional info, as might a TrainSpeedReporter.
Right, that is still allowed.
> So if you want to make the LnReporter report something more specific than Object or String, that's fine with me. Particularly if the toString of that is the same, so it doesn't break all those scripts.
> 
That I am having a little trouble with.  In particular, I see enter and exit as events, and I don’t know how ( or why ) we would encode an event as part of the state of the tag.  ( as part of the state of the identifiable object that is )
> But I think there's a problem with making it report IdTag:
>
> IdTag represents a tag that might be attached to a specific piece of rolling stock to uniquely identify it
> That's not a report. It has [...]
I disagree because the IdTags and the RailCom tags ARE encoding the message received from the layout.

The actual report received from an RFID reader  is just the IDTag number.  The reporter adds information to the tag about when and where the tag was seen.

The RailCom objects potentially add additional information but only if the decoder and the reporter hardware report that to us.

pabender, 2019-01-21:
Sent from my iPhone

> On Jan 21, 2019, at 11:27 AM, Bob Milhaupt <notifications@github.com> wrote:
> Report the value of a CV as a result of an Ops-mode programming operation addressed to the mobile decoder when it is within a transponding "zone".

CV reading is handled by the ops mode programmer.
> At some future date, it may be useful to add ways for JMRI and/or scripts to send a "where are you" query (or a set of them) and update the appropriate Reporter object based on the returned "here I am" report(s).
> 
That could be useful.

pabender, 2019-01-21:
@bobjacobsen,

Sent from my iPhone

> On Jan 21, 2019, at 1:57 PM, Bob Jacobsen <notifications@github.com> wrote:
> 
> > The actual report received from an RFID reader is just the IDTag number. The reporter adds information to the tag about when and where the tag was seen.
> 
> This seems quite backward to me, so I guess I just don’t understand the hierarchy here. I’ll step back now.

This is the way it has been implemented.  ( I did not originate this code, but I have been working with it for a while).

RFID tags really just have a number inside of them.  The hardware gives us nothing more than that number, but the application we have wants us to know the where and when we saw that tag and what that is attached to ( which might be a piece of rolling stock or something else ).

It really isn’t any different than a turnout getting feedback information from another device.

Could the implementation change, perhaps, but I don’t how you would implement this as a reporter as you suggested.

icklesteve, 2019-01-21:
Hi folks,

I've been looking at MERG CBUS reporters recently.

CBUS reporters currently report the 5-byte IdTag, however there is a new format in discussion on the MERG forum which may change this to a 2-byte IdTag, with 3 bytes of user data.
I was contemplating extending the IdTag ( to something like CbusRC522Writeable ) to account for the user data element, maintaining the unique key in the IdTag.
 In this case, a "where are you" query would go to the IdTag table to get the reporter location, definitely not changing the reporter value,
Steve.

pabender, 2019-01-21:
So now that I am back looking at this on a larger screen, let’s back up a second.  Rather than look at what we do with the reports of a reporter. There are two general use cases ( along with more variations on the themes ):

1. The original application is taking some piece of information received from the layout, optionally doing some decoding of the information, and then displaying it ( either in a table or on a panel.) in human readable form.
2. Position tracking.  This is currently used in panels for block following and in operations for recording known locations of pieces of equipment, but there are other possible uses as well.

All of the reporters currently handle the first application just fine.  It is the second application where the current LocoNet reporter doesn’t really do what we want.

I am thinking of the variation of the block following case where  we have one block with an associated reporter followed by n blocks with occupancy detection but no associated reporters.  The report should follow the block detection activation/deactivation.

Both IDTag and RailCom reporters record the parsed information from the actual report by updating the Corresponding tag information accordingly ( at a minimum, location is stored as the last reporter to see the object, and the time of observation is recorded to go along with that ).  The object can then be passed from one block to the next, following the occupancy sensors.

LocoNet reporters currently report “1234 Enter“ or “1234 Exit” (or words to that effect) as a string.  @bobjacobsen has requested that the LocoNet reporter maintain it’s current report for compatibility with existing scripts... 

In the block following application, we certainly could pass the whole string from block to block, but we really just care about the number or some identifier we associate with that number.  

Because we might see both Enter and Exit reports, I don’t know how block tracking works with a LocoNet reporter now.  Presumably when a transponder moves from one block to the next, an exit report is generated by the hardware ( on actual Digitrax hardware, which I don’t have available.  My Z21 always gives an exit report for address 0 ).  If I am trying to follow blocks, I see some ambiguity ( and a possible race condition ) here.  In particular,does the exit happen before or after the transition to the next block?  And which content label or labels should move on?

@devel-bobm has pointed out there is also a “presence report” where a reporter identifies a particular transponder is in the associated transponding zone.  My assumption is that the “presence” report Enter or Exit, and my reading of the LocoNet reporter code is that that report is just ignored.

I am not sure how to resolve @bobjacobsen’s request to keep the reports the same with the existence of a report type we don’t handle, as noted by @devel-bobm, and an application where having the enter and exit in the report can cause different results.

devel-bobm, 2019-01-22:
@pabender, my experience with Digitrax transponding is that when a loco moves from one "transponding zone" to another "transponding zone", the "Entry" message can appear on LocoNet _before_ an "Exit" message, but it does not _always_ show up in that order.  I belive this variation relates to the quality of track/wheel pickup, but could also relate to vagaries of LocoNet message timings.  

There is also no requirement that "transponding zone"s are implemented on adjacent trackwork.  In other words, it is legal to have non-transponding trackwork between transponding zones.  For such paths, one would (likely) see "Entry" and "Exit" messages for "zone M" before seeing "Entry" and/or "Exit" from "zone N".

Similarly, the sequence of "transponding Entry" and "block occupied" messages can vary, and the sequence of "transponding exit" and "block clear" message can also vary.

Any scheme which relies upon a specific sequence of the LocoNet "Entry" and "Exit" messages, and/or any specific sequencing of those messages with respect to specific "block occupied" or "clear block" messages is likely to be "fragile".  

As an aside, with the BDL16x/RX4 transponding solution, there is not necessarily a relationship between a "transponding zone" and an "occupancy block".  (There is a direct relationship with the BXP88 and BXPA1, though.)

To the best of my knowledge, the Transponding "Presence Report" implies only that at the time the report was generated, the particular mobile decoder was within the "transponding zone".  How it got there is unknown, and whether it is still there at any later time may be difficult to ascertain.  When the mobile decoder is consisted using CV19, the decoder will not (so far as I can tell) generate any transponding activity at the consist address.  The only way a CV19-consisted decoder will transpond is if the decoder happens to receive DCC track signal messages to its "native" address.  Getting a "Presence Report" for "decoder 312" in "zone 15" is not reliable if that decoder has CV19 programmed to 121 and the consist is or has been moved since the "Presence Report" was seen.

This behavior seems to impose a limitation on transponding - if you want transponding "Entry" and "Exit" messages, the consist _must_ be controlled by addressing one loco of the consist at its "native" DCC address.

pabender, 2019-01-22:
> On Jan 21, 2019, at 8:35 PM, Bob Milhaupt <notifications@github.com> wrote:
> @pabender, my experience with Digitrax transponding is that when a loco moves from one "transponding zone" to another "transponding zone", the "Entry" message can appear on LocoNet before an "Exit" message, but it does not always show up in that order. I belive this variation relates to the quality of track/wheel pickup, but could also relate to vagaries of LocoNet message timings.
> 
This is exactly the race condition I was concerned about.

The other identification systems we use don’t really have the “exit” concept.  You have either seen the tag somewhere or you have not.

> There is also no requirement that "transponding zone"s are implemented on adjacent trackwork.  In other words, it is legal to have non-transponding trackwork between transponding zones.
> 
In this situation, we want block following to work between the two transponding zones.  Block following let’s us fill in the gaps.

> This behavior seems to impose a limitation on transponding - if you want transponding "Entry" and "Exit" messages, the consist must be controlled by addressing one loco of the consist at its "native" DCC address.
> 

This probably has something to do with Digitrax’s preference for Universal Consists.

pabender, 2019-01-22:
> I think that we have lost track of the problem.
> There are many physical reporters available, pre built or kit.
> The most basic is the sensor.
> Some reporters say something is here or not.. an occupancy sensor.

Occupancy sensors are very simply NOT reporters.

> Some reporters say car A just arrived or left.
> Some reporters say car A just arrived West bound.
> Some reporters say car A just left East bound loaded.
> Etc etc.
> JMRI needs be able to provide the infrastructure to support all these types of reports and munge them into a JMRI view of a reporter.

That looses sight of what My question was that started this, and that is how to very specifically handle LocoNet reporters which provide a string report.  

There isn't a one size fits all solution because different manufacturers have decided it is important to provide different information.  

We already have other reporters that handle RFID tags and RailCom devices.  Part of my motivation for doing this is I want to be able to use all three types of reports to identify where equipment is on the layout.  We can't do that with the 

> And while we are here dump the toString method, it is so messed up, just use getters and setters.

This statement looses sight of something @bobjacobsen said earlier:
> It's real important that Reporter continue to report an Object. That's fundamental to what it does: Provide a way to bring any kind of information from the layout into JMRI, not just binary information.

I have no intention of breaking the general pattern of allowing an Object as the report.  I don't know what the next application is that might want to use the Reporter infrastructure, which may or may not even have to do with the position of a piece of equipment.

Pugwash1, 2019-01-22:
> Occupancy sensors are very simply NOT reporters.
@pabender They are different bean types but from a business perspective they are the same. A sensor is just not very clever.
All the reporters and sensors should be giving all the information they have in a consistent way that can be easily data mined  and re-purposed to improve the efficiency and profitability of the railroad. Lets be prototypical. ;)
To that end, they should all give the number of objects they know about, their ids, their  velocity,  current/voltage number of shorts, last short etc etc.
Now back to the real world. In the case of a LocoNet Transponding Zone its an array of the number of objects it thinks are in it, and when they entered, and the other fields unknown. If one was to do the same thing with an on/off sensor its an array of 0-1 containing the id for an unknown object.
toString() should return what a sensors returns, so its consistent.
This is pretty much what we did back in the 80's before DCC with clipper, C, assembler, relays and bar code pen readers. We got velocity by having a bar code front and back of each car, the advantage of O scale. It worked some of the time.

pabender, 2019-01-22:
> Now back to the real world. In the case of a LocoNet Transponding Zone its an array of the number of objects it thinks are in it, and when they entered, and the other fields unknown. 

The array may work with Digitrax hardware, but it will not work for all devices that send LocoNet transpinding messages to describe block contents.

For example, the Roco Z21 I have connected to a Roco 10808 ( which is a RailCom detector ) reports the “Enter” messages for every address, but it only reports that 0 exited.

The LocoNet reporter code also handles Lissy messages and those report northbound and southbound.  I am not sure they ever show Enter or exit.

Pugwash1, 2019-01-23:
I dont know the details of either device, so this may be out in left field

> For example, the Roco Z21 I have connected to a Roco 10808 ( which is a RailCom detector ) reports the “Enter” messages for every address, but it only reports that 0 exited.

Assuming the message is decoded correctly Is it trying to tell you that there are no locos left in the block? Or if 2 locos go in and one comes out does it just report 0 exited?

> The LocoNet reporter code also handles Lissy messages and those report northbound and southbound. I am not sure they ever show Enter or exit.

This sounds like a classic block boundary device,  if it goes North it left the south block and arrived in the north block and vice versa.

pabender, 2019-01-23:
> On Jan 22, 2019, at 8:40 PM, Steve G <notifications@github.com> wrote:
> I dont know the details of either device, so this may be out in left field
> 
> For example, the Roco Z21 I have connected to a Roco 10808 ( which is a RailCom detector ) reports the “Enter” messages for every address, but it only reports that 0 exited.
> 
> Assuming the message is decoded correctly Is it trying to tell you that there are no locos left in the block? Or if 2 locos go in and one comes out does it just report 0 exited?
> 
It reports 0 exited whenever any device leaves the block.
> The LocoNet reporter code also handles Lissy messages and those report northbound and southbound. I am not sure they ever show Enter or exit.
> 
> This sounds like a classic block boundary device, if it goes North it left the south block and arrived in the north block and vice versa.
> 
I don’t have access to any of these devices either, so I am just looking at the code and what it shows.

The Roco 10808 connected to the Z21 can be setup to emulate a Lissy also.  I have not tried that yet.  In native CAN mode, the 10808. Will tell you which way the locomotive is facing on the track.  It is entirely possible that a Lissy does the same thing with northbound and southbound.

 ( Orientation is a RailCom feature supported by other RailCom detectors as well )

pabender, 2019-01-23:
I just had a thought on this.

Rather than worrying about the exact details of what is supported now or may be supported in the future by any specific reporter infrastructure we really just need to have a set of key/value pairs added to the IdTag interface.

Currently, the basic ID tag provides a way to store and retrieve identification information parsed from an on layout message.  It also has references that are supposed to contain the last reporter to modify the tag and the time at which it was modified.

By adding a collection containing key/value pairs, each individual reporter can add and maintain the key/value pairs it knows how to interpret for later retrieval.

I am going to play with this idea and see what it looks like.  More to follow.

pabender, 2019-02-07:
Closed via #6493

closed 2019-02-07 by pabender