JMRI/JMRI #6477: Manager LEGACY_PREFIXES

icklesteve, 2019-01-21:
In LEGACY_PREFIXES in Jmri.Manager, it's flagged that MR type is a legacy name which needs updating.
On shutdown, every MERG CBUS reporter gets flagged.
Seeing as MERG CBUS reporters use MR, ( like Sensors use MS ) I am assuming this is a false-positive.
Can the LEGACY_PREFIXES now be removed?

Ta,
Steve.

icklesteve, 2019-01-21:
Hi Bob,
Running JMRI via MERG simulator create a reporter with address "123" ( no quotes ).
Close JMRI and in the user prefs there's a file called legacy_bean_names.csv with MR123
Message also shows up when running PanelPro from ant,
Can replicate in 4.14 and head of master.

bobjacobsen, 2019-01-21:
Thanks. 

This is _exactly_ the reason I'm so insistent on getting to system prefixes being separately parsable.

I'm not sure how to fix this while still preserving any existing "MR"-prefixed files.  The plan (see the top of #4670; you can ignore the discussion below that) had another year or so before we had to obsolete those.  

Will work on it, not sure what the outcome will be yet.

icklesteve, 2019-01-21:
If it looks like a load of hassle leave as is.
We're aware of it, 
Most users won't notice it,
It won't be there forever!

Steve

icklesteve, 2019-01-28:
Now observing clean shutdown, thanks @bobjacobsen

closed 2019-01-28 by icklesteve