JMRI/JMRI #6487: NSIS usage needs documentation

devel-bobm, 2019-01-23:
Recent issues related to JAVA 11 support for Windows has highlighed that we do not have much documentation of our use of NSIS for Windows install package creation.   The only thing I see is in the instructions for using Netbeans IDE to develop JMRI code.

If anyone chooses to provide some JMRI NSIS usage documentation, a few things that ought to get some attention:

- we use a "special" NSIS package which supports "large strings", not the standard one available from the NSIS sourceforge.net site.
- Our automated build system does not automatically create the Launcher.exe file as part of the package build process (because of the "long strings" item, above).  Launcher.exe must be built locally and "committed" to the JMRI repository in order for changes to be reflected in the released package.
- Our installer and launcher implementations examine certain registry entries to find the JAVA JRE.  As JAVA changes, we may need to look at different registry entries, which means that the installer .nsi and the launcher .nsi file would need changes.
- Our installer implementation specifically calls out various .jar files which must be deleted in an "overwriting install".  That list needs to be kept up-to-date in the installer's .nsi file.
- "Version" numbering must be manually incremented when changes are made to the installer .nsi and/or the launcher.nsi file.

mattharris, 2019-01-23:
The need to use a 'special' build for the Launcher is documented in the 'Compilation Instructions' section near the top of the source file at https://github.com/JMRI/JMRI/blob/master/scripts/WinInstallFiles/LaunchJMRI.nsi#L14 but, admittedly, this isn't exactly conspicuous.

Until now, we've 'gotten away with it' as there have been relatively few changes needed with the launcher, and I've done the necessary compilation myself.

I do agree that this does need attention, hence have assigned this to myself to start putting something together.

bobjacobsen, 2019-01-23:
See also the [scripts/HOWTO-distribution.md](https://github.com/JMRI/JMRI/blob/master/scripts/HOWTO-distribution.md) documentation which has some NSIS info in the "local build option" section.

closed 2021-04-14 by stale[bot]