JMRI/JMRI #6488: Netbeans-related help needs NSIS-related updates

devel-bobm, 2019-01-23:
Netbeans documentation page needs update with respect to proper NSIS "special build" and proper installation into the JMRI development environment.

This partially addresses one sub-issue from Issue #6487 .

mattharris, 2019-01-23:
For the 'package-windows' target, there is no need for the 'special build' of NSIS to be used. This is only needed in the case for compiling the LaunchJMRI.exe launcher application.

dheap, 2019-01-24:
I had trouble using the `package-windows` target on Mac. I eventually got it working and documented it at https://github.com/JMRI/JMRI/issues/4628#issuecomment-438874501

Would be good to document somewhere...

dheap, 2019-01-24:
It is, thanks.

Suggest also at http://jmri.org/help/en/html/doc/Technical/NetBeans.shtml#nsis

closed 2021-04-14 by stale[bot]