JMRI/JMRI #6510: Eclipse as IDE is showing errors/warnings that 'ant warnings' isn't

bobjacobsen, 2019-01-29:
JMRI infrastructure uses a JMRI-distributed ecj.jar Eclipse complier to check for warnings and errors via the `ant warnings` and `ant warnings-ci` targets.  This is configured via the `java/ecj.warning.options` and `java/ecj.warning.options-ci` files.  From `build.xml`:

```
    <!-- The ecj.warning.options file defines which warnings are on and off -->
    <!-- By default, we set them explicitly, so that new ones will show up -->
    <!-- See http://help.eclipse.org/galileo/index.jsp?topic=/org.eclipse.jdt.doc.isv/guide/jdt_api_compile.htm -->
```

But this is not catching some errors and warnings that are showing when running with Eclipse as an IDE.  It would be good to fix this.

For context, see PR #6478 where this was uncovered.

bobjacobsen, 2019-01-30:
JMRI is using version 4.10 of ecj.jar now.

There seems to be up-to-date documentation of the options here:https://help.eclipse.org/mars/index.jsp?topic=%2Forg.eclipse.jdt.doc.user%2Ftasks%2Ftask-using_batch_compiler.htm&anchor=warn

Will take a look at adding some, but might be a few days.  Somebody else is welcome to start...

Pugwash1, 2019-01-30:
What I've tried was to add
warn:+nullAnnot(javax.annotation.Nullable|javax.annotation.Nonnull|javax.annotation.ParametersAreNonnullByDefault)
This nets 6300+ warnings and the one error for nonnull applied to an int.
If I then add these
-warn:-inheritNullAnnot
-warn:-nullUncheckedConversion
-warn:-nullAnnotRedundant
-warn:-nullAnnotConflict
I get 2000+ warnings and the 1 error.

bobjacobsen, 2019-02-20:
PR #6580 is a step forward on this.

We're far from being able to run on +nullAnnot due to the large number of warnings it generates, though. More work to be done on that....

danielb987, 2020-08-25:
> What I've tried was to add
> warn:+nullAnnot(javax.annotation.Nullable|javax.annotation.Nonnull|javax.annotation.ParametersAreNonnullByDefault)
> This nets 6300+ warnings and the one error for nonnull [...]

Can I run these tests from command line, for example by ant? And if so, how?

bobjacobsen, 2020-08-25:
```
% ant -p | grep warnings
 tests-warnings              compile test sources with ECJ warning generation on
 tests-warnings-check        compile test sources with ECJ warning generation on and fail if any warnings
 warnings                    compile source with ECJ warning generation on, omitting tests
 warnings-check              compile source with ECJ warning generation on and fail if any warnings, omitting tests
```

The ones with -check are meant for CI, so they issue a fail return code.  The ones with tests- include java/test, the ones without don't.

The control files are:
```
% ls java/ecj.warning.options*
java/ecj.warning.options
java/ecj.warning.options-ci
java/ecj.warning.options-test
java/ecj.warning.options-test-ci
```

danielb987, 2020-08-26:
I thought some of these items should be easy to fix, especially those related to Bundle. Bundle.getMessage() will never return null, so all Bundle.getMessage() can be annotated with @Nunnull, which is easy to do with a php script.

But I have run into trouble. I have changed `apps.startup.Bundle.getMessage` to:

`static String getMessage(@CheckForNull Locale locale, String key, Object... subs) {`

and changed `apps.startup.TriggerRouteModel.performAction()` at line 62 to:

`InitializationException exception = new InitializationException(apps.startup.Bundle.getMessage((Locale)null, "TriggerRouteModel.RouteNotDefined", this.getUserName()),`

The method getMessage() is annotated `@CheckForNull` for the locale, and performAction() gives a null for the locale. But checkstyle reports:
```
35. ERROR in F:\Projekt\Java\GitHub\JMRI\java\src\apps\startup\TriggerRouteModel.java (at line 62)
	InitializationException exception = new InitializationException(apps.startup.Bundle.getMessage((Locale)null, "TriggerRouteModel.RouteNotDefined", this.getUserName()),
	                                                                                               ^^^^^^^^^^^^
Null type mismatch: required '@Nonnull Locale' but the provided value is null
```
Am I missing something?

bobjacobsen, 2020-08-26:
Note sure if this is the cause, but did you change the superclass method that’s invoked by the next line:  `getBundle().handleGetMessage(locale, key, subs);`

If that `locale` is `@Nonnull`, perhaps it’s reasoning backwards.

closed 2021-04-14 by stale[bot]