JMRI/JMRI #6523: Track Current Meter Not Working

swkline, 2019-02-06:
Track current meter not working in JMRI. Reads 00.0% at all times.
Genuine Arduino Uno Rev3; genuine Arduino motor shield; genuine Polulu MC33926 motor shield. Jumpers and trace cuts are all correct.
TX LED on Uno blinks as expected when Track Current Meter is executed in JMRI, but track current meter does not update.
Tried with two different computers.
Otherwise works.

Cross posted to GitHub DCC++

pabender, 2019-02-06:
Can You send us a copy of the message log showing this?  

We need to see if the value is actually zero or if it is a non-zero value, and knowing that the TX light on the Arduino blinks doesn’t tell us that.

swkline, 2019-02-06:
Thanks for the quick response.  Excerpt from log follows.  The message repeats every .5 seconds or so when Track Current Meter is active.  Also, a DCC++ compiler warnings are included at the end.

07:48:57.346 ( 91401) ERROR [AWT-EventQueue-0] DCCppReply.java:407  - DCCppReply value index too big. idx = 2 msg = a10
07:48:57.377 ( 91432) ERROR [serial.SerialDCCppPacketizer Receive thread] AbstractMRTrafficController.java:1021  - Unexpected exception in invokeAndWait: {}java.lang.reflect.InvocationTargetException
    java.lang.reflect.InvocationTargetException
    	at java.awt.EventQueue.invokeAndWait(Unknown Source)
    	at java.awt.EventQueue.invokeAndWait(Unknown Source)
    	at javax.swing.SwingUtilities.invokeAndWait(Unknown Source)
    	at jmri.jmrix.AbstractMRTrafficController.distributeReply(AbstractMRTrafficController.java:1013)
    	at jmri.jmrix.AbstractMRTrafficController.handleOneIncomingReply(AbstractMRTrafficController.java:1056)
    	at jmri.jmrix.AbstractMRTrafficController.receiveLoop(AbstractMRTrafficController.java:868)
    	at jmri.jmrix.AbstractMRTrafficController$2.run(AbstractMRTrafficController.java:792)
    	at java.lang.Thread.run(Unknown Source)
    Caused by: java.lang.NumberFormatException: For input string: ""
    	at java.lang.NumberFormatException.forInputString(Unknown Source)
    	at java.lang.Integer.parseInt(Unknown Source)
    	at java.lang.Integer.parseInt(Unknown Source)
    	at jmri.jmrix.dccpp.DCCppReply.getCurrentInt(DCCppReply.java:765)
    	at jmri.jmrix.dccpp.DCCppMultiMeter.message(DCCppMultiMeter.java:39)
    	at jmri.jmrix.dccpp.DCCppTrafficController.forwardReply(DCCppTrafficController.java:85)
    	at jmri.jmrix.AbstractMRTrafficController.notifyReply(AbstractMRTrafficController.java:295)
    	at jmri.jmrix.AbstractMRTrafficController$RcvNotifier.run(AbstractMRTrafficController.java:1223)
    	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
    	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
    	at java.awt.EventQueue.access$500(Unknown Source)
    	at java.awt.EventQueue$3.run(Unknown Source)
    	at java.awt.EventQueue$3.run(Unknown Source)
    	at java.security.AccessController.doPrivileged(Native Method)
    	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
    	at java.awt.EventQueue.dispatchEvent(Unknown Source)
    	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
    	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
    	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
    	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
    	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
    	at java.awt.EventDispatchThread.run(Unknown Source)

DCC++ compiler log.  From DCC++ issue #42.  My error log is identical.

```
 R.currentBit=0;                                       /*   reset current bit pointer and determine which Register and Packet to process next--- */ \   

                                                                                                                                                    ^
```

C:\Users\David Danhauer\Documents\Train\BaseStation-master\BaseStation-master\DCCpp_Uno\DCCpp_Uno.ino:433100 warning: backslash and newline separated by space

```
   R.tempPacket=R.currentReg->activePacket;            /*     flip active and update Packets */ \        

                                                                                                ^
```

C:\Users\David Danhauer\Documents\Train\BaseStation-master\BaseStation-master\DCCpp_Uno\DCCpp_Uno.ino:449:103: warning: backslash and newline separated by space

}                                                                                    /* END-ELSE */ \

```
                                                                                                   ^
```

C:\Users\David Danhauer\Documents\Train\BaseStation-master\BaseStation-master\DCCpp_Uno\DCCpp_Uno.ino:450:88: warning: backslash and newline separated by space

```
                                                                                    \ 

                                                                                    ^
```

C:\Users\David Danhauer\Documents\Train\BaseStation-master\BaseStation-master\DCCpp_Uno\DCCpp_Uno.ino:196:59: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

CurrentMonitor mainMonitor(CURRENT_MONITOR_PIN_MAIN,"");  // create monitor for current on Main Track

```
                                                       ^
```

C:\Users\David Danhauer\Documents\Train\BaseStation-master\BaseStation-master\DCCpp_Uno\DCCpp_Uno.ino:197:59: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

CurrentMonitor progMonitor(CURRENT_MONITOR_PIN_PROG,"");  // create monitor for current on Program Track

```
                                                       ^
```

pabender, 2019-02-08:
it looks like JMRI is having trouble parsing the input received from the Arduino.

I've written a test that confirms that and will see what i can do about fixing it.

pabender, 2019-02-12:
@swkline did you ever tell us what version of JMRI you are using?

If you are running version 4.14, I think this issue was fixed by PR #6339.

I now have my test running, and I only made a 1 line change to the test setup to better match the way the code in question is called.

swkline, 2019-02-13:
4.15.2 works.  Thanks.
On a related note, does JMRI know that the current signal from the Populu is about 3x lower than the Arduino motor shield?

pabender, 2019-02-13:
@swkline 

The value JMRI uses for current is the value reported by the DCC++system divided by 1024.

1024 is the maximum value that can be represented by the 10 bit DAC on the Arduino.  This is true no matter what is connected to the input.  

In other words, JMRI has no way to know what what motor shield is connected, so the only choice we have is to use the maximum value the Arduino can report.

Since the fix is in 4.15.2, I am going to close this issue.

closed 2019-02-13 by pabender

harryplotter33, 2021-11-22:
Hi i have a problem like this issue i think !

I'm new to dcc but i like electronics and trains !
I start to make my dcc++ command station with arduino 1 r 3 and original motor shield.
I use a 19V 3.95 A switching power supply wich il connected to a LM2596 stepdown converter to make 15 V necessary to dcc++ system filtered by a large cap 2200 microFarad.

So i start to use this system with JMRI (now version 4.24) on a windows pc, but with a large type of different decoders i have the same problem!
So i can often power on and off lights and some functions, but if i start to run forward or backward the loco stops after few seconds. JMRI disable the power !
Why ?
in the JMRI console i get this message

**2021-11-22 22:46:21,544 dccpp.DCCppReply                      ERROR - DCCppReply 'p2' not matched by '' [AWT-EventQueue-0]**

what can i do ?

danielb987, 2021-11-22:
@mstevetodd 
Could this problem be a mismatch between JMRI version and DCCpp version? That the JMRI needs an upgrade to a newer version?

mstevetodd, 2021-11-23:
Not likely.
The newer test version might provide a clearer error message, but no version of JMRI is expecting a "2" as the power state (should be "0" or "1").
It sounds like he's running the original DCC++ (not the newer DCC++EX). It would be helpful to know the exact version he's running.

danielb987, 2021-11-23:
@harryplotter33 
Can you give us the complete output from the JMRI console?

harryplotter33, 2021-11-24:
Thanks to all, 
I was using uncompatible versions of dcc++ and JMRI; Now i'm working with the newer dcc++ex and jmri 4.24 (latest stable version) , all is ok ! SOLVED !!
Thank you very much !