JMRI/JMRI #6529: DCC++ issues since 4.14

MARCFOUR, 2019-02-06:
There is an issue reported regarding DCC++.  Throttle function commands go into a repeating cycle and the command is issues every .250ms  as seen in DCC++ Traffic Monitor. Speed commands and Track power on/off are not affected, only Function commands. Timeouts are reported and eventually PanelPro will hang and require a restart to recover.  

 I see this in JMRI 4.14 and in 4.15.2.  Person who reported it says 4.12 does not display this issue and he remains on 4.12 at present.

Marc N Fournier

atanisoft, 2019-02-08:
@MARCFOUR do you have more details on this? I have seen this with my ESP32 CS but not on the Arduino one. On the ESP32 it appeared to be due to a timeout condition in the LwIP stack which is still being debugged.

MARCFOUR, 2019-02-08:
Unfortunately I do not have much details except I can replicate what was reported on Groups.io:JMRI Users.  Running DCC++ simulator in JMRI 4.14 or JMRI 4.15.2 encounters the issue when I run the JMRI virtual throttle. Function commands are repeating over and over.  The user reporting the issue says his PanelPro encounters timeout and eventually hangs and assumes the hang is caused by the function commands repeating. I see them repeating every .250mil sec.  Can not add more than this.

MARCFOUR, 2019-02-08:
The Goups.io JMRI thread is 
https://groups.io/g/jmriusers/message/156639

atanisoft, 2019-02-08:
thanks, I'm not in that group but I'll take a look at the thread.

john92648, 2019-02-09:
Hi ,
I'm the user who has placed the message 156639. Here is the DCC++ Monitor Logfile which shows the over and over sending commands and caused what I think the timeout errors so I have each time to restart Panel Pro.
[monitorLog.txt](https://github.com/JMRI/JMRI/files/2847661/monitorLog.txt)
Otherwise I can not track Commands in the DCC++ Monitor because the runs of the function commands. Power ON/OFF and any Speed commands works fine.
Regards Hans

atanisoft, 2019-02-09:
Thanks Hans, I saw this once on my setup with my esp32 CS. I'll see if I can find any clues from your log and also see if I can get a bit more details from panelpro itself.

costing, 2019-02-17:
Hi,

This is related to a background function refresh feature that I have introduced in the DCC++ interface. In 4.14 it is set to a too high refresh rate that causes timeouts as the Arduino side cannot deal with all of them. It was fixed in the mean time and 4.15.2 and 4.15.3 should work fine. Could you give 4.15.3 a try?

250ms is the expected rate, at this frequency I don't have any timeouts and no interference with the programming track either.

The reason for the above is to automatically set the loco to the state it was in after a power cut (derailment, short, explicit power off/on ...).

Cheers,

.costin

atanisoft, 2019-02-17:
> 250ms is the expected rate, at this frequency I don't have any timeouts and no interference with the programming track either.

Any chance it can be disabled or have the frequency adjusted by the user?

> 
> The reason for the above is to automatically set the loco to the state it was in after a power cut (derailment, short, explicit power off/on ...).

The Arduino (and ESP32 CS) should be doing this anyway, at least outside of functions.. On the ESP32 CS the functions are sent periodically so JMRI shouldn't need to send these pings for that side (JMRI can detect the ESP32 CS already today).

costing, 2019-02-17:
The frequency is not tunable at the moment, but at this rate I don't see any problems. Do you have any issues with it?

Arduino seems to only refresh the throttle state but not for function flags. Or if it's supposed to do it it doesn't work for me. Any power glitch and the functions are lost.

I have no idea about the ESP32 implementation. If it indeed refreshes functions too then maybe based on the version string or another handshake between JMRI and the CS the background refresh could be disabled on the JMRI side.

Cheers,

.costin

atanisoft, 2019-02-17:
> The frequency is not tunable at the moment, but at this rate I don't see any problems. Do you have any issues with it?

I only saw it once on the ESP32 side that I can recall but that was with 4.15.2. If it turns up again I'll do some further digging.

> 
> Arduino seems to only refresh the throttle state but not for function flags. Or if it's supposed to do it it doesn't work for me. Any power glitch and the functions are lost.

The Arduino only supports throttle state, functions are only ever sent once. 

> I have no idea about the ESP32 implementation. If it indeed refreshes functions too then maybe based on the version string or another handshake between JMRI and the CS the background refresh could be disabled on the JMRI side.

As the author of DCC++ESP32 I can say with 100% certainty that with v1.2.0 (release being posted today) it does in fact send both basic loco speed data and functions at least every 50ms.

It appears that JMRI is no longer detecting the ESP32 via the status regex, it is still present but DCCppReply.java no longer uses DCCppConstants.STATUS_REPLY_ESP32_REGEX. This can be used to detect the ESP32 and be used to disable the refresh automatically.

PUlvestad, 2019-02-18:
https://groups.io/g/jmriusers/topic/29892069#157059

gbglacier, 2019-02-19:
As reported, even when stop is issued, or when the throttle is released/dispatched, JMRI continues to send function command packets to the DCC++ base. The continuous stream of packets in the least, will keep sound decoders operating instead of timing out to mute, as many modelers are configuring them today. The only way to stop the packets being generated is to shut down JMRI. This is unexpected behavior, which I would consider to be a bug.

costing, 2019-02-19:
Hi Geoff,

I haven't encountered this kind of sound timeout in the locos so far, that's an interesting feature. I would have thought that once you are done with a loco you could park it and turn the sound off before releasing it. But to each his way.

As far as I can tell JMRI forever remembers the state of a loco that was touched at some point. Releasing it it just a logical operation in the client (GUI, Engine Driver) that doesn't reset any function states. When you reacquire it you find it in the same state you left it in. That state must be refreshed in the loco so that (due to any glitches or whatever) the loco doesn't behave differently than what the interface shows. I for one am tired of flipping sound off/on again after every glitch. And the same for every other function (or function group to be more precise).

I don't agree that the loco should be allowed to behave in an unexpected way. JMRI's state should be _the_ authoritative information, only an explicit state change in it should be reflected in a layout change.

If I understand correctly the scripting features of JMRI, such timeouts could be implemented in that framework. And it would be the correct way to do it since it would turn the function off explicitly. Or making sure it is on when it moves. I'll look into doing the latter, it's an intriguing option to automatize small things.

Cheers,

.costin

dheap, 2019-02-20:
> @costing I haven't encountered this kind of sound timeout in the locos so far, that's an interesting feature. I would have thought that once you are done with a loco you could park it and turn the sound off before releasing it. But to each his way.

It's actually been in some brands of decoders (QSI, SoundTraxx, ...) for many years.

I agree fully with your philosophy (as does ESU, which does not provide such a feature). But not every modeller wants realism in this aspect so the market caters for modeller demand.

I don't mind it so much for steam locos, but it really irks me with diesels.

Dave

gbglacier, 2019-02-20:
> As far as I can tell JMRI forever remembers the state of a loco that was touched at some point. Releasing it it just a logical operation in the client (GUI, Engine Driver) that doesn't reset any [...]

Unfortunately, there are more problems with this approach the more you get into this. The difficulty is beyond JMRI. JMRI does not necessarily see everything going on in a layout. You can restrict JMRI to only operating portions, and even restrict it to operating one of more than one base station by design. I don't think it's a good idea to believe JMRI has access to all data. It doesn't. And refreshing DCC packets to a dispatched/released decoder may not update the functions, if the base station reassigns the slot or queue for that loco. What if the loco is put on a disconnected spur? Or reprogrammed by address without JMRI visibility. Not every modeler routes everything though JMRI. And if you continue to send function packets to all addresses ever seen, does it tie up the slots (registers in DCC++) for the loco address? I don't know enough to say. DCC++ has a refresh parameter when a function is set -- default setting is 4 refreshes. If you really want repeated refreshes why not change that parameter? It's already there. Unexpected system behavior to the end user in an electronic system is not a good idea, period.
Sound decoder time-outs are definitely there and definitely inconsistent. But there are enough decoders out there whose sound times out upon loss of packets (in conjunction with the stopping by loss of packets) to be concerned with this.
I think the easiest way to deal with this is t set a configuration variable in the setup to limit by time or count, and give users an option.
I am not a JMRI implementer, just a user. but the problem you are trying to solve has more ramifications than you think, for other layouts, decoders, and operators.
Please give this more consideration beyond where this started.

Geoff

bobjacobsen, 2019-02-20:
> You can restrict JMRI to only operating portions, and even restrict it to operating one of more than one base station by design.

Could you say a bit more about what you mean by this?

gbglacier, 2019-02-20:
>  You can restrict JMRI to only operating portions, and even restrict it to operating one of more than one base station by design.
Could you say a bit more about what you mean by this?

Sure Bob. 1. Default setting can assign throttles to one base station and other duties to another 2. Layouts do not have to route all throttle controls for all power districts through JMRI. There is typically one base station for throttle DCC, but nothing stops a modelers from operating two sections of a layout from 2 different base stations, or operating signaling and lighting from one base station and throttle control from another, and ignoring power on board radio control to boot. And if I am operating a DCC throttle via JMRI with one base station, I can still operate a throttle wired to the base station directly.
Moreover in the age of Arduino based hardware interfaces, there is no dependency on JMRI if I route a separate control mechanism into, for example, the DCC++ paket control registers. In fact, given DCC++ structure, it would be relatively easy to add.
I think distributed control is more likely than ever to come on board as things progress. It's possible that if LCC takes off, we might be looking at a greater need for multiple cooperative, near independent systems.
My own path is leading me to use multiple DCC systems in parallel for loco control, signaling, animation, and lighting. DCC loco control is incidental, not fundamental to my modeling.
This is likely more than you were looking for, but it is how I am progressing.
Thanks again for your interest. It is appreciated.

>

bobjacobsen, 2019-02-20:
> There is typically one base station for throttle DCC, but nothing stops a modelers from operating two sections of a layout from 2 different base stations

Thanks for the clarification.  What you describe is quite difficult to do with DCC if trains are to go from one region to another.  One large European modular group (FREMO) has come up with a way to do it so they can run _many_ trains, but it’s complex.   http://dcc-mueller.de/switch/switch_e.htm    I’m not sure that’s the use case that should drive this discussion.

gbglacier, 2019-02-20:
Bob,
I don't think it is either. I think the point is that JMRI should not be driving a decoder that is effectively shut down, ie. released dispatched. Otherwise, once JMRI accesses a decoder, it is "live" until JMRI shuts down? That does not make sense to me.. Does it to you? And if yes, can you explain the rationale for it, so I can understand?

bobjacobsen, 2019-02-20:
> I don't think it is either. I think the point is that JMRI should not be driving a decoder that is effectively shut down, ie. released dispatched. Otherwise, once JMRI accesses a decoder, it is [...]

From inspecting the code, this appears to be what the SPROG does (though it doesn’t have the queuing problem previously described, which might be the root of all this).  It also seems to be what ESU and NCE command stations do on their own, without JMRI’s help. Under some circumstances, Digitrax command stations will do this, at least until a quite-long timeout happens.  I don’t think Lenz or EasyDCC do it, though I haven’t really checked.

Why do this?  See previous post by @costing:  Some people think it’s more realistic.  They like it.  Others care more about other things, and don’t like it.

It’s basically futile to try to figure out a “one true way” for DCC systems to work in this respect or many others.  Different people prefer the advantages of different approaches.  With something like an SPROG, which is aimed at a particular market and particular users, you’ll get one approach.  With a high-end NCE or Digitrax command station, you’ll get another.  

It may be that DCC++'s hardware and software are malleable enough that it can live optimally in several worlds:  Sometimes doing one thing, sometimes another.  Normally that’s not a great idea, because it can get complicated (and sometimes argument prone) to set up things that come in multiple different flavors. And model railroaders tend to succumb to technological obsession (I’ve heard somebody describe how to pick DCC addresses containing lots of 1 bits because they can be sent faster than ones with lots of 0 bits) which makes a live-and-let-live approach hard to support.

gbglacier, 2019-02-20:
So how do you shut down a loco being serviced by JMRI? Ever? ...or... what is the rationale for transmitting to it until JMRI is shut down? You still haven't explained this.

bobjacobsen, 2019-02-20:
> So how do you shut down a loco being serviced by JMRI? Ever? ...or... what is the rationale for transmitting to it until JMRI is shut down? You still haven't explained this.

Actually, I have. Twice.  But once more for the record:

The systems I mentioned earlier _don’t_ “shut down a loco” by themselves.  If you want the loco to do something, you tell it to do something.  I.e. F0 to turn off the headlight, F8 to turn off the sound.  With many decoders and systems, that headlight doesn’t go off until you turn off the command station (or at least track power). What does your DCC system do?

Bits that you wouldn’t otherwise send don’t cost anything when you use them for DCC refresh packets.  Why stop sending them?

gbglacier, 2019-02-20:
What does your DCC system do?
Ans: When I dispatch or release it stops sending packets. You are confusing the state of the decoder with the dynamic activity of the base station.
Bits that you wouldn’t otherwise send don’t cost anything when you use them for DCC refresh packets. Why stop sending them?
If the loco is active, I completely agree, my question was how do you stop the packet activity? The base station is acting of activity JMRI initiates. Do you even know whether all base stations generate those packets even to a dispatched loco? DCC++ appears to do so. Others may not.

bobjacobsen, 2019-02-20:
Not going to get into discussion on credentials. If you think I don't know what I'm talking about, you're welcome to your opinion about that. Best of luck with you're further work.

gbglacier, 2019-02-20:
Sorry Bob, this has nothing to do with your credentials. I thought we were discussing what base stations actually do. I am not a JMRI developer. I yield to you and your group. There is always 4.12.
Good luck in all your adventures.

pabender, 2019-02-20:
> On Feb 20, 2019, at 12:04 AM, Geoff Bunza <notifications@github.com> wrote:
> If the loco is active, I completely agree, my question was how do you stop the packet activity? The base station is acting of activity JMRI initiates. Do you even know whether all base stations generate those packets even to a dispatched loco? DCC++ appears to do so. Others may not.
> 

If DCC++ isn’t behaving as expected, then the issue is in the DCC++ code.

I don’t know enough about how DCC++ works to know if the cause of the problem you describe is in JMRI or in the base station code.

dheap, 2019-02-21:
> It also seems to be what ESU and NCE command stations do on their own, without JMRI’s help.

There's no concept of releasing locos with an NCE system.

If the loco has speed commands of zero for certain (settable) speed packet count, the loco is removed from the Track Queue and no further speed/direction or Function Refresh (if enabled) packets are sent.

The loco entry may remain in one or more Cab Context Pages in command station memory, but these are ignored unless updated by the user changing something on the Cab (if still powered up). The Cab context pages even persist across command station restarts (although speed and function states are set to cleared, but no packets are sent to the decoder) and a cab (which has no memory) picks that memory up when turned on/connected.

In the case of JMRI throttles, JMRI can send speed/function group etc. commands to the command station. The command station has only one shared Cab Context page for all JMRI throttles and so each command overwrites the shared context page.

This is all rather long-winded (and a lot of detail has been omitted) and not of direct relevance to the DCC++ case, but I detailed it to explain why NCE systems have no concept of dispatch/release of either command station entries or the decoder itself and they manage quite well with JMRI as it is.

In the end, I agree with others that since JMRI is not actually generating the raw DCC track packets (as in the SPROG case) for DCC++, function refresh should be a command station function, not a JMRI function.

Dave Heap

atanisoft, 2019-02-21:
Perhaps in the case of DCC++ it can be an optionally enabled feature in JMRI to send these "refresh" packets, in addition to user adjustment of frequency

pabender, 2019-02-21:
Looking at the code, I see a couple of issues surrounding the referesh, which can be found in the file:

https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/dccpp/serial/SerialDCCppPacketizer.java

Here are some suggestions based on what i see:

1. The refresh is handled by a refresh thread, but the thread is never stopped.  It might be better to turn this into a timer.
2. It appears the refresh thread will refresh functions even if NO throttles are active.  There needs to be some check that throttles are active before refreshing functions.
3. This is really an application feature, but it is implemented at a lower level in the communication stack.  I think it would be better to push this up to a higher level, perhaps to the throttle code.  Moving this up to a higher level will allow interfaces other than the serial interface to take advantage of this.

costing, 2019-02-21:
Hi @pabender 

1. In practical terms this would only save a thread if you already have a shared executor service somewhere. I'm not familiar with the core to tell where it would be appropriate to attach it. If a new executor is needed then nothing is saved. I don't have a strong opinion on this, either of the approaches is ok for me.

2. Well, yes, and it's a feature :) For example, after a short and power cycling I want everything to be back as it was (i.e. sounds, car interior light ...). I don't want to keep the wagon throttles open to keep the lights on. The internal JMRI state should be reflected fully in the circuit state, as soon/often as possible.

3. I have a setup with 2 JMRI instances, like this:
JMRI (DCC++ TCP client) -> JMRI (DCC++ TCP Server ; DCC++ serial connection) -> DCC++ hardware

I found it best to implement it in the lowest possible layer, the serial interface implementation. This is related to some of the above comments. If/when the hardware would implement it then it could be moved one layer lower even (i.e. removed from JMRI completely).

Cheers,

.costin

atanisoft, 2019-02-21:
> If/when the hardware would implement it then it could be moved one layer lower even (i.e. removed from JMRI completely).
I wouldn't count on the Arduino DCC++ code to be updated anytime soon. The project appears to not have an active maintainer at this time.

On the ESP32 (which supports the DCC++ protocol) it is already implemented and available today to not require JMRI to send the update packets.

bobjacobsen, 2019-02-21:
> In practical terms this would only save a thread if you already have a shared executor service somewhere. I'm not familiar with the core to tell where it would be appropriate to attach it. If a new [...]

One approach is to use `ThreadingUtil.runOnLayoutDelayed(ThreadingUtil.ThreadAction ta, int delay)` (see [the Javadoc here](http://jmri.org/JavaDoc/doc/jmri/util/ThreadingUtil.html#runOnLayoutDelayed-jmri.util.ThreadingUtil.ThreadAction-int-)) and have the task reschedule itself until it's time to stop.

Or you can write a repeating [`java.util.Timer`](https://docs.oracle.com/javase/8/docs/api/java/util/Timer.html) operation. JMRI has a common Timerimplementation to reduce Timer thread use when handling repeating operation.  See the [TimerUtil class](http://jmri.org/JavaDoc/doc/jmri/util/TimerUtil.html)

costing, 2019-02-21:
@atanisoft Do you know if it is planned to port it to the Arduino version too?

@bobjacobsen Great, thanks, will look into it.

atanisoft, 2019-02-21:
@costing as mentioned above, don't count on it. I'm not a maintainer for the DccPlusPlus project and can only maintain my esp32 CS. The Arduino DCC++ has limitations built into it's DCC signal generation as well as locomotive state (it doesn't track functions at all, they are single shot commands sent in a fire and forget manner).

On the esp32 there is a lot more flexibility and memory available so the full state is stored in memory and sent periodically to the track (at minimum every 50ms).

costing, 2019-02-21:
@atanisoft I was afraid this was the case... I'll try to find the esp32 board in my project boxes, your project sounds really cool. Then I would also have a testing platform for the JMRI side of it.

atanisoft, 2019-02-21:
>Then I would also have a testing platform for the JMRI side of it.
Feel free to ping me offline and we can discuss the CS etc. I'm actively working on integration LCC into it and would like to also explore LCC support in JMRI (beyond node list)

mstevetodd, 2020-07-21:
I ran into this same behavior trying to use the DCC++ Traffic Monitor.
However, I notice that there are some controls in the DCC++ Traffic Monitor window itself that seem to control this refresh.
[image]
So, perhaps these commands are only sent while the Monitor window is open? I'll figure how to check for this on the Arduino side.

mstevetodd, 2021-06-01:
The JMRI connection to DCC-EX now disables this refresh on the JMRI side, as it is handled in the DCC-EX code.

closed 2021-06-01 by mstevetodd