JMRI/JMRI #6538: Decoderpro - Lenz  - command station error

187king, 2019-02-08:
I have a lenz system, LZV100 v3.6, LH100 v3.6, LI-USB 23150.
I have a PowerPax between the control module and the programming track
Decoderpro v4.15.2
Win 10
Java 1-8.0_201-b09

I have used various versions of decoderpro for ~8-10 years.   Recently I exchanged ~45 messages with the JMRI resources for a problem (#6471) identifying a Rapido SW1200RS sound equipped loco decoder.  According to the manual it has an ESU Loksound select decoder.  Rapido has agreed to look at the engine if I return it indicating the decoder may be faulty.  My recent problem is the decoderpro program now indicates the 'command Station disconnected, or powered down' so nothing works now?  I have performed a command station reset, decoder reset, etc. to no avail.  Since getting this error I can/have set the long address and read cv's >255 with the handheld.
 
When I try and identify a new loco I get the error, and I am unable to reach the system and program any locos as a result of that error. 

On the USB device, the USB light is solid green, the xpressnet light is solid red, the data light remains off, but it does flash (as it should) when I do program via the handheld so I know it can communicate.

2019-02-07 12:41:15,637 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2019-02-07 12:41:15,693 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Greg\JMRI\log\messages.log [main]
2019-02-07 12:41:15,695 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Greg\JMRI\log\session.log [main]
2019-02-07 12:41:15,722 apps.AppsBase                         INFO  - DecoderPro version 4.15.2+Rfda0762 starts under Java 1.8.0_201 on Windows 10 x86 v10.0 at Thu Feb 07 12:41:15 CST 2019 [main]
2019-02-07 12:41:22,365 gui3.Apps3                            INFO  - Starting with profile DecoderPro_4.8.3edd1b5d [main]
2019-02-07 12:41:23,108 node.NodeIdentity                     INFO  - Using jmri-nApZGcvarhNiaaacfBi_Sr-3edd1b5d as the JMRI Node identity [AWT-EventQueue-0]
2019-02-07 12:41:23,783 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "XpressNet" [main]
2019-02-07 12:41:24,653 liusb.LIUSBAdapter                    INFO  - COM3 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: true  CD: false [main]
2019-02-07 12:41:54,690 lenz.XNetInitializationManager        WARN  - Command Station disconnected, or powered down assuming LZ100/LZV100 V3.x [main]

I tried getting help from LenzUSA but after leaving a message ~ a week ago I have not receive a reply, then tried Lenz in Denmark(?) with no response, so I am reaching out further.  

Any assistance / suggestions you can offer would be greatly appreciated.

Thanks

pabender, 2019-02-08:
> Java 1-8.0_201-b09
>
> I have used various versions of decoderpro for ~8-10 years. Recently I exchanged ~45 messages with the JMRI resources for a problem (#6471) identifying a Rapido SW1200RS sound [...]

When this error occurs, there really are only two possible causes, both of which are listed in the warning message ( more on that in a second ).

The warning message itself is generated because we were unable to identify the command station after trying for 30 seconds.

> 2019-02-07 12:41:24,653 liusb.LIUSBAdapter INFO - COM3 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: true CD: false [main]
> 2019-02-07 12:41:54,690 lenz.XNetInitializationManager WARN - Command Station disconnected, or powered down assuming LZ100/LZV100 V3.x [main]

Assuming the power is on to the LZV100, you have either an issue with a cable ( or a connection that involves a cable ) or you have something mis-configured.

Here are the things you need to check:
1) On your LIUSB, you should see two LEDs illuminated illuminated, one for USB and one for XpressNet.  If these are not lit, you you have a cable disconnected somewhere.
2) verify that your LIUSB is still connected to COM3.  Unfortunately, if you move the USB cable from one USB socket on your computer to another, that is enough for Windows to change the port it is connected to.  Moving it back will likely fix the issue, otherwise, you need to reconfigure JMRI to use the new port.
3) Verify you don’t have a bad cable.  The LEDs you checked above tell you if you have power, but they don’t tell you if the communication wires are ok.
4) if you are using the screw terminals on the LZV100 for the LMAB connection’s to the XpressNet, verify all the wires are still connected.
5) verify the pins in any RJ12 jacks are not bent so they don’t make@ connection with the modular plug.

pabender, 2019-02-08:
Incidentally, if you have a handheld with a din plug on one end of the connector and a modular plug on the other, like comes with an LH90, you can use that cable connected to the DIN plug on the back of the LZV100 to quickly eliminate the XpressNet wiring.

187king, 2019-02-08:
Thanks Paul,  got the USB port # figured out after your reply, now it works
Alway something to learn with MS,  Much appreciated

pabender, 2019-02-09:
Good.  I am going to close this issue since it is resolved

closed 2019-02-09 by pabender